keirf/flashfloppy #381: IMG.CFG giving Error 31 after update to 3.20

dallday, 2020-10-20:
I was using version 2.14 of flash floppy and all was working. I have now updated to version 3.20 and it no longer seems to like my IMG.CFG file.
All the images on the USB stick show *err*31 on the oled display and will not load.
The same USB stick works fine under 3.19
I tried adding a tracks parameter and this works fine in version 3.19 but not in 3.20
Not sure what I have done wrong or what I need to do to correct this

IMG.CFG File contains
# IMG.CFG:  polydos 2 discs
[default]
cyls = 35
heads = 2
#   interleaved:   Interleaved cylinder ordering: c0s0, c0s1, c1s0, c1s1, ...
file-layout = interleaved
tracks = 1-35
secs = 18
bps = 256
id=0
mode=mfm
iam = no

Thanks for a brilliant product 
David

keirf, 2020-10-20:
The default tag is now `[]` instead of `[default]`. You don't need the `tracks=` line so get rid of that.

dallday, 2020-10-21:
Thanks - that works

closed 2020-10-21 by dallday