keirf/flashfloppy #384: Openflops akai s900 not compatible

j0vius, 2020-10-26:
I purchased A openflops board on recommendation to use with a Akai s900 , I just couldn't get it working consistently, got A  gotec and it worked pritty much out the box used the same pen drive same ff config same disk images as before all fine . Updated to latest Ff and used lots of different images all seam fine so unless I had a faulty board the is some incompatibility between open flops hardware and Akai s900 and I wouldn't recommend anyone getting one

simulant-systems, 2020-10-26:
Hi thanks. the Akai s900 isn't actually listed as being a compatible host platform for FlashFloppy.

So providing you use the host platforms listed the OpenFlops hardware has not issues. But for s950 we'd probably need @keirf to test further with Openflops to see if he can add it. Thanks for the feedback.

j0vius, 2020-10-26:
Agreed .. I should say I wouldn't recommend getting one to use with a Akai s900 . . .

keirf, 2020-10-26:
I'd suspect a problem with the Openflops itself, as it should operate just as a Gotek. That's the point of it.

It can on occasion be cached config, and resetting cached config is worth a try (hold both buttons a few seconds after power on, with no stick inserted, until display changes).

Apart from that: Same stick, same fw version, same jumpers: Should behave the same!

j0vius, 2020-10-26:
I did reset after any change I made so I doubt that it's that

simulant-systems, 2020-10-26:
Must of just been a bad board. So far we've sent out over 100 of the OpenFlops boards and we've not had any bad feedback or issues yet except this one. So I suspect you must of received a faulty one perhaps. I'll note we did then send you a standard Gotek as you requested too, so this has been replaced for you. Thanks again and we'll see if anyone else experiences this or if it was a one-off issue.

j0vius, 2020-10-26:
You could test the board I sent back to you to see if it's working in another system ? Or if you want to send me another open flops I would happily test it in my s900  now that I know it's not my hardwhere

simulant-systems, 2020-10-26:
> You could test the board I sent back to you to see if it's working in another system ? Or if you want to send me another open flops I would happily test it in my s900 now that I know it's not my hardwhere

Hi thanks. I actually used the board you sent to install in my Tatung einstein, as I needed another drive in it anyway. And it is working fine in that system so far. So an odd problem indeed!

keirf, 2020-10-26:
I don;t think we can conclude any more at this point.

closed 2020-10-26 by keirf