keirf/flashfloppy #385: Korg 01W FF.cfg to do and error e34

chacals84, 2020-10-29:
Hello,

I have 2 Korg 01W : one FD et One Pro X.

I would like to put gotech usb floppy emulator, the korg 01w have a 720 Ko 2DD disk drive.

My 2 gotech are 1,44 one. I put on one gotech flashfloppy and i have error E34 when i put my usb stick.

I format my usb stick in fat32, and i have this error, i could choose a floppy disk on 100 i have on usb stick.

I put FF.CFG with akai for interface. Do i have to do Img.cfg files  too ? And what parameter do i put in ?

Thank you for your answer.

Best Regards.

P.S. : excuse my bad english, i am a french guy ;) ;)

keirf, 2020-10-29:
What are the image filenames? They should be IMG files (with suffix .IMG) and then they should be detected and work ok.

chacals84, 2020-10-29:
Hello, thank you for your answer.

I have no image on the 100 floppy disk, i would like to choose a disk to format it with the korg menu, and save the floppy disk i have.

I would like to load with the real floppy disk and disk drive and after save on usb stick.

I couldn't do that ? I must have a img file on floppy disk to enable it with flashfloppy ?

Thank you for you answer

Best regards

keirf, 2020-10-29:
Yes, but if you can format it on the Korg and you are sure the disk format is PC-style 720kB disks (right?) then you can just make an exactly 720kB size file called whateveryoulike.img on the USB drive.

chacals84, 2020-10-29:
No, i don't know the format of disk, i would like to knw what FF.CFG or/and IMG.CFG i have to do to make it works. 

I will try tomorow to do like you said a file *.img of 720 Ko ;)

I saw for 01w that we must take the shugart type

thank you, i will tell you if it works.

Best Regards.

keirf, 2020-10-29:
I think the format is probably 800kB. Anyway, attached here is a 720kB IMG and an 800kB IMG. Both need formatting. You can try formatting/writing each one, and see which works.  Then you can simply format the working one, and then copy it as many times as you like so that you always have a fresh formatted disk image to hand.

[img.zip](https://github.com/keirf/FlashFloppy/files/5461469/img.zip)

chacals84, 2020-10-29:
Ok, thank you, but, how i format the img you give  ? Sorry, but, i know PC, but not Korg stuff ;) ;)

keirf, 2020-10-29:
Doesn't the Korg have a "format disk" option in its menus? I know Korg even less than you, I'm sure.

chacals84, 2020-10-29:
yes, it had, but, i have error E34 on screnn and i could not format with korg menu ;)

I write the image you give to me on 000 et 001 disk, i try this before i go to sleep ;)

chacals84, 2020-10-29:
I format my usb stick in fat 32 mode, without make 100 floppy disk (i have a 4 Go usb stick).

I put you 720.img i test i have E31 error, and i test 800.img and same error.

I could i make a 713 Ko Img ? How do you create a xxx Ko Img ?

keirf, 2020-10-30:
Apologies I made a typo creating those images and they're the wrong size!! Hence not recognised.

Here is a replacement zip file with correct sized images :+1: 

[img2.zip](https://github.com/keirf/FlashFloppy/files/5463366/img2.zip)

chacals84, 2020-10-30:
And, me, i forgot to put FF.CFG to have interfac = akai for korg synthetiser.

I test your image when i return from my job office ;) ;)

Thank you for you rapid answer, if you have a donnation page, give it to me.

keirf, 2020-10-30:
Just to note, the line you need in FF.CFG is `host = akai`. Probably for interface you want `interface = shugart` but that is the default anyway.

Donations page: https://github.com/keirf/FlashFloppy/wiki/Donations

chacals84, 2020-10-30:
yes you are right "host" not "interface"

ok for donation page

chacals84, 2020-10-30:
Yes, yes and reyes it works with the img2.zip. and FF.CFG file with interface = shugart and host = akai.

I took the 720.img file, format it the korg 01w show "formatting" the usb floppy emulator % of formating and korg 01w show "coompleted. I put back original floppy disk drive, load original file A, pub back the usb floppy emulator and save on usb stick the image.

How could i put many img file to have multiple image file with multiple floppy disk.

Thank you for your helpness.

keirf, 2020-10-30:
If all you have is the three-digit numeric display, you may want to run in indexed mode which is described here: https://github.com/keirf/FlashFloppy/wiki/Initial-Setup#indexed-mode

Essentially all your files are given a numeric name, and they correspond to the 3-digit number on the display.

Then you can simply duplicate your IMG file as many times as you like: DSKA0000.IMG, DSKA0001.IMG, DSKA0002.IMG, ...

chacals84, 2020-10-30:
I test with 2 images i name FDUFILEA.IMG and FDUFILEB.IMG that correspond on the preload disk file A and B i saved.

Then i have my 3 digit on goteh drive give 000 and 001 for the 2 files, but i don't know which is 000 and 001, but, i have to put oled screen,  rotary button and speaker i buy.

Many thanks i go on donate page you give to me to put 10 euros for you, it's not a lot, but, it's all i could do.

Best Regards.

chacals84, 2020-10-30:
Ok, i find to put all image files with indexed type 0000_myfilename.img (no prefix i take this option). On my 4 Go i could put 5000 720 Ko disk, cool ;)

keirf, 2020-10-30:
That's right you can have any suffix (_myfilename) and the prefix is configurable and can even be nothing at all. I guess you read about that :)

closed 2020-10-30 by keirf

chacals84, 2020-10-30:
Yes, i do donation, i hope you could do good works with it. You could close this issue, thank you very much. Best regards.