keirf/flashfloppy #386: Need Help Adding Disk Images

DickWiggler6, 2020-11-05:
Hey, stumbled upon this website after looking pretty much everywhere else. I downloaded the software in order to format and add disk images to my USB so I can use it with my gotek that I have installed on my mpc(drum machine). But the software doesn't seem to work. I used one version which I think was called floppy manager. It showed that it added partitions to the USB but when I put it in my gotek it said error 34. The other version was called vfd or something like that and I tried to use the multi block function but it said some crazy worded sentence that didn't make much sense but it basically said that something on the computer could be using the USB some how and so it can't complete the function. Any answer will be appreciated I already have a USB that came with the guy I bought my mpc from but I would like to know how to be able to use more than one USB.

keirf, 2020-11-05:
Format the USB stick FAT32 and put IMG files on it. Each IMG is a single disk image file: mounting one of these in FlashFloppy is like inserting a disk into the disk drive. Don't use software like the floppy manager -- that's for the original Gotek firmware and won't work with FlashFloppy.

DickWiggler6, 2020-11-05:
Okay how would I add individual disk images without software? I'm pretty new to this kind of thing. Had no idea what a floppy emulator was until I found out the mpc had one installed.

keirf, 2020-11-05:
What images have you got? What are you actually trying to load into the MPC? Or are you trying to save stuff from the MPC?

DickWiggler6, 2020-11-05:
Not sure what you mean by what images. And yes  I'm saving samples onto the USB from the mpc. I only have 50 partitions on the USB that I bought with it. Id like to be able to keep using additional USB sticks so I don't have to eventually install a regular floppy drive or an sd card reader or something.

keirf, 2020-11-05:
Okay so you have a working USB stick?

keirf, 2020-11-05:
And your Gotek is running the FlashFloppy firmware?

DickWiggler6, 2020-11-05:
Yes and yes it says v 11 so I'm guessing it's version 11.

DickWiggler6, 2020-11-05:
Would it be possible to take disk images from the working USB I have and drag them into an empty one?

DickWiggler6, 2020-11-05:
Copy them I mean

keirf, 2020-11-06:
There is no v11. Maybe attach a picture of your display.

But anyway yes you can simply copy/drag images onto a fresh stick.

DickWiggler6, 2020-11-06:
[image]

keirf, 2020-11-06:
version 0.11. That's really really old. But if it works I guess that's fine :)

DickWiggler6, 2020-11-06:
Well what would change if I updated it?

keirf, 2020-11-06:
Nicer font XD I mean, if it works, why bother.

DickWiggler6, 2020-11-06:
Also what version would be current?

keirf, 2020-11-06:
https://github.com/keirf/FlashFloppy/wiki/Downloads

version 3.21

DickWiggler6, 2020-11-06:
Oh holy shit that's why you said there wasn't a version 0.11 my shit is ancient haha

keirf, 2020-11-07:
I don't think I have anything more to add here. Copying images across to a second USB stick is totally fine, and then you can format/overwrite the contents on your drum machine as you like.

closed 2020-11-07 by keirf