keirf/flashfloppy #387: Roland S-50 Boot Issue

jgale2005, 2020-11-18:
I have a question about integrating FF and a Gotek with my Roland S-50.  

When I boot the S-50 with the Gotek and FF – it takes about 25-30 seconds before the S-50 screen starts its “countdown” as it reads the disk.  While it is waiting those 25 seconds - the Gotek is counting from 1 to 7 very slowly.  The Gotek then starts counting from 8 – 79 faster as the S-50 screen starts its “countdown” loading process.

When I connect the original FDD – the delay time is only about 4-5 seconds before the S-50 begins its “countdown”. I am guessing the first pause is the OS load and the “countdown” is the loading of the sample data. 

The S-50 works as expected after it is booted and I select another file on the Gotek and have the S-50 either load or save a file. The “countdown” starts quickly. I am guessing that the S-50 is looking for some kind of handshaking during the OS load. 

I tried changing a bunch of FF options but nothing seemed to change this.  I replaced the S-50 boot file in case it was corrupted. No Luck.  Not a deal breaker – but any ideas? 

-	FF 3.21
-	Gotek SFRM72-FU-DL
-	Jumper on S0 only

[FF.CFG.TXT](https://github.com/keirf/FlashFloppy/files/5562356/FF.CFG.TXT)

John

keirf, 2020-11-19:
At least the early OS tracks on the disk probably have sector interleave and/or skew. What's really needed is a dump of the physical Roland floppy disk, eg using a Greaseweazle adapter and software (https://github.com/keirf/Greaseweazle) -- these are (fairly) cheap especially in kit form.

jgale2005, 2020-11-29:
Thanks for the response - I'm betting that you are correct.  

I've been testing and everything else works fine - patch loading, sample loading, file saving, disk conversion, etc.  The only issue is a one time delay of 25 seconds on initial boot.  I can live with it as is.

btw - thanks for creating this firmware.  The original Gotek firmware would not even show the display correctly. You've helped to breath life back into my S-50!

I've made a small donation to your site so that you can have a beer or coffee on me.

Thanks again,
John

keirf, 2020-12-02:
Cheers, I will ask around if anyone can do an OS disk dump for me.

closed 2022-06-22 by keirf