keirf/flashfloppy #388: Korg N364/(N264) Music Keyboard with Gotek FlashFloppy LED system reads files but has to turn off to access new image# 0000-0001

LeeMC2020, 2020-11-19:
When Setting up my Korg N364 keyboard I used a previous issues to help me solve how to get everything set up successfully.
-------------------------------------------------
Korg 01W FF.cfg to do and error e34 #385
-------------------------------------------------
==========================
My current (FF.CFG) Setup is as follows:  
==========================
interface = shugart
host = akai
pin02 = auto
pin34 = auto
write-protect = no
side-select-glitch-filter = 0
track-change = instant
write-drain = instant
index-suppression = yes
head-settle-ms = 12
motor-delay = ignore
chgrst = step
ejected-on-startup = no
image-on-startup = last
display-probe-ms = 3000
autoselect-file-secs = 2
autoselect-folder-secs = 2
folder-sort = always
sort-priority = folders
nav-mode = indexed
nav-loop = yes
twobutton-action = zero
rotary = full
indexed-prefix = "DSKA"
display-type = auto
oled-font = 6x13
oled-contrast = 143
display-order = default
display-off-secs = 60
display-on-activity = yes
display-scroll-rate = 200
display-scroll-pause = 2000
nav-scroll-rate = 80
nav-scroll-pause = 300
step-volume = 10
da-report-version = ""
extend-image = yes
==========================
All of my files follow the "DSKA0000_username.IMG" Naming convention with the file size of 720K
Below shows an example of what appears in my USB Drive: H  
I used a black image of 720K that you created for the other User and then formatted it in my Korg N364
Keyboard.  Then moved those file into the formatted images with the WINIMAGE software.
==========================
[image]
==========================
I can read and write to the image files just fine, everything works okay.  But when I try to select a different number (for example go from DSKA0000 to DSKA0001 by pressing the arrows of the gotek drive; the gotek drive shows 0001 but the Korg N364 only recognizes 0000 still until I turn off the Korg N364 and turn it back on then it recognizes the 0001(DSKA0001) Image.  I guess this would also be true if I took the USB out of the keyboard and placed it back in and then changed the number on the Gotek drive; it would then recognize the new numbered image 0001(DSKA0001) that I selected.  This may be that I've set it up wrong but I'm unsure.  I've tried native and I currently have it set up in (indexed mode), but have received the same results.  I have no subfolders as well.  Thanks for any help and all help you can provide in advance.  -Lee

keirf, 2020-11-19:
In FF.CFG try changing interface line to `interface=ibmpc`. Change it back if it doesn't help.

Another thing to try is change the chgrst line to `chgrst=delay-3`.

LeeMC2020, 2020-11-19:
I tried the (interface=ibmpc) first and I received the error message ("drive not ready") meaning that it didn't recognize any of my file information.  

Next, I tried the (chgrst = delay-3); and it did not work as well. 

I tried some more testing and research, and noticed that the only time it will recognize me selecting a new image file DSKA0000 to DSK0001, is if I turn the Korg N364 off and back on.  I was reviewing the manual and from what it states when a disk is inserted or changed it does some type of automatic load function and will display the message (("Directory" & "OK?")).  See Screen print below:

[image]

I can work around this by pressing the (Global) button, and then pressing the (Disk) button, to get the same messages to appear on my screen.  However when I try to read the new image from the next selection 0001(DSKA0001) the Korg N364 still only has 0000(DSKA000) information listed to read or load.  I'm wondering if this is within the Korg N364 Operating System? I removed the USB completely and still the same thing occurred.  I turn the Korg N364 off and then back on and it recognizes the image change from 0000(DSKA0000) to 0001(DSKA0001).

keirf, 2020-11-19:
It will be some physical interfacing thing, like disk change signal is not correctly connected or configured right. I see N364 has a 26-pin drive connector. Are you sure the adapter you must be using to connect to 34-pin Gotek is a correct one?

LeeMC2020, 2020-11-19:
Yes for the most part.  I ordered the wrong adaptor first from eBay.  It was a 26pin to 34pin 1.0 pitch; and it would not fit.  I ordered the correct one the 26pin to 34pin 1.25 pitch from Flexidrive products.  I'm guessing the disk change signal has to be configured in or on the gotek wiring or jumpers?  I have S0 currently.

keirf, 2020-11-19:
Disk Change is on pin 2 in Shugart interface configuration. Schematic indicates that should map to pin 21 at the 26-pin connector. Beyond that, there are variations in how the Disk Change signal can behave. But getting it mapped to the right pin is the first thing.

closed 2020-11-20 by LeeMC2020

penfold42, 2026-11-01:
Sorry for the necro thread, but it looks like you might need a pull-up on the diskchange pin.
i'll be making my own cable so have "done my own research" and saw a resistor on some premade cables.
Tracing this shows its a 2k2 pull up to 5v.
Looking at the N364 schematic shows a (stupid?) pull _down_ 10k resistor inside the synth.