keirf/flashfloppy #391: YAMAHA PSR 6700 emulator does not read fdd 720KB

futek2, 2020-11-25:
Hello keirf and reader.
I downloaded new software to gotek so that I replaced the **MF353C 152MZ** SHUGART type disk drives in the **YAMAHA PSR 6700** keyboard in 1990.
GOTEK with the latest version of the 3.22 firmware at the beginning I tested on my PC and everything is fine when the file images are for a **1.44MB** disk, but when I load images for **720KB**, it cannot be read.
I made changes to the configuration in FF.CFG files to install GOTEK with 3.22 soft and there is an effect that when there are 1.44MB images, it is signaled that the station is seen but it gives a message from UNFORMAT in YAMAHA and they should be 720kB because this is the station.
The LED on the keyboard will turn on and the green LED on the GOTEK will go out.
But when I do the image exchange from 1.44MB to 720kB, both LEDs are off and the GOTEK doesn't work anymore.
I did tests on earlier versions of the software and the problem is similar.
Same thing happens on my pc as i test.
For the YAMAHY PSR 6700 I have the following settings in FF.CFG:
_interface = shugart
pin02 = dens
pin34 = chg
chgrst = pa14_

For my PC:
_interface = ibmpc
pin02 = auto
pin34 = auto
chgrst = step_

I read the settings for YAMAHA from the forum
My observation that when I was doing tests with images of smaller disks, there is a problem.

Please help .
Best regards.
Futek2

keirf, 2020-11-25:
So 1M44 works perfectly including disk changes, while 720k doesn't work at all, on the PSR?

keirf, 2020-11-25:
Okay that can't be it as I believe PSR-6700 is double density only. It may also use a custom disk format.

Perhaps try the linked HFE image file (see below) on your Gotek, and select FORMAT DISK on the PSR 6700. If that works, send me the HFE to have a look at (or you can just go ahead and use HFE image files).

In short, PSR may not use FAT-fromatted 720k disks. I don't know.

https://github.com/keirf/FF_Images/raw/master/Amiga/Blank_AmigaDOS.hfe

futek2, 2020-11-25:
Hello Keirf and reader.
When I have a 1.44MB disk image on the flash drive, the YAMAHA PSR6700 gives me the message UNFORMAT disk. When I choose the format, I should do it from 80 cylinder to 0 and it will end but I do not have any trace on the flash drive.
I take the pnedraiwa and put it into my PC and also nothing changes and my PC reads the pedrai and it's ok nothing changes And when I put it into GOTEK and change FF.CFG to my PC, it still sees that it reads 1.44MB disk image correctly.
I will try with what you wrote to me and let me know.
Futek2

keirf, 2020-11-25:
The PSR hasn't managed to modify the 1M44 image at all. Not surprising if it doesn't support high density images at all.

futek2, 2020-11-25:
Here is a description of the Yamaha Clavinova CVP-75 has the same MF353C-152MZ hard drive.
Here is a description and maybe that will be helpful.
https://torlus.com/floppy/forum/viewtopic.php?p=12659#p12659

How do you propose to make rehearsals?
Futek2

keirf, 2020-11-25:
It's just a double density floppy drive. You probnably have the interface set up right. You just need to have suitable formatted disk images. I think PSR does that itself if you use HFE image and format on the PSR.

futek2, 2020-11-25:
And here they made rehearsals and achieved success.
https://torlus.com/floppy/forum/viewtopic.php?p=20911#p20911

keirf, 2020-11-25:
That is a different host (Clavinova). Can be many differences even if drive is the same.

futek2, 2020-11-25:
I did another test and the YAMAHA PSR6700 does not see a 720KB picture. And if I connect GOTEK to my PC and gives the same 720KB image, it doesn't read either - no disk.
And if I change the disk image to 1.44MB, that's fine.
Maybe I have what to change in the FF.CFG file?
Because it seems that maybe the reading time of a smaller file is shorter and this is the reason that these 720kB images are not read.
Because I did a probe from a PC with a 1.2MB image and it is already reading.
Futek2

futek2, 2020-11-25:
I did tests with subsequent versions of the software, older and the FF_Gotek-v1.2.upd version uploaded to GOTEK will allow me to read 720kB rotor on PC. But it took longer to read than how it reads 1.44MB.
On the YAMAHA PSR6700, I do not read 720kB image, even there is no LED signaling.
And as the image is 1.44MB, after turning on YAMAHA, the plug will be green LED on GOTEK and a moment later LED red on YAMAHA (it looks like simultaneously).
There is also another information that as the pen tray with a 1.44MB disk image for the GOTEK connected to YAMAHA which are turned on, it will also appear after a while in the world of GOTEK LED and YAMAHA. It did not work when the image on the flash drive is 720kB.

futek2, 2020-11-25:
I recorded the disk image in HEF format and I will put it on pendrives and my PC read it as 720kB and it sees everything.
Now I will do the same test on the YAMAHA PSR6700.
Software version in GOTEK FF_Gotek-v3.22.upd.
I'll let you know.
Futek2

futek2, 2020-11-25:
I made 4 attempts:
1- test: I made a 720kB disk image, 2059kb in size in HEF format and transferred to a 720kB disk image on a flash drive. and connected to YAMAHA and it does not recognize and does not read, and there is no LED signaling.

2 - test: I made a 1.44M disk image 40117kb in HEF format and transferred to a 1.44MB disk image on a flash drive. and connected to YAMAHA and it recognizes it but does not read it. UNFORMAT
There is LED signaling in YAMAHA and GOTEK.

3 - test: I made a format probe for this 1.44MB disk image in the form of HEF from YAMAHA and managed to format everything and even saved a 17kB file.

4 - test: I made a test reading from YAMAHA and everything was successfully read and the data was correctly sent to YAMAHA and the message is complete.

This is a small success, but I tried to read what was written on the flash drive with HxC Floppy Emulator and the message "Some Errors found during the Image mounting! DOS Disk? Disk damaged?"

Now I'm looking for what kind of disk format on YAMAHE.

Is this good news?

Futek2

keirf, 2020-11-25:
I think it tells us that the disk format is not PC compatible. So I don't know that you will be able to easily transfer files via floppy between PC and your Yamaha. Perhaps some other method such as MIDI could work?

Can you attach here the HFE image that you successfully formatted in the Yamaha? I can see what the track format looks like.

futek2, 2020-11-25:
I will send you an email.

futek2, 2020-12-03:
Hello Keirf.
After the last one I changed what you advised me, everything is fine.
So I do not change anything anymore because there is no need to work well.
And with a different format, I don't mind because it only applies to the YAMAHA PSR6700.
Thank you for your help.
Best regards.
Futek2

closed 2020-12-03 by keirf