keirf/flashfloppy #393: Micropolis support and hard sectors

ejona86, 2020-11-29:
I have a Vector Graphic Inc. Vector 4 system that uses a Micropolis-compatible floppy drive. The Micropolis format is straight-forward, but it uses hard sectors. I'm interested in adding support. It looks like hard sectors will be the main issue.

I assume there's no way to get this halfway working just by configuration. I don't care about the on-disk format too much.

FWIW, I already added read support to FluxEngine (which can consume SCP files from Greaseweazle). I have [a flux image captured with FluxEngine and converted to SCP](https://github.com/keirf/FlashFloppy/files/5611751/micropolis-pattern.zip) of a test disk that has a collection of patterns on 7 tracks (ignore later tracks; the floppy was known-bad). Sample real-life image files for a Vector Graphic system can be found at [deramp.com](https://deramp.com/downloads/vector_graphic/software/disk_images/Micropolis%20controller/). [Detailed hardware manuals are available](https://github.com/davidgiven/fluxengine/blob/master/doc/disk-micropolis.md#useful-references).

It looks like the FlashFloppy IMG support would be a pain to adapt without benefit, so it would be convenient to use the different Micropolis-specific VGI format used by some emulators. That would make configuration easy and allows for some stranger hardware variations. But if it makes sense to enhance IMG or DSK support, then that would be fine too. The VGI format is "raw" like an IMG format, but with the full sector instead of just user data, as that is how the software communicated with the Micropolis FDD controller.

Basic details of the VGI format (as if it were an IMG):
cyls = 77
heads = 1-2
file-layout = sequential
secs = 16 (hard sectors)
bps = 275 for VGI (256 user data)
id = 0
mode = mfm
interleave = 1
cskew = 0
hskew = 0
rpm = 300

The sector format is MFM-encoded: 0x00 preamble, a 0xFF sync byte, 274 bytes of data, and then 0x00 postamble. There's a simple checksum and an optional yet-to-be-understood ECC.

Looking at `src/image/adf.c`, it seems like it would be straight-forward to add support, except for the hard sectors. How invasive would be to support hard sectors? If a pain, initially I could emulate them in hardware with a [Virtual Sector Generator](https://deramp.com/vsg.html) along with index-suppression=no. But if some other plans of mine pan out, I'd probably try to get direct support in FlashFloppy.

keirf, 2020-11-29:
Concentrating just on read to start with, you can get roughly the right effect by setting `im->tracklen_bc` to 1/16th of your actual track length. This will get you 16 equally spaced index marks, with the first at the "track start".

However I guess you also want a double pulse at the track start. You could possibly fake this by mucking with `tracklen_bc` dynamically. Or you might be better to introduce the concept of a `hardsec_bc` and `hardsec_ticks` and plumb this through to the index magic at the tail end of `IRQ_rdata_dma`.

Writes will need you potentially to mess with or add to the `im->stk_per_rev` logic as this free-runs index pulses when we're not syncing with the RDATA stream.

Also there will be questions over whether index pulses should ever be suppressed: FlashFloppy does that by default while busy (for example seeking to a new track, or when writing out data). It can be turned off to some extent in configuration. Perhaps that would need messing about with some more.

ejona86, 2020-12-09:
I've gotten reading implemented, with hard sector index pulses. It was rockier than I expected, partly due to my unfamiliarity with the code base/architecture and partly due to some bugs in my code that were slow to debug. Most recently, `image_ticks_since_index()` and `image->tracklen_ticks` were different units. But now that it's implemented, it looks quite simple.

I'm cleaning things up and then I'll create a PR.

keirf, 2020-12-10:
Yeah the ticks thing, that's a pain. I should give them different names but, how many names are there for abstract ticks?

keirf, 2020-12-13:
Worth a look at #400 too: That's another way to support hard sectors (and potentially a wider range of hard sector formats).

ejona86, 2020-12-15:
It's not clear to me part of #400 you are referencing. Do you mean supporting OP_index of HFE would be a good way to support hard sectors?

Information for the various hard sector formats is pretty scattered. But from what I can tell they are all basically the same, except for number of sectors. [StackOverflow](https://retrocomputing.stackexchange.com/questions/2484/what-systems-used-hard-sectored-floppy-disks/2486#2486) and FluxEngine list hard-sector floppies, so I looked through those. I didn't come to conclusions about DEC Pro 350, MITS Altair 8800 BT, and AES Superplus; I probably found enough information for the first two, but it'd take more time to wade through. The Heath H17 (and thus probably the H-89A), NorthStar Horizon, Helios II, Compudata Exidy Sorcerer, and probably the Zilog MCZ use same hard sectoring as Micropolis. The IBM Displaywriter appears to be soft-sectored, but with a different index sleeve hole based on the format; maybe that caused it to be miscategorized. In other searches I saw one mention of an 8" hard sector format which put the index hole in a different part of the disk (further or closer to the center of the disk), but it seems that would require a different electrical interface.

So I went with the nr_hardsecs approach, because it seemed it would be pretty complete. I think it wouldn't be too hard to extend that to be a "list of index locations", relative to bc. HFE could then fill that array in progressively within `hfe_rdata_flux`. (I imagine setting a `hfe.hardsec_pos = 0` at the beginning of each track and setting the cur_bc value to the current array position when OP_index is seen, incrementing hardsec_pos. It seems the HFE code only supports starting at the beginning of the track.) But for now I imagine HFE could be handled by just assuming the number of sectors based on the initial index timing.

keirf, 2020-12-15:
Yes arbitrary index positioning with OP_index. It seems an interesting approach. But the VGI support is not that invasive.

The thing yet to be tested in practice is how the hard-sectored host handles FlashFloppy's irregular track timing -- when seeking between tracks, pulses may be lost; when writing back data to Flash, pulses may be lost. This may break the host assumption that pulses are produced in good time with no breaks all the while the disk spins. I think other dedicated HS projects include enough RAM to hold an entire disk image, which makes it rather easier/possible to achieve hard real time requirements.

What I mean is: The proof of the pudding, and much of the actual difficulty, will be in getting a real VG system to play nicely with FlashFloppy. Consider the mess that writes can make if the host gets out of sync on which sector is currently being played out.

ejona86, 2020-12-24:
> The thing yet to be tested in practice is how the hard-sectored host handles FlashFloppy's irregular track timing -- when seeking between tracks, pulses may be lost; when writing back data to Flash, pulses may be lost

I definitely think the approach should be "don't lose pulses." Pulses seem more important than getting the data right. If we keep pulses working, then I'd expect OS-level retries to paper over data issues.

I was digging into pulse irregularity I was seeing even though index suppression was disabled. `fake_fired` just looks wrong, so I disabled that. I started realizing that changing `index.timer` for hard sectors prevents `index.prev_time` from aligning with the track start which causes confusion. I also saw there was already free-running code for index pulses, which I had been considering adding for sector pulses.

I thus tried a new approach: leave `index.timer` alone and add a new timer for the other pulses. You can see the results in my [vgi-index-list branch](https://github.com/keirf/FlashFloppy/compare/master...ejona86:vgi-index-list?expand=1) (floppy.c and floppy_generic.c are the only interesting parts). The extra pulses are stored in a list. I have `index_assert()` schedule the timer for the other pulses, and then those other pulses free-run until the end of the track.

I had an earlier approach where I tried splitting the index pulse and the sector pulse timers, as would be done in old hardware. In this case the index pulse comes a half-sector before the actual start of the track, and each timer fires at a consistent interval. I introduced an `index.offset` to reposition the index pulse and the consistent timer interval made free-running easy. The results can be seen in my [vgi-sector-pulse branch](https://github.com/keirf/FlashFloppy/compare/master...ejona86:vgi-sector-pulse?expand=1). It didn't turn out super poorly, but is more error-prone and I didn't figure out how I wanted to stop sector free-running (so it free-runs all the time).

-----

Reads don't seem to be a problem. Between tracks the 15 ms head settle time seems plenty. For SS, the current 10 ms timer seems like enough and may impact only one sector. I expect we could add an extra sector worth of delay there and not cause trouble. If something goes wrong the OS will retry which provides an extra 200 ms.

Writes will be interesting. There's no naturally-occurring hardware delay at the end of the write, other than <1ms for the write head to deactivate. But generally the writer should do a verification read after the write. The version of CP/M I have only allows an application to write 2k at a time (sixteen 128 byte "sectors"), but the Micropolis track size is 4k. That means there is likely at least 100 ms of slack after the write until the verification read. FORMAT and maybe DISKCOPY may behave specially and cause trouble; it seemed FORMAT wrote the entire disk and then read-verifies the entire disk. My machine also supports MS-DOS 2.0, and I don't know its behavior.

keirf, 2020-12-24:
100ms will be enough most of the time. Also I am generally planning a move to write-back dirty buffers when heads are next stepped. This will hide write latency behind seek/settle/setup latencies. Not perfectly, but it's probably a more forgiving time to insert somewhat arbitrary delays.

ejona86, 2020-12-24:
> Also I am generally planning a move to write-back dirty buffers when heads are next stepped.

That's what I was originally expecting before digging into the code. I expected the full track to be in memory and shared between read/write and then on track change write it out (potentially optimized to track dirty parts). That seems especially useful for sector skew.

Although I hadn't considered head changes. Head changes would double the necessary memory, which seems fine. It may be pushing things too much to write out both head's tracks (which may be non-contiguous in the file) and read in 1-2 tracks within 15 ms. But I agree that every OS is probably most resilient to unexpected delays when changing tracks.

ejona86, 2020-12-28:
I just added HFEv3 hard sector support in my [vgi-index-list branch](https://github.com/keirf/FlashFloppy/compare/master...ejona86:vgi-index-list?expand=1). I've been unable to convert recordings to HFEv3 and retain the sector pulses, so I've only tested reading against [heath_kf.hfe and northstar_nst.hfe](https://hxc2001.com/vrac/HS/). Once I'm able to verify write support I can make a PR.

If I get HFE working, then I guess I'll delete the VGI code in my working branch. There are issues with HFE as the tracks aren't aligned well, but I hope that's not much of a problem for synthetic files (vs recordings).

ejona86, 2021-05-08:
Closing in favor of tracking with #400

closed 2021-05-08 by ejona86