keirf/flashfloppy #396: Add OLED Display Support 64 * 128 SH1107 (SSD 1107)

George-CNC, 2020-12-07:
Hey.
There is an excellent display, more compact and convenient. 0.96 OLED 64x128 SH1107 (SSD 1107)
But unfortunately they do not work even on the latest firmware 3.22.
I beg you to add support for this display in the new firmware.
Thanks!

George-CNC, 2020-12-07:
SH1107 64X128

Controller "sh1107", Display "64x128" [Description]

    U8G2_SH1107_64X128_1_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_4W_SW_SPI(rotation, clock, data, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_4W_HW_SPI(rotation, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_2ND_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_2ND_4W_HW_SPI(rotation, cs, dc [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_2ND_4W_HW_SPI(rotation, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_3W_SW_SPI(rotation, clock, data, cs [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_3W_SW_SPI(rotation, clock, data, cs [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_3W_SW_SPI(rotation, clock, data, cs [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_3W_HW_SPI(rotation, cs [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_3W_HW_SPI(rotation, cs [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_3W_HW_SPI(rotation, cs [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_6800(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_6800(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_6800(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_8080(rotation, d0, d1, d2, d3, d4, d5, d6, d7, enable, cs, dc [, reset]) [full framebuffer, size = 1024 bytes]

SH1107 64X128

Controller "sh1107", Display "64x128" [Description]

    U8G2_SH1107_64X128_1_SW_I2C(rotation, clock, data [, reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_SW_I2C(rotation, clock, data [, reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_SW_I2C(rotation, clock, data [, reset]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_HW_I2C(rotation, [reset [, clock, data]]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_HW_I2C(rotation, [reset [, clock, data]]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_HW_I2C(rotation, [reset [, clock, data]]) [full framebuffer, size = 1024 bytes]
    U8G2_SH1107_64X128_1_2ND_HW_I2C(rotation, [reset]) [page buffer, size = 64 bytes]
    U8G2_SH1107_64X128_2_2ND_HW_I2C(rotation, [reset]) [page buffer, size = 128 bytes]
    U8G2_SH1107_64X128_F_2ND_HW_I2C(rotation, [reset]) [full framebuffer, size = 1024 bytes]

George-CNC, 2020-12-08:
Is it possible to add some library code to the FF.KFG file?

keirf, 2020-12-08:
This is not high on my todo as the display is addressed horizontally rather than vertically (so all the fonts have to be rotated) and the existing 128x64 displays do fit if the panel is unstuck from the pcb, and the pcb laid flat behind the panel, inside the Gotek enclosure.

George-CNC, 2020-12-08:
> 
> 
> This is not high on my todo as the display is addressed horizontally rather than vertically (so all the fonts have to be rotated) and the existing 128x64 displays do fit if the panel is unstuck from the pcb, and the pcb laid flat behind the panel, inside the Gotek enclosure.

Thanks for the answer.
This screen won't work?

keirf, 2020-12-08:
The display output is rotated 90 degrees: It is a 64x128 display rather than 128x64. Hence the work needed to support it is non-trivial.

George-CNC, 2020-12-08:
спасибо за ответ!

closed 2022-06-22 by keirf

philippetev, 2024-08-18:
Sorry to bring back this old one, but has any progress been made on this issue?