keirf/flashfloppy #397: Disk offline

ducdreamer, 2020-12-07:
Hello Everybody,
I got myself A Gotek SFRM72-TU100K fd emulator  for my msx computer.
the default fw works but is terrible
So I have succesully flashed and verifyd it with the latest Flashfloppy FW
 When the computer boots its display FF ( I got A two digit screen.
 And when I use a fat32 formatted usb stick containing FF.CFG
I get E3 error
In the cfg file 
Interface = shugard
Host = msx
on the board I have S0 and JC jumpered.
As it has only A two digit screen the information is not that clear.
But the FW is working.
When I hold the two buttons it shows r S
display gets dark and comes up with FF
So what can be the problem.
A fake Gotek?
I hope somebody can give me A hint
Greetings Herman

keirf, 2020-12-07:
Probably error is E34 and you have no valid image files on your USB drive.

ducdreamer, 2020-12-07:
Hello Keirf,
thanks for your quick answer
there are nofother files on de drive.
only FF.CFG
Am i missing some files?
When I put the USB back in the computer there is also a new empty file on de USB
IMAGE_A.CFG
But when I do format the result is the same disk offline

keirf, 2020-12-07:
You need to some IMG files suitable for MSX on your USB drive.

ducdreamer, 2020-12-07:
Hello keirf,
I have put a MSX dsk file on the USB and it works.
In the original FW you must initialize the USB by holding the two buttons and format the individual disks.
Then I wrote a liddle basic progam and save it to the USB.
I could read it back and run it
Here you just put the image files on the USB and it works.
Great.
But is it possible to save your own programs and read them back?
that would be great
Thanks for your great work
Greetings Herman

keirf, 2020-12-07:
You can save onto the currently mounted/attached IMG file on your USB drive, yes.

ducdreamer, 2020-12-07:
Hallo ,
Thats great
I just create a empty img and save the programs on that image.
Greetings Herman

closed 2020-12-09 by keirf