keirf/flashfloppy #398: Kaypro DSDD 'Raw' image not working

snhirsch, 2020-12-08:
From comments in the sample cfg files I received the impression that raw sector images for Kaypro are supported directly.  However, I'm not able to get this working.  They seem to be recognized by the FF firmware, but cannot be read by the computer.  If I convert them to IMD format and use HxC in batch mode, the resulting HFE file does work properly.  What am I missing?

keirf, 2020-12-08:
Please attach a working HFE here, and also preferably a corresponding raw sector file (IMG) too.

snhirsch, 2020-12-08:
Sure thing.  Here 'tis.  The HFE file was created from the original IMD using 'batch convert' in the HxC application.  It works correctly under latest release F-F.  The sector image was extracted from the same IMD using IMDU.  The F-F appears to mount it without complaint, but the computer cannot read the emulation.

[images.zip](https://github.com/keirf/FlashFloppy/files/5662315/images.zip)

keirf, 2020-12-08:
What suffix do you give the raw image file on the FlashFloppy USB stick? It should be .IMG (case insensitive).

keirf, 2020-12-08:
Also, are you specifying `host = kaypro` in FF.CFG?

snhirsch, 2020-12-08:
I tried both .IMG and .DSK.  And, yes, I had 'host = kaypro' in FF.CFG.  F-F seemed happy enough with selecting the image.  I also tried tried <filename>.kaypro-dsdd40.{img|dsk} to be absolutely sure the firmware understood what the file was.  No joy.

keirf, 2020-12-09:
Ok, I tried the IMG/BIN you provided on latest FlashFloppy and dumped it to HFE using Greaseweazle. I attach the HFE here. Can you give it a try? I can't actually see anything wrong with the track layout. I note that your HFE does have a single sector on both sides of cylinders 40 and 41 (ie outside range of a normal 40-cyl disk) -- does the Kaypro ever access them? (ie. does the cylinder count ever increase to 40 or 41?)

[dump.zip](https://github.com/keirf/FlashFloppy/files/5664577/dump.zip)

keirf, 2020-12-09:
Phew, and here's some more: The track layout for disks can be defined in an `IMG.CFG` file, and this gives us more flexibility than `host=kaypro` as we can tweak the parameters. I have attached here an IMG.CFG that you can try putting in the root of your USB drive. It will override the layout for 200kB/400kB/800kB IMG files. So you can try this too. That gives plenty of avenues to explore now!

[IMG.CFG.zip](https://github.com/keirf/FlashFloppy/files/5664727/IMG.CFG.zip)

snhirsch, 2020-12-09:
I placed that IMG.CFG file in the root directory of the USB stick, alongside a bunch of HFE files and a single IMG file of the Kaypro diskette.  Unfortunately, although the HFE files work fine, the computer is still unable to read the IMG.  I can see it selected, then a step from track 0 --> 1.  It hangs for a moment and complains 'BDOS error on C, bad sector'.  What am I missing?

keirf, 2020-12-09:
Did the dumped HFE that I provided work?

snhirsch, 2020-12-09:
Sorry - no, it does not work.  As a control, I went back to the HFE that was batch-created from IMD (product of using td02imd on original Teledisk image).  It works fine.

keirf, 2020-12-09:
Ah I see the head number is 0 on both sides of the disk. That needs a tweak in IMG.CFG and also means that `host=kaypro` never worked properly (I'll remove it in favour of providing an IMG.CFG in the next release). So please try the following IMG.CFG:

[IMG.CFG.zip](https://github.com/keirf/FlashFloppy/files/5667303/IMG.CFG.zip)

snhirsch, 2020-12-09:
Bingo!  That indeed works :-).  What an odd arrangement with the head numbering. FYI, a real K10 floppy has interleave of 4 (e.g. 0,5,3,6,1,9,4,7,2,5 for side 0 and 10,18,13,16,11,19,14,17,12,15 for side 1).  Not sure this makes a huge difference, though.  It may actually be slightly faster on reads with interleave of 3.

I'm a happy camper now.  Really appreciate the fast response on this issue.

It's quite possible that previous Kaypro F-F users were working with single-sided images.  Might explain why no one had called this out.

keirf, 2020-12-09:
Ok that's great, I'll get this sorted for next release. Regarding interleave, performance increases as you reduce it until it suddenly drops off a cliff. So if it's okay at 3:1, that's good. The gap between 4-5 (14-15 on side 2) is only three anyway!

closed 2020-12-09 by snhirsch