keirf/flashfloppy #403: ATARI 1040ST Write problems

ASTuser1, 2020-12-19:
Hi,

i´ve recently equipped my 1040ST (3 MB RAM, TOS 2.06) with a SFR1M44-U100K and programmed it with the latest V3.22 Release. I am using a Maxell 8 GB Stick.

Reading is perfect, but iI have strange write issues, my .img-Files got corrupted when i´m trying to write to them. 

The following images show a comparison between the original and the copy, which i made with the TOS Copy Function. Top Windows shows the Copy i´ve made, Bottom Windows shows the original.

[image]

[image]

[image]

As it seems, each 512 Bytes and the following Byte, as well as some bytes in between contain invalid data. The Format is the standard 80 Track/9 Sectors, 1024 Bytes per Cluster, 720K

While copying, reading of the 720K-Image is done in about 50 Seconds, writing around 6:30 Minutes(!).

I´ve played around with numerous parameters in FF.CFG and also in IMG.CFG, but unfortunaly that doesn´t solve the issue. 

CrystalDiskMark gives me the following results fpr the USB Stick:

[image]

I have attached three files - one contains the orginal .st Image, the second one is a Copy made with the Standard TOS-Copy-Routine, the third one is the copied and afterwards formatted image. 

I´m not sure if this can be nailed down to the Stick-Performance.

i would be very happy about a solution ;-)

Best regards !

[images.zip](https://github.com/keirf/FlashFloppy/files/5718784/images.zip)

keirf, 2020-12-19:
Try another USB stick? ;)

ASTuser1, 2020-12-19:
I´ll try to get hold of another one, since this one is the only one that i have. Any Sticks recommended?

keirf, 2020-12-19:
Doesn't need to be anything special or particularly large. A cheap branded 16gb stick for example.

ASTuser1, 2020-12-21:
Nope, even a 128 GB Stick with approx 18 MB/Sec Sequential and 2 MB/sec Random write speed and doesn´t solve the issus.  The first two Bytes of some 512 Byte-Data Blocks still have not the correct Value and Random Bytes instead of Zeros also still apperar.

ASTuser1, 2020-12-21:
In addition - yes, some of the Sticks of the Brand that I used before have indeed serious speed issues when writing 4 KB-Blocks, The Stick itself isn´t very old, approx one and a half year. Maybe the Type of Flash being used could be a issue (zeoring out a Page which takes a specific amount of time, then write actual data)

keirf, 2020-12-21:
What Flashfloppy version? If very recent could be worth trying an older version just in case there's been a regression.

Also worth running the logfile version of the firmware and gathering a FFLOG.TXT file. Perhaps we'll see CRC errors.

keirf, 2020-12-21:
Another thing could be worth trying is writing to an HFE image file and looking for errors in that.

ASTuser1, 2020-12-21:
I´ve used the actual V3.22, as well as a V0.9X from 2018 - after i found a thread from a guy which also used FF with a 1040ST, but he had a speed issue. 

The only thing i can report by now is that i have around 304 RPMs instead of 300 (measured last night with a program written for the ST). But i´m not sure how reliable this program is.
 
I´ll try later to adjust the RPM-Parameter in the cfg and if this doesn´t help, i´ll try the V3.16 and if it doesn´t still work, the Log-Version. I also have an Saleae-Clone available, if further measurements are needed.

Thanks for your effort - i appreciate it !

ASTuser1, 2020-12-22:
Updated Status - V3.16 .hfe same problem, but i can at least report a little progress.

First of all, V3.22 log output:

`Build: Nov 17 2020 14:51:44
Board: Standard
Config: Flash Slot 6 (ver 2, size 69)
Interface: Shugart (pin2=chg, pin34=rdy)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 0101495ac5bddd01ef7c30a6535f44c32be7e4348e749ba33af0581321199af
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 104 items
0:F: 'mid.IMA' 
Cache 104 items
Mode: Native
Current slot: 1/1
Name: 'mid' Type: ima
Attr: 20 Clus: 0004001d Size: 737280
Fast Seek: 1 frags
Cache 33 items
Cache 104 items
Before: After: Current slot: 0/1
Name: 'Atari Magazin 003 (19xx)(Atari Magazin)_st' Type: img
Attr: 20 Clus: 00000003 Size: 737280
Fast Seek: 2 frags
Cache 33 items
Cache 104 items
Before: After: Current slot: 1/1
Name: 'mid' Type: ima
Attr: 20 Clus: 0004001d Size: 737280
`

So i went from img and .hfe-images to .st-Images. Works better, but not under each circumstances.

Copying a .st-Image to a .st-Image by using the Builtin TOS Copy program still gives errors.

By using the program "Fastcopy III" ,  .hfe to img. and vice versa also leads into errors, namely "RNF/Seek error" on certain Tracks.

Now, with .st-images, (both source and the destination Image) its the following situation.

Assume that i have set up Fastcopy to a step rate of 12 ms both for reading and writing.

- Disk is being read in without problems.
- after that, the Gotek-Counter shows Track 79
- Fastcopy wants the destination disk, so i switch the image
- Gotek-Counter rolls back towards first Track

But then this happens:  

- the Gotek Track counter display does not reach Track 1, it stops somewhere else (anywhere between 2 and 79) 
- Fastcopy already starts with the writing process, then complains about a Write error with the usual RNF/Seek error.
- a Retry will set the Track counter from the last (error) position finally to one 
- and the copy is being made without any errors. The copy is 1:1 identical to the original .

Setting the Step rate to 3 ms (FastCopy´s default value) results in no errors at all for the entire process. Which doesn´t make much sense for me, because i would assume that a slower step Rate, instead of a faster step rate, should -avoid- problems. 

Now, i´ve read that TOS 2.06 uses by default a step rate of 6ms, but trying that step rate in Fastcopy even results in no errors at all. 

Remember, all this information is only valid for .st-images under Fastcopy, where HFE and IMG still lead to RNF/Seek errors on certain (random) tracks.

So, in conclusion:

- Still not able to copy Disks and unable to write error-free Files directly under TOS, Image-Format does not matter at all
- HFE and IMG-Files cannot be written without errors on the ST, whether though TOS, nor through Fastcopy
- ST-Images can be copied through Fastcopy with a Step Rate below 12 ms.
- its said that TOS 2.06 uses a Step rate of 6 ms, but i cannot tell for 100% sure if this is really true
- it -might- have to do something with the Speed of the track change

Reading a disk is fine, FC Stops at Track 79. tells me that i should insert the destination disk, so i switch the image. Track counter of the Gotek is still at Track 79. 

So i tell FC to proceed, then the Gotek-Track counter rolls back, stops at Track 26 and Fastcopy already wants to write, complains that a write error has occured. Repeat? Yes, Counter finally rolls back to Zero.

ASTuser1, 2020-12-23:
Update: After using a unmodified ff.cfg, , HFE-Files can be also written within Fastcopy III at least with 3ms Step Rate.

Here is the current log (source is the .st-image in Slot 0, destination the HFE-Image in Slot 1, zz.img contains Fastcopy III on Slot 2.
 
First step done was copying with Fastcopy, second try through TOS
`
[lost 51879]
 (4096)... 16255 us
Write 20992-25087 (4096)... 16285 us
Write 0-4095 (4096)... 12115 us
Write 2560-6655 (4096)... 23867 us
Write 5120-9215 (4096)... 24321 us
Write 8192-12287 (4096)... 12556 us
Write 10752-14847 (4096)... 16323 us
Write 13312-17407 (4096)... 19795 us
Write 20992-25087 (4096)... 15247 us
Write 0-4095 (4096)... 12579 us
Write 2560-6655 (4096)... 16331 us
Write 5120-9215 (4096)... 24640 us
Write 8192-12287 (4096)... 12193 us
Write 10752-14847 (4096)... 15423 us
Write 13312-17407 (4096)... 15899 us
Write 15872-19967 (4096)... 15433 us
Write 18432-22527 (4096)... 15761 us
Write 2560-6655 (4096)... 15465 us
Write 5120-9215 (4096)... 16274 us
Write 8192-12287 (4096)... 15976 us
Write 10752-14847 (4096)... 15420 us
Write 13312-17407 (4096)... 25087 us
Write 15872-19967 (4096)... 12161 us
Write 18432-22527 (4096)... 16305 us
Write 20992-25087 (4096)... 16280 us
Write 0-4095 (4096)... 15835 us
Write 8192-12287 (4096)... 15252 us
Write 10752-14847 (4096)... 15434 us
Write 13312-17407 (4096)... 24129 us
Write 15872-19967 (4096)... 12573 us
Write 18432-22527 (4096)... 16251 us
Write 20992-25087 (4096)... 16325 us
Write 0-4095 (4096)... 15751 us
Write 2560-6655 (4096)... 15886 us
Write 5120-9215 (4096)... 16481 us
Write 0-4095 (4096)... 19401 us
> USBH_USR_DeviceDisconnected
> USBH_USR_DeInit
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5583
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra Fit
 Serial Number : 0101495ac5bddd01ef7c30a6535f44c32be7e4348e749ba33af0581321199af
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 104 items
0:F: 'Atari Magazin 003 (19xx)(Atari Magazin)_st.hfe' 
Cache 104 items
Mode: Native
Current slot: 1/2
Name: 'Atari Magazin 003 (19xx)(Atari Magazin)_st' Type: hfe
Attr: 20 Clus: 00000039 Size: 2108416
Fast Seek: 1 frags
Cache 28 items
[... 128 lines cut ...]
Write 18432-22527 (4096)... 15876 us
Write 2560-6655 (4096)... 15251 us
Write 5120-9215 (4096)... 19152 us
Write 8192-12287 (4096)... 16355 us
Write 10752-14847 (4096)... 18806 us
Write 13312-17407 (4096)... 25221 us
Write 15872-19967 (4096)... 12181 us
Write 18432-22527 (4096)... 15905 us
Write 20992-25087 (4096)... 15881 us
Write 0-4095 (4096)... 15766 us
Write 8192-12287 (4096)... 14917 us
Write 10752-14847 (4096)... 18716 us
Write 13312-17407 (4096)... 25147 us
Write 15872-19967 (4096)... 12266 us
Write 18432-22527 (4096)... 16312 us
Write 20992-25087 (4096)... 16311 us
Write 0-4095 (4096)... 15869 us
Write 2560-6655 (4096)... 15865 us
Write 5120-9215 (4096)... 16299 us
Write 0-4095 (4096)... 15787 us
Cache 104 items
Before: After: Current slot: 0/2
Name: 'Atari Magazin 003 (19xx)(Atari Magazin)' Type: st
Attr: 20 Clus: 00000022 Size: 737280
`

keirf, 2020-12-23:
I wonder why it would stop early seeking to track zero. Perhaps the track 0 signal line is bad?

ASTuser1, 2020-12-23:
Very good idea - I will check that. 

In case that this wouldn´t sort out the problem, i have a (used) spare WD1772, and if everything else fails, ´i´ll connect my Saleae. 

Many thanks for your effort !

ASTuser1, 2020-12-23:
So here we go.  TRK00 is fine.

After changing for testing purposes to HxC the same error occurs.

Now i was trying to set the Step rate under TOS manually with the help of a Tool and, guess what, when i´m trying to test out the selected step rate inside this tool, the same corruption as usual occurs  . I have to verify it, but I think that this tool doesn´t switch the Write gate at all,

ASTuser1, 2020-12-24:
Well,  i hooked hooked up the Analyzer, used TOS-Copy.WDATA always starts with a series of pulses which could lead into 0xFF written twice (or 0x3F and then 0x00) at the beginning of each Sector.

Using HxC:

Full View (reading, then writing a few Tracks, Disk (in order to write) was once not recognized, i had to repeat it)

[image]

Particular view of a sector being written:
 
[image]

Full Track:
[image]
 
Now i will do the same with Fastopy instead of TOS-Copy.

ASTuser1, 2020-12-24:
With Fastcopy:

[image]
[image]
[image]

keirf, 2020-12-24:
Looks okay. I know it's tricky on ST but I urge you to try another floppy ribbon. I understand you may have to desolder the existing cable and fit a pin header. Old ribbons are a scourge on reliability, and you really need to be able to discount it.

keirf, 2020-12-24:
Since this is reproducible on other drives this is not a FlashFloppy issue.

closed 2020-12-24 by keirf