keirf/flashfloppy #405: Ensoniq EPS16+ write to HFE (superdisk) failures

JC-III, 2020-12-21:
I recently installed a gotek running flashfloppy in my Ensoniq EPS16+. 
Generally all working ok.
I'm starting to get files onto the USB. I've used some HFE superdisk format (2550K) so I can get more, and larger, instruments onto each image.
Normal size (800K) images IMG and hfe both seem to work fine - read/write including spannign disks..
When I save to a Superdisk hfe I'm getting a lot of errors (mostly write verify). Smaller files are more successfulk, as are images that have more free space, but struggling to get *any* large instruments on a fuller image to save,

Is this just expected unreliablilty with the superdisk format? 
I've tried a few USB sticks and whilst some seem slightly better there's little practical difference in failures of larger instruments.
I was on flashfloppy v3.15, now updated to 3.22.

keirf, 2020-12-21:
Can you point me at or attach a superdisk image? Also you can run logfile firmware (see the wiki) and collect an FFLOG.TXT file which may provide diagnostic info.

JC-III, 2020-12-21:
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/5725221/FFLOG.TXT)

[Superdisk_E16_SD.zip](https://github.com/keirf/FlashFloppy/files/5725223/Superdisk_E16_SD.zip)

See attached, I didn't get a 'write verify error' this time, rather the write got stuck in a loop towards the end and I had to restart.

keirf, 2020-12-21:
Okay, the last cylinder is inaccessible because FlashFloppy always uses cylinders 254 and 255 for Direct-Access mode to the USB drive. Is it possible to make a 253-cylinder image instead of 254-cylinder? Otherwise possibly I could disable Direct-Access mode if a 254- or 255-cyl image is mounted.

JC-III, 2020-12-21:
re. "Is it possible to make a 253-cylinder image instead of 254-cylinder?"  - not sure if/how I could do this. 
I'm using '_Multiple EFE2IMG & HFE' to generate the images.

JC-III, 2020-12-21:
In case it helps, this is a log whereb I got the actual 'write verify error' message on the EPS16.
[FFLOG.TXT](https://github.com/keirf/FlashFloppy/files/5726247/FFLOG.TXT)

keirf, 2020-12-21:
I will probably need to make you a new firmware to test.

JC-III, 2020-12-21:
OK, let me know - happy to try that out.

keirf, 2020-12-22:
Here's a link to an automated build to try. Download the linked Artifact (requires you to be logged into Github). Unzip, and then unzip the non-debug zip, and install the firmware using the UPD file.

https://github.com/keirf/FlashFloppy/actions/runs/438471970

keirf, 2020-12-22:
I realised there's an even bigger fix, which is not to skip multiple cylinders when seeking towards cyl0 from high-numbered cylinders. That fix is in the following build:

https://github.com/keirf/FlashFloppy/actions/runs/439085386

JC-III, 2020-12-22:
Yes - I had tried it, and was still getting some verify errors, and when I tried to save to a nearly full disk ans it got up to the 250's sectors I got a 'File operation Error'. I'll try this new one - thanks.

JC-III, 2020-12-23:
That seems to be working well. I've made several 'superdisk' images with large instruments, and disks near to full and so far all saves and resaves have been successfull - no failures yet. I'll try some more tomorrow - but that's looking very promising.

keirf, 2020-12-31:
This is fixed in latest release v3.23

closed 2020-12-31 by keirf