keirf/flashfloppy #407: Durango F85 possibility

Tubatstuff, 2020-12-26:
I'm probably one of the very few people with not one, but two Durango F85 working systems.  Late 1970s system, using 100 tpi 5.25" disks on Micropolis-interface drives.   Encoding is proprietary GCR with 12 512-byte sectors on 77 tracks for a bit over 900K per soft-sectored disk.  Bit clock rate is around 300KHz and uses DSDD floppies.  The system uses a WD1781 FDC with lots of supporting logic.

I can accommodate the details on the interface differences (e.g. pin 6 = READY);  

I'm quite familiar with STM32 MCU development;  what I'm wondering is, off the top of your head, can the code in FF be coaxed to handle this oddball format?  I can encode the bitstream any way that will work.

Thanks,
Chuck

keirf, 2020-12-26:
Yes, encode as HFE. It is a raw bitstream format. It's a bit weird as the streams of the two sides of a cylinder are interleaved, and the individual bytes are bit-wise little endian. But not too hard to generate. I have examples in C and Python.

Tubatstuff, 2020-12-26:
Thanks, Keir.   Next step is to figure out the details in converting a Catweasel stream into FF format.   Shouldn't be too hard.   A lot of my old floppies are starting to degrade after more than 40 years, so I'd best start preserving them.

closed 2020-12-31 by keirf