keirf/flashfloppy #408: Detailed description of HFE file format.

Tubatstuff, 2020-12-27:
All I can find on this subject is the 1.5 page PDF for HFE 1.0.   Leaves a lot to be answered, particularly for later versions of HFE.

Is there **anywhere** a more detailed discussion is posted?

keirf, 2020-12-27:
The canonical resource would be the HxC implementations. In the libhxcfe loader sources, part of hxcfloppyemulator on sourceforge. There's an implementation in FlashFloppy too: https://github.com/keirf/FlashFloppy/blob/master/src/image/hfe.c

keirf, 2020-12-31:
I would move this to Discussions but the action has gone awol in the Github UI. So I'll close it instead. If you want to discuss HFE stuff some more, do it under Discussions as this is not a FlashFloppy issue and doesn't belong here. Cheers :)

closed 2020-12-31 by keirf