keirf/flashfloppy #411: Better Amiga WB integration?

callimero, 2020-12-30:
After a short intermezzo with hcx I am back.

I need to use a OLED now because after beeing in FFManager always rebooting is a bit mehhh...

Is there a better WB integration coming like hxc has it, meaning go back to WB, switch screens...

keirf, 2020-12-31:
No plans to be honest. What's your setup? Have you considered FF OSD (On Screen Display) hack?

callimero, 2020-12-31:
Ok, thought it could be a simple thing to implement, no idea how the other implemented it, is it maybe harder than it seems?

In fact I dug out a oled from my parts box and will try. OF COURSE :) I am having more fun tinkering than using stuff....

I use a Picasso GFX and a Flickerfixer on a modern Monitor, so I guess OSD is nothing for me ?

keirf, 2020-12-31:
Ah yes, no OSD with RTG. But since you have a big box Amiga your best bet is OLED and rotary encoder on your Gotek. I guarantee it will be a more enjoyable experience than the file selector!

callimero, 2020-12-31:
Failed to convert my SPI OLED to I2c...

Now ordering one.... BUT all I found have 0x78 or 0x7A adresses?!

keirf, 2020-12-31:
It depends how you define an i2c address. The 78 here is probably what I call 3c. I recommend 128x64 by the way as the smallest can be made to fit the gotek case

closed 2022-12-19 by keirf