keirf/flashfloppy #413: I can't rotate the oled screen

5146jorge, 2021-01-02:
Hello, thank you very much for your work
I have the latest version and a problem has arisen, when I try to rotate the screen there is no way I use this command:

display-type = auto -rotate 

But there is no way, what am I doing wrong?

A greeting and sorry for the inconvenience

keirf, 2021-01-02:
Remove the space: `auto-rotate`

5146jorge, 2021-01-02:
It does not work tried two goteks and two usb memory same result the text does not rotate
this is my command line:

display-type = auto-rotate

keirf, 2021-01-02:
Ah sorry, try: `display-type = oled-128x32-rotate`

Substitute 128x64 above, if your display is 128x64 rather than 128x32

5146jorge, 2021-01-02:
Nothing does not work in the two goteks they do not rotate nor does the inverse command do anything

keirf, 2021-01-02:
Please zip and attach your FF.CFG here

5146jorge, 2021-01-02:
## FF.CFG: Example FlashFloppy Configuration File

# Place in the root folder or FF/ subfolder of your USB drive.
# NOTE: If FF/ exists, IMG.CFG must reside there, not the root folder.

# Uncommented lines below are the default settings.
# Uncommented options cannot be overridden by settings in other config files.

##
## DRIVE EMULATION

# Floppy-drive interface mode (interface pins 2 and 34)
# jc: Specified by jumper JC (open: shugart, closed: ibmpc)
# shugart: P2=DSKCHG, P34=RDY (Amiga, Atari ST, many others)
# ibmpc: P2=unused, P34=DSKCHG (IBM PC interface)
# ibmpc-hdout: P2=HD_OUT, P34=DSKCHG (not generally needed: prefer 'ibmpc')
# jppc: P2=unused, P34=RDY (Japanese PC standard)
# jppc-hdout: P2=HD_OUT, P34=RDY (Japanese PC alternate: prefer 'jppc')
# akai-s950: Legacy alias of 'jppc-hdout', previously used for Akai S950
# amiga: P2=DSKCHG, P34=DRIVE_ID (not generally needed: prefer 'shugart')
interface = jc

# Host platform: Improves image-format detection for generic types such as IMG
# acorn: Acorn ADFS
# akai: Akai (S01, S20, S950), Korg, SC Prophet 3000
# casio: Casio (FZ-1)
# dec: DEC (RX33, RX50)
# ensoniq: Ensoniq (ASR, TS, etc)
# fluke: Fluke 9100
# gem: General Music (S2, S3)
# ibm-3174: IBM 3174 Establishment Controller
# memotech: Memotech
# msx: MSX
# nascom: Nascom
# pc98: NEC PC-98
# pc-dos: PC DOS Format (geometry determined from Bios Parameter Block)
# tandy-coco: Tandy Color Computer (CoCo)
# ti99: TI-99/4A
# uknc: UKNC / DVK Soviet PDP-11
# unspecified: Common default geometries (including IBM PC)
host = unspecified

# Pins 2 & 34 output (drive->host) manual configuration
# auto: Auto-configure from interface= setting
# nc: Unused/No Connection [eg. if pin is an input (host->drive) on your host]
# low: Always 0 volts (0v)
# high: Always 5 volts (5v)
# rdy: Drive ready (Ready = 0v)
# nrdy: Logical complement of above
# dens: Density mode (High Density = 0v)
# ndens: Logical complement of above
# chg: Disk changed (Changed = 0v)
# nchg: Logical complement of above
# Values: auto, nc, low, high, rdy, nrdy, dens, ndens, chg, nchg
pin02 = auto
pin34 = auto

# Forcibly write-protect images, or respect the FAT read-only attribute?
# Values: yes | no
write-protect = no

# Filter glitches in the SIDE-select signal shorter than N microseconds
# Values: 0 <= N <= 255
side-select-glitch-filter = 0

# Rotational offset of disk after a track change
# instant: No rotation during track change
# realtime: Emulate rotation of disk while track is changing
# Values: instant | realtime
track-change = instant
[... 176 lines cut ...]
display-scroll-pause = 2000

# LCD/OLED long filename scroll rate during navigation (ms per update)
# Values: 0 <= N <= 65535
nav-scroll-rate = 80

# LCD/OLED long filename pause before scroll, during navigation (milliseconds)
# Values: 0 <= N <= 65535
nav-scroll-pause = 300

##
## MISCELLANEOUS

# Speaker volume for head STEP
# Values: 0 <= N <= 20
step-volume = 20

# Report the specified version number to host software
# Values: <quoted-string> ("" means report real version)
# eg. da-report-version = "v3.0.0.0"
da-report-version = ""

# Automatically extend certain types of truncated image file (SSD,DSD,TRD)?
# Values: yes | no
extend-image = yes

5146jorge, 2021-01-02:
[FF.zip](https://github.com/keirf/FlashFloppy/files/5760603/FF.zip)

keirf, 2021-01-02:
I just copied your FF.CFG onto my drive and it rotated perfectly. Are you sure you're running an OLED? An LCD display doesn't work with -rotate nor -inverse.

kroy-the-rabbit, 2021-01-03:
This was working fine until I updated to 3.23.

kroy-the-rabbit, 2021-01-03:
> I just copied your FF.CFG onto my drive and it rotated perfectly. Are you sure you're running an OLED? An LCD display doesn't work with -rotate nor -inverse.

I walked back versions, 3.22 seems to be where it got broken.  With 3.21, the flip works like it should, but 3.22 and 3.33 are ignoring my `auto-rotate`

kroy-the-rabbit, 2021-01-03:
If it helps, I have whatever [this screen is](https://www.amazon.com/gp/product/B0761LV1SD)

keirf, 2021-01-03:
v3.22 explicitly broke `auto-rotate` which has never been a recommended `display-type` option. The correct line is:
```
display-type = oled-128x32-rotate
```
Which is what was in the FF.CFG you attached, and I tested, and it worked for me. Does it work for you?

5146jorge, 2021-01-03:
I did it!
But ... I can't explain it :(
at the end of so much putting out usb try on other computers and others started to work

closed 2021-01-03 by keirf