keirf/flashfloppy #414: How to contribute to the Wiki?

Ben-Chapman, 2021-01-04:
hey @keirf, I have some minor updates I'd like to propose to the [Firmware Programming](https://github.com/keirf/FlashFloppy/wiki/Firmware-Programming) page on the Wiki. These would include some additional detail on how to configure the programming headers for USB programming, and an associated diagram for jumper configuration.

What's the best way to get these updates over to you for review?

keirf, 2021-01-30:
You can possibly fork the wiki repo and do a PR? Not sure if that works for wiki repos. Or point me at your repo and I can manually pick commits from it.

Ben-Chapman, 2021-02-13:
hey @keirf, sorry for the delay with this. It doesn't look like you can fork the Wiki (but the wiki content does with across with a `git clone` of the main repo).

You can find the proposed changes at https://github.com/Ben-Chapman/FlashFloppyWiki/pull/1/commits/44afe76ffbc811f94a11f411e287f1fb1ce5e826

It's a straightforward proposal:
  - Modification of Firmware-Programming.md
  - Addition of two supporting images

Please let me know if you have any questions / feedback on the changes.

keirf, 2021-02-13:
Thanks, pulled the changes in.

closed 2021-02-13 by keirf