keirf/flashfloppy #415: [Wishlist] Add scrolling to display long filenames in the image-selector program *enhancement

Takehaniyasubiko, 2021-01-07:
Currently, if a filename is really long, the image-selector program won't display it in full, which can lead to confusion when using the TOSEC archives. For example, sometimes there are various cracked versions included, but they share most of the filename, so you have to see the filenames in full to know which version is being selected.

Would it be possible for the image-selector program to scroll long filenames, so that we can which file we're actually selecting?

Tallguy297, 2021-01-19:
I second this request.
Could the file manager also be modified to show these long TOSEC archive file names?

Thank you.

keirf, 2021-01-30:
Not by me. Use OLED or OSD.

Takehaniyasubiko, 2021-01-30:
> Not by me. Use OLED or OSD.

I am using an OLED screen and the very long TOSEC filenames are not displayed in full. Besides, my Gotek is mouned where the original floppy would be, so it's a pain to look at the back of Amiga to select an ADF. The image-selector program is much quicker and practical for such situations.

keirf, 2021-01-30:
It's impossible to display them in full within the slot assignment screen or on OLED (in hxc mode) as the HXCSDFE.CFG only stores about 50 characters per slot.

keirf, 2021-01-30:
Well, ok, the config file could get another revision and make the slots larger, to be fair. Well let's leave this open, it would be nice to have.

Takehaniyasubiko, 2021-01-31:
Can you please modify the image-selector program, so that it can scroll the filename, just like the OLD OSD does?

closed 2022-06-22 by keirf