keirf/flashfloppy #416: Motor Sound from JB pins very quiet

Hiddenevil, 2021-01-10:
I have been trying without success to install a buzzer inside my hotel. But where as in the guides I have seen on YouTube, showing a clear buzzing sound for the floppy motor. On my hotel it is barely audible. I have to have my ear very close to the buzzer to hear the motor access.

Model: SFRC922

keirf, 2021-01-10:
How have you wired it up, and what sounder/speaker are you using exactly? It will be an issue with the circuitry/sounder, not the Gotek.

Hiddenevil, 2021-01-10:
Hi there

I've tried it with several types of piezo transistor. Just the bog standard
sort you use in a pc, see attached pic. I can get it to buzz if I wire it
up to the MO pins. But as soon as the drive is accessed, it proceeds to
tick every second. Not the spaced out drive tick, but a more rapid tick
which only stops when the drive is being accessed.

I even tried doing a mod but it did nothing for me.

Ohh btw if I hook the buzzer up to the same points as the DRIVE ACTIVITY
LED. The buzzer performs exactly as it does when connected to the MO pins.
I don't know if that is of any help.

Firmware was updated November last year I think. So NOT the most recent but
recent enough. When I was given the unit it didn't have the flash floppy
firmware installed on it.

Many thanks
James

keirf, 2021-01-10:
Attach a photo of your wiring.

Hiddenevil, 2021-01-10:
The buzzer goes directly on JB one, theres no really wiring to it. I've
tried swapping the red and black wires on the pins but that only stops it
from sounding at all. Or if it is, it's even quieter and thus inaudible
without a stethoscope lol

I did try this, which i saw online and thought it worth a shot as things
weren't working out for me. However it resulted in the same output on the
buzzer.

Hiddenevil, 2021-01-10:
Picture of the board, minus buzzer on the JB pins.

The gotek is used externally with an internal DF0: swapper switch, fitted
inside the a500+.

Hiddenevil, 2021-01-10:
Sorry forgot to attach image

keirf, 2021-01-10:
No picture I'm afraid. At the end of the day it does work for a lot of people. Worst case your JB output pin is fried somehow. You can't really confirm this without another Gotek to test against.

Hiddenevil, 2021-01-10:
True, but if it was fried surely it wouldn't output anything at all?

If I recall when I put a meter over it last Friday night. I was getting 5v
on the rear pin and the other was ground.

As I don't have another gotek, I don't know if that's normal.

keirf, 2021-01-10:
The JB pin should be 0v at rest and pulses high (to 3.3v approx) to make a sound. It certainly shouldn't be 5v. The other pin on the JB column is indeed tied to ground.

Hiddenevil, 2021-01-10:
Well then, at least that's something I guess. It certainly did show a value
of 5v and not 3v. I'll test it again when I get back to my bench later,
just to be sure.

Many thanks for your help, I've been struggling with this since before
Christmas and I was all but ready to give up.

keirf, 2021-01-10:
Yes, test it with nothing connected to it and see what voltage there is. That's useful to know.

closed 2021-01-30 by keirf