keirf/flashfloppy #417: Ensoniq ASR-X [Non Pro] and Disk not readable

andr229k, 2021-01-18:
E: I have this Ensoniq ASR-X [standard version] installed with a Floppy Emulator incl. FlashFloppy. The Ensoniq ASR-X has internal version 2.67 [2.50+ is important to be able to read example ASR-10/EPS 16+ disk as per following [link](https://syntaur.com/tech_asr-x.html)] 
Jumper = S0
Interface = Shugart
Host = ensoniq

S: It initiates the handshake to start loading and then gives the error “Disk not readable”.
D: The attached disk works fine, when using HxC as firmware.
A: Tried various USB-sticks.

**Attachments:**
1. Error message
2. HxC Info [[Link](https://hxc2001.com/download/floppy_drive_emulator/support.htm)]
3. Disk example

[image]
[TECHSTART-30.zip](https://github.com/keirf/FlashFloppy/files/5828196/TECHSTART-30.zip)
[image]

keirf, 2021-01-18:
Have you tested on HxC yourself? And if so, what HxC type and what settings?

On FlashFloppy does the load make any progress or does it immediately fail with no obvious disk access?

andr229k, 2021-01-19:
Ticket can be closed, issue was around the USB-stick itself.

closed 2021-01-19 by andr229k