keirf/flashfloppy #420: 20x4 LCD display not working

shift838, 2021-01-29:
I recently purchased a new GOTEK drive and loaded FlashFloppy 3.23 FW.  I have followed the guide to hook up a 20x4 LCD display with I2C backpack. when the unit powers up only the display characters are solid white blocks.  adjusting the contrast control only dims or brightens them.  characters are not readable.  (a 16x2 display works fine)

I have included in the FF.CFG file on the root of the USB 

display-type = lcd-20x04

I have also tried 

display-type = auto

but neither works.  What am I missing ?

omikron88, 2021-01-29:
Chck the backpack I2C address. It has several choices often, selectable by
jumpers  or zero resistors.

shift838, 2021-01-29:
I had a bad ground wire!

closed 2021-01-30 by keirf