keirf/flashfloppy #421: Yamaha PSR 1700 compatibility

KARAPAHUU, 2021-01-31:
Hello everybody and a VERY VERY GREAT thanks for all this work !!!!

I worked for a long time to replace the floppy drive of my Yamaha PSR 1700 keyboard and I never found a really explicit tutorial.

I bought a GOTEK SFRM72-FU.DL drive because the keyboard only reads 720kb floppy disks.

I made a 24-34 pin converter using this table, after several test to find the good configuration

[image]

For this I'm using the Yamaha electronic diagram

[image]

On the GOTEK only the S1 jumper is present
The original firmware 126.F01 was replaced by Flashfloppy 3.25 (cable to cable method)
No special configuration (FF.CFG original)
The USB stick is formatted in FAT 32 and I created IMG files with USB_Floppy_Manager_v1.40i (I attach an empty IMG file) 
[FIMG000.zip](https://github.com/keirf/FlashFloppy/files/5899692/FIMG000.zip)

I am so happy and ready to answer your questions. I am preparing a video tutorail and add the link in this issue soon

( so sorry for my poor english)

keirf, 2021-02-07:
Added to the compatibility list https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#yamaha
Added warning that this is not easy (need to make an adapter to the Yamaha mainboard or FFC connector).

closed 2021-02-07 by keirf

KARAPAHUU, 2021-02-07:
Ok thanks !

This is the link to my video to explain how craft the converter

https://www.youtube.com/watch?v=s5ZfBuPnyA0&t=19s

now-its-dark, 2024-08-24:
Here is an adapter design for this purpose, you can find them prebuilt in various places 
https://github.com/marucome09/24PFFCto34P_conv

There is also a super overpriced option from Flexidrive https://floppyemulator.com/products/34-24p1-25adaptor/