keirf/flashfloppy #422: Numbering of sides on display

SvOlli, 2021-01-31:
I'm now using FlashFloppy in a Commodore 1581 replica drive, thanks for having this option.

There is one thing that's "odd" using D81 images. The 1581 does the numbering wrong. Side 1 contains sectors 0-19, side 0 sectors 20-39. I've read through the FF.CFG text, but could not find any "switch sides in display" option. Did I miss it, is it done completely different, or is there a feature missing? ;-)

keirf, 2021-02-01:
The thing is the display shows 'physical' cylinder and head information. So 0 means lower, 1 means upper. And, for example, if accessing a HFE image file marked for double stepping, the HFE image's cyl 20 (say) will display as 40 on the Gotek. The actual side swapping for D81 happens within the image handler code. It's fair to say that 1581 numbers the sides differently than standard: probably the /HEAD signal is simply inverted and noone noticed.

SvOlli, 2021-02-05:
A good explanation, but it still leaves my "biggest" question uncovered. Is there a better solution than me compiling a special 1581 version that does an XOR 1 in the output of the side?

keirf, 2021-02-05:
No

SvOlli, 2021-02-05:
Okay, thanks! :)

closed 2021-02-05 by SvOlli