keirf/flashfloppy #423: Submission: Example Host IMG Profile for Osborne CP/M

dandoore, 2021-02-01:
Tested on every type of disk image I could find on an OCC2 with DSDD, the only oddity is that the Vixen format files have some different sizes:

## IMG.CFG for Osborne systems.

# NOTE: The tags match on filesize alone. If you wish to define an explicit
# tagname match, you can for example add 'occ1' to the square-bracketed tags
# to limit matches to filenames of the form *.occ1.{img,ima,dsk}

# Osborne 1 OCC1/1A (for units without double density upgrade fitted) - SSSD 100Kb
[::102400]
cyls = 40
heads = 1
secs = 10
bps = 256
skew = 2
mode = fm
iam  = no

# Osborne Executive OCC2 (also OCC1/1A with double density upgrade fitted) - SSDD 200Kb
[::204800]
cyls = 40
heads = 1
secs = 5
bps = 1024
mode = mfm
iam  = yes

# Osborne Vixen OCC4 (also Executive OCC2 with double sided upgrade fitted) - DSDD 400Kb
# Note: Multiple sizes for OCC4
[::409600]
cyls  = 40
heads = 2
secs  = 5
bps   = 1024
skew  = 2
mode = mfm
iam   = yes
[::409728]
cyls  = 40
heads = 2
secs  = 5
bps   = 1024
skew  = 2
mode = mfm
iam   = yes
[::409856]
cyls  = 40
heads = 2
secs  = 5
bps   = 1024
skew  = 2
mode = mfm
iam   = yes
[::419840]
cyls  = 40
heads = 2
secs  = 5
bps   = 1024
skew  = 2
mode = mfm
iam   = yes

keirf, 2021-02-07:
It's funny that there are multiple sizes for OCC4. Perhaps I should make that specifiable in the `[]` syntax. Or perhaps it is easier to make OCC4 a blanket default here (ie. specify it as `[]`)?

dandoore, 2021-02-09:
Blanket default would work, even though OCC4 format is probably the least prevelent so multiple sizes (or a greater than but less than range) in the [] would probably be cleaner.

commit 0b1ed450 (2021-02-15): Example IMG.CFG file for Osborne CP/M systems. Thanks to Dan Doore.
Closes #423
closed 2021-02-15 by keirf