keirf/flashfloppy #424: Amiga 1200 Gotek shows up as "DF0:????" in WorkBench

ShamblerDK, 2021-02-05:
When I insert the USB-key with an ADF-image on, looking at the OLED I can see that it reads it just fine. But when the Amiga boots it doesn't even seem to try and read from it. Also the red LED is constantly on but the green LED never lights up.

FlashFloppy version: 3.23
Gotek config: Set to DF0

Am I looking at a floppy cable issue?

keirf, 2021-02-06:
Jumper at position S0 only? The green LED should flash a little at boot. Could be bad cable indeed.

ShamblerDK, 2021-02-06:
Thank you for your reply. I found another cable and it's indeed working
just fine now. I just wish the audio emulation through the buzzer was a bit
louder :-)

You may close the issue.

closed 2021-02-06 by keirf