keirf/flashfloppy #425: Issues with Alesis DataDisk (WD1772)

FuSoftware, 2021-02-07:
Hello,

I'm currently trying to get my Gotek, using the FF 3.23 version, to work with an Alesis DataDisk using the 1.2 ROM version, and the WD1772-PH02-02 chip. The Gotek 34-pin interface is configured as shugart (as per the chip's specification, and I've confirmed using the original floppy drive that the interface is shugart, and that the WD1772 is working).

I'm currently stuck with the DataDisk reporting that there is "NO DISK IN DRIVE" when trying to either read from it, or format it. Meanwhile the Gotek usually reports on the 7-Seg display either " 0_" or " 1_".

I'm using a custom IMG.CFG format as found on the HxC forums regarding that device :

```
[datadisk]
cyls  = 80
heads = 2
secs  = 5
bps   = 1024
id    = 1
mode  = mfm
iam   = no
rate  = 250
gap3  = auto
rpm   = 300
```

I've tried 720k, 800k, 1.2M and 1.44M image files with similar results.

Is there a known working configuration for that device, or is there something I'm missing when it comes to get it to work ?

Thanks in advance !

keirf, 2021-02-07:
Have you tried jumper at JC (or interface=ibmpc in FF.CFG)? No disk in drive doesn't sound like it's trying to read the disk at all?

FuSoftware, 2021-02-07:
Oh damn, that worked. My own research and testing of the original drive showed it was shugart, but I guess I was mistaken. Putting the jumper did the trick, thanks!

closed 2021-02-07 by FuSoftware