keirf/flashfloppy #427: Ensoniq MR61 image format

baritonomarchetto, 2021-02-09:
Hello everyone, sorry for opening a issue request for my knowledge lacks.
I am trying to use a Gotek drive with Flash Floppy in my Ensoniq MR61. The instrument supports HD/DD disks.
The Gotek drive has only jumper S0.
I have configured FF.CFG specifying "host = ensoniq" and tryied both "interface = ibmpc-hdout" , the default "interface = jc" and "interface = shugart" with no luck.
The synth sees the floppy image, starts formatting it if I give it the command but then stops with a "can't format floppy" message.
I am using the attached image (zipped because the site gives an error otherwise).

Anyone could help? 

Thanks in advance! :)

[DSKA0001.zip](https://github.com/keirf/FlashFloppy/files/5951105/DSKA0001.zip)

baritonomarchetto, 2021-02-10:
Some additional info, just in case this can help figuring out what the issue is. 
With "ensoniq" and "shugart" as the only "not default" varibales and a disk image like the one I uploaded, the keyboard starts formatting the disk. Flash floppy starts counting, counts up to 79-80 (takes a couple of minutes) and then fast reverts back to 0 and then cycles fast from "0" to "2".
I don't know what does this means

baritonomarchetto, 2021-02-11:
Made some progress: I have tryied another USB stick and the formatting process succeed. It is strange that the "old" stick was not working because I use the same kind of sticks with other FF-keyboards with no issues.

There's still a problem: if I try to write on the just formatted image, the keyboard returns a "Disk's been changed!" error, which from the manual should be displayed "whenever you try to load data from a disk, and there's a different floppy disk in the drive."

Not sure how Flash Floppy handles images and if with some existing option of the configuration file this could be fixed.

Any suggestion would be much appreciated :)

keirf, 2021-02-11:
The DSKA0001 you attached is actually an HFE image, not an IMG. And it is double density.

See below a zip containing probably-good DD and HD IMG files for Ensoniq. This assumes MR-61 uses the Ensoniq 800k/1600k custom disk layouts. The IMG files are preferred if they work!

If not, I also include blank DD and HD HFE files. You can try formatting these and send them back here and I can tell you what format they are. Possibly then you could still use IMG files of the appropriate size.

[mr61.zip](https://github.com/keirf/FlashFloppy/files/5964155/mr61.zip)

baritonomarchetto, 2021-02-11:
Hi Keirf, I had mixed results with those images.
With one stick (lets call it "A" stick) I was able to format the ddimg.img file, but not the other three;
With the other stick ("B") I was able to format both hdimg.img and hdimg.hfe, but not the other two.

Anyway, none of the images worked: I always had a "Disk's been changed!" error trying to save something to disk.

Here attached is the formatted hdimg.hfe from stick "B"

(thanks for the help so far!)

[DSKA0000.zip](https://github.com/keirf/FlashFloppy/files/5964736/DSKA0000.zip)

keirf, 2021-02-11:
Okay the format is normal 1M44 (PC). So try the attached 1M44 IMG file. I also include a 720k IMG file.

You may also want to try `interface=ibmpc-hdout` in FF.CFG configuration file and see if that helps.

[mr61.zip](https://github.com/keirf/FlashFloppy/files/5966405/mr61.zip)

baritonomarchetto, 2021-02-11:
Great! Works like a charm now!!

As you said, I had to specify `interface=ibmpc-hdout` in FF.CFG file. Here attached is a formatted image, just in case it could be of some use for someone.

Ensoniq MR61 compatibility confirmed then!

jumper to S0
interface = ibmpc-hdout
host = ensoniq

Keirf, may I ask another thing before closing the ticket? May I suggest to upload some default, unformatted, images for the most common floppy disk formats? I find it the most "complex" thing to do every time I install the emulator on a new instrument. Maybe they are somewhere on github, but I cannot source them

Thanks for the great work!

[DSKA0000.zip](https://github.com/keirf/FlashFloppy/files/5967161/DSKA0000.zip)

keirf, 2021-02-15:
MR61 is now explicitly mentioned in the wiki (https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#ensoniq).

Empty images, and your MR61-formatted image, are added to the FlashFloppy image library: https://github.com/keirf/FF_Images

closed 2021-02-15 by keirf

baritonomarchetto, 2021-02-15:
I was referring to unformatted images, so that one can source an image resembling a specific disk (i.e. DS/DD, DS/HD, etc) and start from there. In example, my issue here was merely related to the use of the wrong image and having such blank, unformatted images would have been useful being the type of disk known.
I am not a floppy expert unfortunately, so maybe my requested (suggestion) makes no sense

baritonomarchetto, 2021-03-18:
Hello Keirf, just a head up to tell you that I have just installed flash floppy in a MR76 too: works like a charm!
Same settings as MR61, as one would expect. I used directly the images from the MR61 on the MR76 and I was able to upload all files I had saved on the MR61. Cross compatibility it is :D
Thanks