keirf/flashfloppy #428: Problems with PSR-1000

0C-34, 2021-02-09:
Hi, 
I modded a gotek for my grandfather so he could use it with his old Yamaha PSR-1000 synth.
The gotek has a oled display, a Rotary encoder and a beeper from an old pc.

The modded gotek works flawlessly in ibm compatibles (tested in a 386 and a pentium3 machine) and an amiga500, but on the synth it lights up, it lets me scroll trough the files, but I doesn't load anything and remains stuck on track 0,0.

What can I do to give more meaningful details? 
Did anyone already resolve this? I have only seen 1 person on a random forum solving this, but it's for a very old version of FlashFloppy (1.4.0) and it doesn't really help. 

Have a good day.

keirf, 2021-02-11:
I don't have any info on PSR-1000, but at a guess: Try changing jumper S1 to S0. Try removing jumper JC if you have it installed.

This may get you somewhat further!

0C-34, 2021-02-11:
>Try changing jumper S1 to S0. Try removing jumper JC if you have it installed.

Oh, I forgot to put the drive select jumper e.e
I removed all the jumpers when i got the gotek and used the config file to specify if it had to be a shugart or a ibmpc floppy drive (I used the alternative logfile version to check if it was working and it was) but I completely forgot about that.
As soon as I get around testing I'll report if it works, so that you can add the PSR-1000 to the compatibility list on the wiki, thanks and have a good day!

0C-34, 2021-02-14:
> This may get you somewhat further!

Thanks a lot!
The Yamaha PSR-1000 is plug&play and it just needs the gotek to be set in IBMPC mode and a drive set, it works either with the jumper on S0 or S1 with no problems.
Thanks again for the help

closed 2021-02-14 by 0C-34