keirf/flashfloppy #430: Distribution Media Format (DMF)

marx-veix, 2021-02-14:
Hi,

Could it be possible to add DMF support to FlashFloppy?
https://en.wikipedia.org/wiki/Distribution_Media_Format

With gratitude,
Marx-Veix

keirf, 2021-02-15:
Should already work -- try renaming files to have IMG suffix? Is DMF a standard suffix for these images?

Leo-PL, 2021-02-21:
I remember it worked with v3.21, but IIRC with v3.22 it stopped again. I'll flash v3.21 in a moment and check back.

keirf, 2021-02-21:
If so we can work out the offending patch and fix it.

Leo-PL, 2021-02-21:
I'll have to recover this configuration somehow first, just restoring v3.21 didn't help, as well as removing IMG.CFG altogether, neither did applying IMG.CFG from #348 - in which I noticed a typo in the image size section - was 170320, not 1720320. I'll try to debug this further and get back with more information to you. What I currently get is "Not ready" error on attempting to copy a file from the disk - the directory lists properly.

keirf, 2021-02-21:
It's not clear it's ever worked reliably, for some reason. It's weird as it's a pretty straightforward format really.

keirf, 2021-02-21:
I probably recommend playing with IMG.CFG settings. Various `interleave` values, perhaps vary `gap3` a bit too.

Leo-PL, 2021-02-24:
From information I could find in a fragment of Hardware Bible by Winn L. Rosch, cskew should be 3, not 0, interleave is 2:1. But setting this gave me "sector not found" errors immediately, right now I can access at least sector 0 to list root directory.
I found the patent describing DMF here: https://patents.google.com/patent/US5745313A/en - I'm trying to deduce parameters from it.

keirf, 2021-02-24:
What would be ideal is if someone has a real working DMF floppy disk and then dump it using Greaseweazle (or Supercard Pro, or Kryoflux, or FluxEngine, or...) then we can directly see the track layout.

Leo-PL, 2021-02-26:
I'm currently trying to get a hold of one. When I do find one, I'll take a capture with Greaseweazle. It's a godsend that one can use a plain "Blue Pill" board for that. I might try my luck with HDCopy, too.

Leo-PL, 2021-02-28:
I found a very important quote in the Hardware Bible, by Winn L. Rosch:

> Before CD-ROM media took over the job of program distribution, software publishers sought and developed ways of shoehorning more data onto every floppy disk. The leading extra-capacity alternative was Microsoft's Distribution Media Format (DMF). This variation in the high-density design allowed Microsoft to fit 1,720,320 bytes on a standard high-density 3.5-inch floppy disk in place of the more normal 1,474,560 bytes (nominal 1.44MB). The DMF format differed from the standard format in that it used 21 sectors per track instead of the normal 18. DMF squeezed more sectors on each track by reducing the interrecord gap (the space between sectors) down to nine bytes.
> The differences went deeper, however. Each track used a 2:1 interleave factor so that sectors did not appear in order. This interleaving resulted in slower reading because the disk had to spin around twice for each track to be read. The DMF format also skewed the sectors on adjacent tracks by three sectors so that Sector 1 on Track 4 sat next to Sector 1 on Track 2.
> The small interrecord gap made DMF disks difficult to write to with normal floppy disk drives. In fact, Microsoft called DMF a read-only format. In any case, you cannot write to DMF disks using ordinary software. However, several special utilities are available for copying and even creating DMF disks. Note that Microsoft enforces a limit of 16 entries in the root directory of a DMF disk by only allocating a single cluster to service as the root, so DMF-formatted disks usually use a subdirectory structure for their contents.
> Microsoft operating systems cannot ordinarily read DMF floppies. Consequently, DMF floppies were used only when software required multiple floppy disks for its installation. The first disk of the installation package—usually called the setup disk—loaded software to reprogram your floppy disk controller to read the DMF disks.

And the last paragraph is the clue. When I booted the system from Win95 installer bootdisk, reading the DMF image indeed worked. I set the cskew to 3, but I recall this also worked with default settings, with cskew 0. I think it might be beneficial to quote this paragraph somewhere in documentation. I'll do some more testing w.r.t cskew value and get back to you, although this should probably only affect read performance, not as much as compatibility.

Previously I was attempting to read DMF floppy from within MS-DOS 6.22 which predates DMF format.

Leo-PL, 2021-03-02:
Now, knowing of `/dev/fd0u1680` existence, I even went so far as to test writing DMF images on FlashFloppy, and it works. Again I used Debian 1.3 for easier testing. 
I started by copying an existing image, calculating its `md5sum` - correct, no surprise here, and zeroing it out with `dd`. On my PC, `hexdump` showed all zeroes indeed. Back in FlashFloppy, I copied back the image I backed up before testing to the device, and verified with `md5sum` - again correct. To be extra sure I `diff`ed it with original image I copied it from - no differences.

So now that even wrting works, It might be worthy to update the parameters in source code. I wonder if there isn't actually a bug lurking in the detection logic, as the only parameter I've actually had to override was cskew, and also FlashFloppy was operational, even if cskew was set to 0, the same as default in source code.

keirf, 2021-03-02:
Ok, so what are the conclusions:
1. DMF needs custom driver support not present out of the box in Windows. I can add a comment on this to the IBM PC section of the Host Platforms page of the wiki? Is there anywhere better to put it.
2. Should DMF have a non-zero skew? I guess we haven't seen an actual real dump of an original DMF disk, and the documentation refderenced above is ambiguous to me (I can't visualise/interpret "Sector 1 on Track 4 sat next to Sector 1 on Track 2").
3. Is there a claim that DMF images are not recognised on latest firmware without IMG.CFG?

Leo-PL, 2021-03-02:
Ad 1: Not exactly, it needs version of DOS or Windows that is "recent enough to contain support". Installer bootdisk for Windows 95 is new enough as is Windows 95 itself, while MS-DOS 6.22 and WfW 3.11 aren't. Linux 2.0.y and later is capable as well, and can even deduce proper floppy size from BIOS Parameter Block during mounting through ordinary `/dev/fd0`.

Ad 2. Yes, though controller on my 386SX machine seems not to care about it. The mentioned [patent](https://patents.google.com/patent/US5745313A/en) contains a visualisation in figure 5:
[image]

There is no mention of head skew in the patent whatsoever, or in any other documentation I could find, so I  can only assume that it should stay at 0.

Ad 3: Without any configuration at all - FF.CFG nor IMG.CFG, no. However, enabling:
`host = pc-dos`
in FF.CFG is enough to break the detection.
I tested this by doing factory reset, and loading Windows 95 installer without any configuration file present on USB drive. Without any configuration, detection worked, when I put `host = pc-dos` into FF.CFG I started getting "sector not found" errors in host OSes.

keirf, 2021-03-02:
Okay, so (3) should be, broken with `pc-dos`. Ok. I can look into that.

keirf, 2021-03-03:
Do you have a DMF image you can send me that's working except with `pc-dos`? Then I can debug what's going on. It will be something simple.

Leo-PL, 2021-03-03:
I do - how would you like me to share it? I think I could also create one in the meantime, to avoid any copyright-related woes.

keirf, 2021-03-03:
You can zip it and attach it to the ticket. Or upload it somewhere and provide a link. On Google Drive you could restrict it to my email.

Leo-PL, 2021-03-03:
[dmf_test.zip](https://github.com/keirf/FlashFloppy/files/6075043/dmf_test.zip)
Attached. I kept existing filesystem from Win95 DMF disk I used previously, but replaced data with chunk of randomness - detection with `host = pc-dos` still breaks.

keirf, 2021-03-03:
Here is a build with some fixes:
1. It sets the cskew to 3 in default config (no `host=`)
2. It detects DMF and sets interleave=2 and cskew=3 when `host=pc-dos`.

Note you need to be logged into Github to access the link below. You should unzip, then unzip the normal (non-debug) zip within, and then grab the firmware UPD file and update your Gotek with it. Then test.

https://github.com/keirf/FlashFloppy/suites/2170258479/artifacts/44592687

Leo-PL, 2021-03-03:
And works flawlessly for me. Many thanks!

sigurbjornl, 2021-03-03:
Ran across this thread while stumbling with Win98 floppy install images on my OpenFlops.   Didn't work with 3.20, even with the IMG.CFG files but works flawlessly with this update.  Thanks!

commit c5fa3294 (2021-03-04): img: Fix pc_dos layout for DMF (1.68MB)
Fixes #430
commit 963ed184 (2021-03-04): img: Fix pc_dos layout for DMF (1.68MB)
Fixes #430
closed 2021-03-04 by keirf