keirf/flashfloppy #431: ZSK MSC embroidary machine

vinibali, 2021-02-21:
Hi!
A friend of mine has an old ZSK from the first years of the 90s. I was not able to read any code with Gotek's original firmware, however  I tried every combination of the jumpers.
This machine works only with 720K drives, does anybody here had some luck with the default settings?

vinibali, 2021-02-21:
The ideas @keirf mention are working with this device?
[Embroidery Machines #37](https://github.com/keirf/FlashFloppy/issues/37)

keirf, 2021-02-21:
So are you using FlashFloppy or original firmware?

vinibali, 2021-02-22:
FlashFloppy at the moment, the first part of the comment was just some historical information :)

keirf, 2021-02-22:
Is your aim to format/save a new disk. Or do you have existing disks to preserve/transfer?

I would use a double density HFE image and try formatting it on the embroidery machine.

closed 2022-06-22 by keirf