keirf/flashfloppy #440: [FloppyMeister] New reference design requests FF firmware port

bleeptech-nl, 2021-03-02:
Keir, as per our deiscussions, I am designing a new floppy emulator board, called FloppyMeister, and would request a ported version of FlashFloppy running on it.

**General**

- The schematic will serve as open reference design for anyone
- One or more implementations of the he boards will be made available for all in some shape or form

**Specs**

- A new and faster STM32 microcontroller to break past the RAM and Flash barrier. We're looking at the F411 atm.
- Peripherals: USB, (micro)SD, beeper, Oled display, On Screen Display with keyboard input (where possible), rotary encoder, buttons, status leds, expansion pins, ..more..
- Floppy connector signals fully forwarded to the MCU to enable more firmware options

**Status**
Currently drawing the schematics. Peripherals are done, MCU pins have to be determined.
Once I have a full first schematic version I will post it. 
After that a PCB layout will be made and sent off for prototype production. Shipping and soldering included it will be at least some weeks before the fist board is ready to be tested.

Thanks, regards,
-Richard

bleeptech-nl, 2021-03-03:
Ok so here are some pin requirements we discussed

> OSD needs SPI on the output pin; 
MOSI on Green video signal? Cannot share this SPI then I presume.
Can you explain why SPI is needed?

> All floppy inputs should be on same GPIO bank. There are also constraints on pin indexes within the bank, as some inputs are routed to EXTI interrupts, which are grouped 0, 1, 2, 3, 4, 5-9, 10-15 (ie only lower numbered indexes get a unique interrupt).
I count 12 incoming floppy signals. (CHNGRST, INUSE, SEL0-3, MTRON, DIR, STEP, WD, WE, SIDE). Do they all need an EXTI interrupt? Which of those should get a unique interrupt, and which can be shared?

> CSYNC and Amiga KBCLK need a timer channel.
What about VSYNC?

> WRDAT, RDDAT need to be on timer channels.
In which modes are all these timers running? This determines pin choice.

> LCD/OLED needs to be on I2C.
Roger

penfold42, 2021-03-03:
> Ok so here are some pin requirements we discussed
> 
> > OSD needs SPI on the output pin;
> 
> MOSI on Green video signal? Cannot share this SPI then I presume.
> Can you explain why SPI is needed? I assumed this was some DMA output.
> 

You can’t share it unless you use an external output buffer which in turn needs an output enable GPIO pin.
SPI is used to clock the pixels out quickly and accurately. DMA is used to keep it fed.
15kHz 320pixel video needs a pixel speed of ~8 MHz and any jitter is immediately visible.

keirf, 2021-03-04:
Floppy input signals needing EXTI: CHGRST, SEL0-3, MTRON, STEP, WE, SIDE.

Note that four select inputs may be overkill: we could stick with SELA,SELB plus physical jumpers? That would leave us 7 EXTI IRQs required, which just fits (and actually, CHGRST and MOTOR can share, so it could be just 6).

VSYNC also needs an EXTI IRQ, as does CSYNC at the moment, ... Actually if they are on timer channels that may be okay, as they can trigger IRQ too... 

WDAT is in Input Capture mode; RDAT is in PWM mode 1.

I'm not sure I will really pull the OSD stuff into main FlashFloppy. Another option is to simply stick two MCUs on the F4 board for those who actually want OSD. It only adds a few dollars. FF_OSD could certainly be ported off F1 to its own F4, F7 or whatever.

bleeptech-nl, 2021-03-06:
I have investigated drop-in replacing the F105 with the F411; all pins and even all capabilities match except we loose the display I2C, which is replaced by the VCAP1 pin. Given the Gotek pin layout this can only be fixed by both a hard and a software change, which basically means a redesign.

Ok, so I have allocated the MCU pins and optimised to one wich maximises peripherals available while still keeping me fairly happy on the PCB layout side things.
Please verify the following and tell me if something is not right:

I have operated under the folowing assumptions
- it doesn't matter much which timers to use eventhough they are not all the same.
- can use different channels of ther same timer for different functions 
- SPI SS (CD) cannot be hard-tied low to save pins because we don't know which devices require signalling on it

```
EXTInt line -> signal mapping:
0  _SELx
1  _SIDE
2  _STEP
3  _SEL0
4  _WE
-
5  _BTN_X
6  _BTN_R
7  _BTN_L
8  ROT_B (or GPI polling, or Quadrature Timer)	
9  ROT_A (or GPI polling, or Quadrature timer)	
-
10 
11 
12 
13 _CHNGRST
14 _MTRON
15 _SD_CD (or GPI polling)

Timers:
_RD  	TIM3_CH3  PWM out mode
_WD  	TIM2_CH1  Capture mode
_ROT_A	TIM4_CH4  for Quadrature input, or else EXTI or GPI (rotary encoder)
_ROT_B	TIM4_CH3  for Quadrature input, or else EXTI or GPI
BUZZER  TIM3_CH1  PWM out mode, or just bit bang GPO

```

**Notes**
All pins now in use
18 floppy signals, including _CHNGRST
USB OTG 2 wire (Host)
External SPI Serial Flash (optional)
SPI SD card wit _CD card detect low* and detect interrupt (* more prevalent, cheaper and commonly available types. Suggest a config setting).
I2C (O)LED interface + bluepill OSD
I2C 8-port I/O expander 8 (optional) + future expansion (on same bus)
1 generic user expansion pin with TIM1, GPIO and EXTI12
SEL0 and SELX where default is 1 and changeable to 2 or 3 with some jumpers. Will probably require internal pull-ups.
Each emulated drive (SEL0/X) a status LED controlled by mcu so they can for example also be used as active-drive notification
single-Wire debug (which are actually 2) + SW Output
U(S)ART1 port for debug (can obsoleted for SW Output) + OSD commands(RX)/display-out(TX) (see below)
U(S)ART6 future expansion
buzzer out on a PWM timer, or GPO
Rotary Encoder (quadrature output) + select, and 2 seperate push buttons. All on same EXTI block. Will have to figure it out in the ISR.

**Other comments**

> I'm not sure I will really pull the OSD stuff into main FlashFloppy. Another option is to simply stick two MCUs on the F4 board for those who actually want OSD. It only adds a few dollars. FF_OSD could certainly be ported off F1 to its own F4, F7 or whatever.

Well, that is kind of a let down since that was one of the reasons to do this redesign.

I am not a fan of sticking a blue pill on the board. STM stocks drying up, prices soaring a few dollars more is a lot, I have no space on the board for another MCU and on top would also have to add another USB port for upgrades and program twice. I also don't think it is good design to add another MCU where it cone be done with one so I'm not really a fan of that option. 

I need it somewhere though, but I would agree if you would say that OSD on the floppy emu is not the best design also. To be honest I can live with delegating it away off-board to some abstract display/input device that uses USART1 for sending commands and receiving screen updates. 
That would enable me (and others) to maybe stick it on another controller that I already have/is available, or split screen and input to seperate devices, or build an IR remote, hook it up to a PC or whatever without being tied to an I2C source, the blue pill or its firmware.
Yes I realise USART1 is debug now but it can probably be both.  
Can you add that in stead, also on the current firmware so I can start and hook something up soon?  I can make a ticket to discuss the details.

> WDAT is in Input Capture mode; RDAT is in PWM mode 1.

You meant the other way around?

Some FF code comment I saw (forgot which project/file) mentioned 2 PC pins on the floppy bus with different signals on pin 2 and 34. One was DENSEL and seemd to have reversed direction. Does that ring a bell? What is that, and is that of any concern?

Thoughts?

keirf, 2021-03-07:
Sounds reasonable indeed.

bleeptech-nl, 2021-03-07:
I [posted the generic OSD request](https://github.com/keirf/FlashFloppy/issues/445).

Also can you please answer the questions I posted? I need to know if you think the timers and external interrupt lines are good to work with.

keirf, 2021-03-07:
Yes the assignments are good. You are welcome to route through the OSD wires also if you like. I mean, we can have a go at integrating OSD, it's just not guaranteed.

bleeptech-nl, 2021-03-07:
I'd rather have the USART1 generic OSD/input then since, well, it's generic and pretty much a sure thing rather than an iffy. 

I can go about testing with the USART1 soon and when the final board is there we can always still look at hooking it up through the expansion lines if that is still desirable. For those wanting the Blue pill OSD that stays an option too, also on the new board.

bleeptech-nl, 2021-03-10:
I am looking again at the output bufers for the board. I found some that can take a lot of beating but the output is inversed. I did not find an invert-output option on the GPIOs of the STM32 series, so that means a software change.

Is it a problem to invert the outputs in the software (specifically the PWM RD_DATA output)?

The propagation delay of the buffers is 250ns on either flank (2MHz max). Looks still in spec to me. Is that an issue?

keirf, 2021-03-10:
250ns isn't super fast but probably okay. Possibly rise/fall times matter more than propagation. Inverted vs not doesn't really matter, you probably do want an ID pin or three if possible so that features such as inversion can be dialled in per board design.

bleeptech-nl, 2021-03-10:
Only one pin left Im afraid and one isnn't really enough to store multiple configuration IDs in. Can it be done in the config file/storage?

keirf, 2021-03-10:
Yeah possibly.

bleeptech-nl, 2021-03-15:
I notice programming the Gotek requires 5V on the TX pin (method 2, USB).
Why is that?

keirf, 2021-03-15:
> I notice programming the Gotek requires 5V on the TX pin (method 2, USB).
> Why is that?

It's nothing but a flag to the system bootloader to put itself in USB-DFU mode.

bleeptech-nl, 2021-03-15:
So, to make your answer more precise then; USART_TX also has an alternative as USB_OTG_FS_VBUS which is connected to 5V power as a means of USB cable detection, which in turn selects the USB boot.

But [AN2606 ](http://www.st.com/resource/en/application_note/cd00167594.pdf) (14.1, table 27) says this: 
[image]
Which sets USART1_TX to transmission mode **output push-pull**, which is then hard connected to 5V with a jumper..
Which is confirmed:
[image]

From boot version 2.1 and up the TX is internally left as input until it receives the 0xF7 start code on RX to indicate USART1 is being used to boot from. Only then it is configured output. USB is tested first so that is ok.
However, 2.1 starts on F105s with date code 227 or higher, so this is a risk for users of a Gotek with lower date code.
I have a Gotek here with date code 017. If 2010 (got to love those date codes) then I'm shorting the pin when USB programming.

keirf, 2021-03-15:
I never really investigated that errata: Does PA9 get reconfigured as input after USB is initialised? If so this would only be a very short period that PA9 is out of spec. Anyway, it's somewhat moot since there's nothing can be done to fix this issue in affected chips, and noone has ever indicated that their chip has been killed by this erratum (a damaged PA9 IO pad would not be observed by most users in normal Gotek use).

bleeptech-nl, 2021-03-15:
Well, that depends on the version of the boot loader then. Pre 227 date code it has boot loader <=V2.0 where USART1_TX is initially set as output on PA9. This will (may) then cause a short if hard tied to 5V, which the errrata warns for.

In V2.1 (227+) they start out with PA9 as input and check the USB first. After USB they check the USARTS and read RX. I'm assuming that only once they read the USART start code they set TX to transmit output push-pull, since that is the safe solution.

Like you said, if anything gets damaged it is probably only the pin and one can only guess as to how much trickles back as feedback. Still, it may be wise to check the date code first.
Question here is what does the date code mean since it does not include the decade. Mine is a good example. I bought my 017 somewhere in 2020 so how old is the mcu?

As for newer versions of the STMs, I've looked at a few and they don't have the issue.

bleeptech-nl, 2021-03-15:
Have you tried entering USB mode without bringing VBUS/USART1_TX high? 

I cannot find anything about how the STM devices detect an USB host except the reference in that 1 errata, but I do read that USB is detected using enumeration polling on the line and pulling up the data line for detection, and AN2606 mentioned not needing that pullup.
[image]

keirf, 2021-03-15:
> Question here is what does the date code mean since it does not include the decade. Mine is a good example. I bought my 017 somewhere 2nd half of 2020 so how old is the mcu?

Date code is YWW (Year, Week) so that's 17th week of 2020 I would guess. Pretty unlikely to have been hanging around for a decade!

keirf, 2021-03-15:
> Have you tried entering USB mode without bringing VBUS/USART1_TX high?

I have not (actually I never really use DFU myself), and the STM docs are needlessly vague about it, but if you Google around you will find it generally accepted that the connectivity-line bootloader detects USB cable by polling VBUS on PA9, and noone has success unless that pin is HIGH.

closed 2022-06-22 by keirf