keirf/flashfloppy #442: Read and Write Files Empty

masonchamberlain, 2021-03-04:
I have a Milltronics Partner 3 CNC mill I am trying to get FlashFloppy set up with to be able to transfer CNC programs back and forth width. This is an aging industrial machine, the CNC software is a program running inside DOS 6.22.
I was able to determine it uses the IBMPC protocol, which is set up with the jumper. 
I created a .img file with some test files in it and connected it to the machine. The mill will load the contents of the flash drive and show the files I have loaded. When I attempt to import them, the file shows up in the system as empty. The file name is transferred, but the contents of the file have been entirely replaced by spaces.
The same issue happens when I try to copy programs from the mill back to the flash drive. Everything appears to go well from the mill side, but not when reviewing it on the computer side. It transfers the file name, loosing the extension in the process. The contents of the file are replaced by a single line of spaces.

I have tried formatting the floppy image differently and that drastically affects what I am able to do, everything from sector not found errors to crashing DOS and requiring a reboot. I am not sure what to try next.

keirf, 2021-03-07:
What is the size of the IMG file? FAT formatted I assume? This ought to be an easy one since you can reasonably assume the CNC is an embedded PC, it seems.

masonchamberlain, 2021-03-08:
[Tests.txt](https://github.com/keirf/FlashFloppy/files/6102182/Tests.txt)
So far I have been testing a 1.44MB file, as that is what size floppy disc is used. I have been testing FAT12 and FAT16 with different cluster sizes, the results of those are in the attached text file. I have yet to try FAT32 with a 1.44MB image. Pending the results of those tests, I will move on to testing a 1.2MB image size unless there is something I am missing here.
I do know the CNC is using a slightly stripped down version of DOS on whatever hardware it has. Most commands that are not relevant to file structure manipulation have been removed, but it is possible to get to the main DOS shell and run commands from it.

masonchamberlain, 2021-03-08:
I tried using a different program to make the image files, and the combination of that and using a .IMA rather than a .IMG seemed to work.
I wouldn't mind knowing why .IMA works and .IMG does not, but it is useable.

keirf, 2021-03-08:
There is absolutely no difference between IMG and IMA handling in FlashFloppy. Probably it is something else you did at the same time as changing the filename.

closed 2021-03-08 by keirf