keirf/flashfloppy #448: Ensoniq ASR-10 (rack) not receiving Disk Change

jeremybernstein, 2021-03-18:
I'm using the latest example FF.CFG, but my ASR-10 isn't detecting Disk Change events (I have to reselect the Floppy interface each time I insert a new disk in order to force a reload).

My Gotek has only the S0 jumper set, which I believe is correct. How would I go about troubleshooting the problem?

keirf, 2021-03-18:
https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#ensoniq

Note the required setting for interface: `interface = ibmpc-hdout`

jeremybernstein, 2021-03-18:
Yes, I meant that I'm using the latest FF.CFG with the correct additional settings for the ASR-10, as found on the Wiki. The FF is working, I can see and load images. But switching between images requires going to the system menu and reselecting the floppy interface, which isn't expected.

keirf, 2021-03-18:
Are you certain your FF.CFG is being picked up?

Also try changing chgrst setting to `chgrst = delay-3`.

jeremybernstein, 2021-03-18:
Yes, I've tried some breaking changes (interface = shugart, for instance) to confirm this. I also tried the chgrst = delay-3 setting, without success.

keirf, 2021-03-18:
Is your ASR-10 working with a real floppy drive, do you know?

jeremybernstein, 2021-03-18:
The floppy drive worked before I swapped it out, but I only used it for the OS. I'd need to plug it back in and test whether disk changes were being properly handled, though (the ASR-10 should however respond to disk change, in general).

jeremybernstein, 2021-03-18:
According to [this link at HxC](https://torlus.com/floppy/forum/viewtopic.php?t=510&start=75), pin2 should be dens and pin34 diskchg, which I guess I can try when I'm back at my machine. But I'm not optimistic.

keirf, 2021-03-18:
Yes, that is what `ibmpc-hdout` specifies. Hence wondering if your ASR-10 or its cable is working properly. Failures at pin 34 are quite common as it's at the far end of the ribbon connector and receives a lot of stress from removals and insertions.

jeremybernstein, 2021-03-18:
I'll pop the machine open tomorrow and check it out.

jeremybernstein, 2021-03-19:
You were right, pin 34 has no continuity on the cable. Built a new one and it's working perfectly. Sorry for the noise, and thanks for taking the time to help me figure it out.

closed 2021-03-19 by jeremybernstein