keirf/flashfloppy #452: Indexed mode - support more then 1000 images

tazzz, 2021-04-02:
nav-mode = indexed is limited to DSKA0000 - DSKA0999 images.
This is hard-coded in main.c: 
indexed_mode:
...
/* Expect a 4-digit number range 0-999. */
                if ((i != 4) || (idx > 999))
                    continue;
Is there a specific reason for this?  I am trying to have more than 1000 images in indexed mode.  Track 255 direct access mode does support this: CMD_SELECT_IMAGE_INDEX : 0x04 has two bytes parameter or potential 65535 images.
This is not maintainable using manual selection.  But this is  not a problem with direct access

Would it be possible to add a parameter in FF.CFG (something like nav-mode = indexednolimit) to remove the 1000 limit?

keirf, 2021-04-02:
The limit was mainly because the original 3-digit display, which I assume is main target for indexed mode, can only display up to 999. Also there's a slot_map[] that gets populated, and that's fixed size. Maybe that could be dealt with....

keirf, 2021-04-02:
What's the actual usage scenario anyway: file selector software? There may be other interfaces you could use such as CMD_SELECT_NAME.

tazzz, 2021-04-02:
It's on the BBC micro/master: indexed mode with shurgard - indeed for file selector (*DIN)

tazzz, 2021-04-02:
Did not found a lot on cmd_select_name but from what I can tell from da.c
it relies on 512 byte sector size. This is not supported as far as I can
tell on the BBC.

keirf, 2021-04-02:
The da interface is hardcoded for 512-byte sectors. Somehow the Beeb must already be writing 512-byte sectors.

tazzz, 2021-04-02:
Just compiled from git flashfloppy with these two small changes in main.c
34     uint8_t slot_map[10000/8]; /* changed from 1000 - this could go up to 65535 but not sure about gotek memory */
1843 if (i != 4)  /* took out the idx check on max 999 */

Did some quick tests (with 1200 ssd's) and it just works : on startup it recognizes the 1200 discs and *din works as expected (*din 1200).  The display I have gets truncated (1200/120) but that is not such a big deal.
Thank you for the pointers.

iancourt, 2021-06-20:
Would it not be possible to have 100 ‘banks’ of 1000 disk images?

e.g. User can select ‘b00’ to ‘b99’ on the three digit display first, then select disk ‘000’ to ‘999’ beneath?

Not sure how feasible this might be, just an idea. 😊

closed 2022-06-22 by keirf

richard-broadhurst, 2024-01-24:
Does FF now support indexed mode in sub folders, if so, does it just search the whole disc for them?
Otherwise how does it get part the format limit of 510 files in the root directory?