keirf/flashfloppy #453: Gotek with AT32F415RCT7 Instead of STM32F105

andr229k, 2021-04-03:
I was not sure, if was going nuts till I discovered, that a vendor delievered a Gotek to me with a AT32F415RCT7 chip, instead of the normal used STM32F105. A user on Facebook mentioned that STM is having micro shortages, which may effect more Gotek devices in the future. Keir mentioned that these chips are 32k ram, which is lowered than the STM32F105.

Picture of chip:
[image]

I made a short guide as of how to write the flashfloppy firmware to these deivces:
[How to flash AT32F415RCT7.pdf](https://github.com/keirf/FlashFloppy/files/6253324/How.to.flash.AT32F415RCT7.pdf)

---------------
Gotek Installed with Flashfloppy
S: 7-Segment display is blinking FF or showing a code 23 also with a USB-stick inserted with .hfe files as indexed. OLED shows snow only.
D: The PCB-design looks similar, but is using a AT32F415RCT7 instead of STM32F105 due to the STM micro shortages
A: Provided information on the Facebook group and opened this issue ticket.
N: Investigate, if some changes to the firmware can be made in regard to ram usage and tests of functionality.

keirf, 2021-04-03:
I may be able to reduce RAM usage to 32kB and provide a test firmware for you.

andr229k, 2021-04-03:
> I may be able to reduce RAM usage to 32kB and provide a test firmware for you.

Perfect, Im ready when you are :)

omikron88, 2021-04-03:
Did you see Pi Pico, it is great toy for project like FF

keirf, 2021-04-03:
Here is a quick attempt at a 32kB RAM firmware. Download the artifact linked at the bottom of the linked webpage (you need to be logged into github!): https://github.com/keirf/FlashFloppy/actions/runs/715182292

Use the HEX file in the non debug zip file contained within. My guess is that there will be further issues than just RAM.

keirf, 2021-04-03:
If any of these devices are still on factory firmware it might also be interesting to know if that seems to work correctly. More of a pain to set up, however, and you'd probably need an IBM-compatible host to test against.

andr229k, 2021-04-03:
> Here is a quick attempt at a 32kB RAM firmware. Download the artifact linked at the bottom of the linked webpage (you need to be logged into github!): [...]

LED is working with this one, OLED is just blinking.

I have not tested it in use yet.

bleeptech-nl, 2021-04-04:
Here is the datasheet.
http://www.arterytek.com/download/DS_AT32F415_V1.04_EN.pdf

Problem with these clones (even the dataheet is a clone :D) is that you never know which peripherals are cloned 100% or even if they stay that way in future batches. Ive taken a look at a few and I dare not use one for a gotek replacement. Other than that, the cones are getting scarce too. This one you cannot even get at LCSC.com. It's just not known, neither on octopart.

STM32 mcus themselves are near impossible to get now, especially anything with 64kB RAM or above. That may have triggered this.

As for the PCB being similar; I believe the STM32F415RC is a drop in replacement for the F105 at least pin wise, so a good clone would be too, so question is how good is it?
Binary compatibility depends on what's used by the code and if that changed. Most stays the same but not everything.

It's a strange move. A quick look at the vendors for the Goteks shows stocks in the thousands up to 100,000 pieces [available](https://www.aliexpress.com/item/32287025620.html?spm=a2g0o.productlist.0.0.75d35854bBu3KW&algo_pvid=7f947b0c-4487-4b0a-939c-13cce7c1779e&algo_expid=7f947b0c-4487-4b0a-939c-13cce7c1779e-5&btsid=0b0a01f816175407665041522e6bd0&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_) so if they have that many, and I've seen those numbers before, why the need to put another MCU on?

keirf, 2021-04-04:
It's unknown right now how prevalent this may be. If it's an easy fix and this specific part is common, perhaps it will get supported. It certainly reads the USB stick and drives the 3-digit display okay. The timers I guess will be okay. The I2C peripheral may have differences, as may DMA (after all this is a F4 clone, not really F1, so some peripherals are expected to differ anyway). The halving RAM is quite unfortunate as 64kB was quite tight for track buffering already. No doubt some higher-density formats with slower sticks will be unreliable.

omikron88, 2021-04-04:
What about to build our own "Gotek"
A tiny board with the Floppy buffers, few buttons and cheap Arduino display.

The question is. MPU candidate.

What about Pi Pico as it has 256 KB RAM and huge Flash.
But do not know if an external program Flash, with some waits will not slow
down it too much.

bleeptech-nl, 2021-04-04:
I'm designing one, will probably be around 5x6cm PCB. That's about as small as you can get, unless you start throwing out connectivity and expansion, but of course you can go bigger.
A big issue is indeed the MCU with the current scarcity. Things are bound to pick up again in the future though because there obviously is a market for them.

Not sure Pi pico will work with Cortex M0, 28 GPIO, 1 timer, and USB 1.1. Also it's a ready made board so that means you have to solder stuff together whatever you do.

andr229k, 2021-04-04:
What is important with this topic is not look at the alternative expansions, but how we can be prepared for usage of lower spec MCU like AT32F415RCT7 in the future. I have therefore today shipped one of these to Keir for further testing.

I was asking myself following:
•	How would Jean handle this with his HxC-firmware as the flashing-bridge won’t be able to write to this? I guess his firmware is also too hard on 32kb?
•	Are these Gotek all coming from the same place as the Factory firmware is not available to be downloaded anywhere?
•	How do you identify in advance, if your seller is selling you a Gotek with AT32F415RCT7 or STM32?

I contacted seller and asked for an explanation and he forward my inquiry to his Manufacturer.

keirf, 2021-04-05:
You can be sure HxC will also need modification for these chips, as it certainly uses available RAM as block cache if nothing else. I couldn't say whether the job is harder than for FlashFloppy. It's probably similar.

GOTEKsystem, 2021-04-06:
The price of stm32f105rbt6 has increased 10 times, and now we are running out of stock, so we have to replace it with at32f415cbt7, which can maintain high cost performance.Because most of our applications are on other devices, customers directly use the functions of our own firmware.
At32f415rct7 is 64 pin, but it's hard to get too. Now we mainly use at32f415cbt7, which is 48 pin. All the models that use AT still keep the original functions. But the object code can't be used directly and needs to be changed.  ST is cortex-M3, AT is cortex-M4, But I believe it should be very easy for keirf.   It took us only a few days to complete the modification and test. About ram, AT only has 32K RAM, but our application actually uses less than 25K, You need to compress the use of ram, because AT's chip with USB host only has 32K ram maximum.
We have recently made 2 new model, one is the model with 128 * 64 OLED screen, and we modified the shell to adapt to the screen, the other is the model of floppy emulator with USB device function for transfer data with PC. visit goteksystem.com.
If you have any other necessary hardware adjustments, please let me know. We can add them in the next update.

keirf, 2021-04-06:
@GOTEKsystem Thanks for your support. I will send you an email, but my brief thoughts are:
1. The new 100LQD model with OLED and rotary encoder looks excellent! This covers the main hardware modifications requested by users. I assume that this will use the LQFP64 MCU, and that the usual programming/debug header remains available (so that programming via serial or DFU is possible?)
2. For more basic models, the programming/debug header is required for successful programming of alternative firmware via serial/DFU methods. So for example, model U144K has been a problem, as it has a different PCB and serial pins are not on any header.
3. Next most important is a J7 header, so that users can retrofit a rotary encoder. Again, missing from U144K. This may be less important if your 100LQD is a good price and readily available.

keirf, 2021-04-09:
I have two tickets tracking this issue, so I am closing this one in favour of #454

closed 2021-04-09 by keirf