keirf/flashfloppy #454: New GOTEK with AT32F415CBT7

GOTEKsystem, 2021-04-06:
Over the past few months, because the price of ST chip(STM32F105RBT6 Cortex-M3) has increased greatly, and We're running out of stock, we have changed the design of most models and replaced them with ARTERY chips(AT32F415CBT7 Cortex-M4).
        AT chip supply is relatively stable, the product made of AT chip can maintain the original cost performance.
        At present, we still keep some models that still using ST chips, (example: SFR1M44-TU100K, SFRM72-TU100K, etc), but we are not sure how long it will last, because the price of ST chip continue rising. 
        If you insist on buying the model using ST chip, you can order from the seller to produce the model still using ST chip. The price will be relatively high, and the production cycle will take about 2-4 weeks.
        In the future, if the price of ST chip is reduced and the supply is stable, we will continue to produce the model using ST chip.
        For customers who want to DIY by changing the firmware of our products, please note that you should confirm with the seller and purchase the model using ST chip. 
        Another better way is to contact your DIY firmware developer to change the software to apply to AT chips. We can provide you with necessary help to solve the problem of software transplantation as soon as possible. Please contact me by email guiie@163.com.

Snocksman, 2021-04-06:
OK, it seems that i bought a Gotek Drive with an Artery Chip. The Chip on my PCB is labeled AT32F415RCT7. Does that mean, that I can´t use FlashFloppy with my Gotek Drive ?

keirf, 2021-04-06:
@Snocksman Yes until FlashFloppy is updated with support. The chip has 256k Flash but only 32k RAM.

Snocksman, 2021-04-06:
OK, thats bad news... But will FlashFloppy be updated for those Devices (I hope yes !!!) ?

keirf, 2021-04-06:
Yes the writing is on the wall for stm based goteks. I can't really see the factory going back if Artery MCU prices are lower.

Snocksman, 2021-04-06:
OK, than I will sit back and wait for an Artery ready Firmware release ! Where can I donate for a Coffee ?  ;D

tizzulo, 2021-04-07:
Came here to mention this as well, I received some Artery based drives today as well and noticed the target looked different in the ST Flash tool.
If it's any help, here is a photo of the new PCB.
[image]

Snocksman, 2021-04-07:
@tizzulo Maybe it´s better to flash this Chip with the Artery Flash tool you can download here: https://www.arterytek.com/download/Artery_ISP_Programmer_V1.5.46.zip (as mentioned in Issue #453)... With this Flashtool & Driverpackage you can use the front USB of the Device to flash the Chip (without USB to TTL).

RetroClinic, 2021-04-07:
Oh dear, where are the headers for the rotary encoder? good to see it's got a different part number though, so it can be identified without removing the lid.

Snocksman, 2021-04-07:
Hey, cool... In my PCB Version the rotary encoder connectors are still there !!!  :D

[image]

RetroClinic, 2021-04-07:
> Hey, cool... In my PCB Version the rotary encoder connectors are still there !!! :D
> 
> [image]

Hmm, that's the standard SFRC922D PCB, originally for the ST chip, and the later one is SFRC922AT3, so looks like there are two PCB variants being produced :-(

tizzulo, 2021-04-07:
> @tizzulo Maybe it´s better to flash this Chip with the Artery Flash tool you can download here: https://www.arterytek.com/download/Artery_ISP_Programmer_V1.5.46.zip (as mentioned in Issue #453)... [...]

Thanks for the suggestion. Unfortunately, after flashing the [test build of the firmware](https://github.com/keirf/FlashFloppy/issues/453#issuecomment-812928138) shown on that issue, my SFRC922AT3 unit is showing "uPd" on the screen, just like the one I flashed with USB-TTL.

tizzulo, 2021-04-07:
I am more than happy to continue to test new firmwares on my units, btw, since I don't have anything better to do with them until then!

Snocksman, 2021-04-07:
Ok... I tested the testbuild, too. For me it worked partly... Some adf loaded fine  (giganoid for example), but some did not and my Amiga stopped loading. But I had no uPd message.

keirf, 2021-04-07:
The PCB variant is to allow use of a LQFP48 chip. I think this is preferred as it's surely cheaper.

andr229k, 2021-04-09:
For the overview I have here uploaded the different chip variations from AT, I have also received a few AT32. Those I have received are marked with red, the one that will be common as mentioned by GOTEKsystem is marked with green

[image]

GOTEKsystem, 2021-04-09:
Yes, we produced 200 pieces with 64 pins, but the model is 415rct7, not marked red 415rct7-7. Did you get a 64 pin package of the same size as 105rb? Can I have a look at the pictures? However, this kind of package is also difficult to buy in large quantities, so we now use 48 pin for mass production。

andr229k, 2021-04-09:
> Yes, we produced 200 pieces with 64 pins, but the model is 415rct7, not marked red 415rct7-7. Did you get a 64 pin package of the same size as 105rb? Can I have a look at the pictures? However, this [...]

I got these two:

Left = AT32F415RCT7
Right = AT32F415RBT7

[image]

GOTEKsystem, 2021-04-09:
Yes, right, we are now using the right one, 48 pin, model 415cbt7, in mass production，This model is in stable supply。

keirf, 2021-04-09:
The issue with the 48-pin model is: Where do we connect a rotary encoder? As it has no header that we would normally use for this.

@GOTEKsystem Do you have any plan yet which pins you would connect Rotary DAT,CLK,SW to on the 415cbt7?

bleeptech-nl, 2021-04-09:
I can imagine them removing the rotary pins if they plan to release their own emulator with on board rotary encoder.

as for the IC models;
CBT7 = is LQFP 48 pin package, 7x7 mm (on Tape)
RCT7 = LQFP 64 pin package, 10x10 mm
RCT7-7 = LQFP 64 pin package, same thing but in a 7x7 mm (non standard) size

@GOTEKsystem Did you use a new pin configuration for the LQFP48 or is it still the same as the LQFP64 (in other words, do you use the same firmware)?

bleeptech-nl, 2021-04-09:
@andr229k Can you provide us with clear (with enough lighting) hi-res pictures of both the Top and Bottom of the 48 pin board?
Maybe can build a new schematic and see what we're dealing with.

keirf, 2021-04-09:
@bleeptech-nl The firmware will be modified as some pin assigns are different. Eg. the LQFP64 button pins don't exist on LQFP48.

andr229k, 2021-04-09:
> @andr229k Can you provide us with clear (with enough lighting) hi-res pictures of both the Top and Bottom of the 48 pin board?
> Maybe can build a new schematic and see what we're dealing with.

Yes I can friend, please see below. I have attached pictures of both: [I dont really have a hi-res camera, so it was to best effort]
[image]
[image]

GOTEKsystem, 2021-04-09:
I saw it, on 64pin ,  PC10-DT ,  PC11-CLK, PC6-SW,  but on 48pin, there is no portC, I changed it to PF6(pin35)-DT, PF7(pin36)-CLK, PA6-SW, NOTUSE-PA15, The next version of PCB, SFRKC30AT3, I added a 6-pin interface, which is used to connect the rotary。
This version of PCB, we will also use in SFR1M44-U100LQD, which is a model with OLED screen and rotary。

keirf, 2021-04-09:
@GOTEKsystem Many thanks, the full header will be very helpful!

For this initial 48-pin revision I think Rotary will be discounted or require direct soldering to the MCU(!) -- Resellers had best save existing reserves of 64-pin versions for their Rotary-modded drives!

Would anyone care to send me an existing 48-pin model?

keirf, 2021-04-09:
Here are my thoughts on supporting the plurality of Gotek models (STM32F105RB, AT32F415CB, AT32F415RC). The different Cortex cores (M3, M4) are going to require separate firmware builds anyway. So I propose to introduce a separate build for each of the above, and then merge them into a unified update (UPD) file. The update bootloader will be modified to select the correct build from the UPD file. So from user point of view: All Gotek models are treated the same for update, and the different CPUs, pin assignments, etc will be invisible.

I will continue to provide an 'old' Gotek UPD file as existing bootloaders in the field of course will not understand the new unified format.

There will be increased complexity for initial firmware flashing (serial, DFU, stlink) as each build will have a separate HEX file. Of course resellers are perfectly capable of identifying the MCU -- physically or possibly via DFU/serial-reported chip id -- and using the correct HEX. I will also have to document the HEX files in the wiki, perhaps with identifying PCB photos.

The benefit of dealing with this upfront is that we can get an initial port done to AT32F415 and this can be published to end users without fear, as long as we continue to have a **simple** story for end-user updates in the field.

bleeptech-nl, 2021-04-09:
@andr229k Thanks, but can you also picture the **bottom side** of the 48 pin board? Need to see the traces on that side :)

GOTEKsystem, 2021-04-09:
In fact, on sfrc922at3, the port we use to download code has two pins pa13 and PA14, which can be used as DT CLK and ja-pa5-sw. I don't know if these two pin are occupied in other changes。Of course, this brings trouble to the code version, but these pins are not occupied on sfrkc30,

GOTEKsystem, 2021-04-09:
[image]
[image]

andr229k, 2021-04-09:
> @andr229k Thanks, image quality is good, but can you also picture the **bottom side** of the 48 pin board? Need to see the traces on that side :)

[image]

bleeptech-nl, 2021-04-09:
Thanks @GOTEKsystem and @andr229k :)

keirf, 2021-04-09:
> In fact, on sfrc922at3, the port we use to download code has two pins pa13 and PA14, which can be used as DT CLK and ja-pa5-sw. I don't know if these two pin are occupied in other changes。Of course, this brings trouble to the code version, but these pins are not occupied on sfrkc30,

How many at3 units will there be? Or: For how long do you expect to ship them before kc30 replaces it?

I could do a separate build for at3, or perhaps allow configuration for at3 via FF.CFG config file to override the default pin assignment (which will be for the kc30 pin assigns).

bleeptech-nl, 2021-04-09:
>  or perhaps allow configuration for at3 via FF.CFG config file to override the default pin assignment (which will be for the kc30 pin assigns).

Good idea to expand this to all firmware supported models. Let the user determine the model in the config. It will free up the 4 configuration pins on the F105 enhanced model too, and allow for things like active low and active high SD card detect and inverted outputs.

andr229k, 2021-04-09:
@GOTEKsystem I’m in contact with the seller who sold me these with 48-pin. I have explained the seller the situation and we came to an agreement, where I can receive a full refund by returning them.

I would highly recommend to advise your re-tailers to be very "exact" about what they are selling, because this change may lead to some misunderstandings between your retailers and their buyers in the future.

GOTEKsystem, 2021-04-09:
Because the price of st105 is very high now,and hard to get， and the production cost has exceeded our selling price, So now we have changed  to use 48 pin chips, and even the 64 pin chips of at415 are not available, This situation will continue for at least three months， If FlashFloppy can support 415 48 pin chips, I don't think it's a problem。 Because if you return the product, it is a loss of time and cost for both，Most of our products  purchased by customers are directly using our own firmware, on their machine，So the seller may not know whether it is suitable for you to  DIY with FlashFloppy。We have also reserved some ST chip models, such as sfr1m44-tu100k and sfrm72tu100k, which will be more expensive. In addition, next time you can order model with board sfrkc30at3 and we reserve the interface. We will probably start to use this board next month。

tizzulo, 2021-04-09:
> 
> I could do a separate build for SFRC922AT3, or perhaps allow configuration for SFRC922AT3 via FF.CFG config file to override the default pin assignment (which will be for the SFRKC30AT3 pin assigns).

My opinion on this matter would be that whichever is less trouble for you is preferable....

Because this different chip with the 32kb RAM seems to be intended to be only in production for ~3 months, and the supply goes back to the old MCU there wouldn't be a huge number of them out in the wild, comparatively.

It seems to me that it's not a clear choice because we're not certain if this will end up being a stopgap measure, and there will not end up being a large number of 32k Goteks sold comparatively - in which case there's no problem at all with a specific build for these low RAM units  - or if we foresee this supply issue potentially lasting longer than intended, or wonder about a problem like this happening again in the future. Though, in either case, I would be willing to bet the supply issue resolves and these end up being an unusual bunch, in which case a specific build would probably make more sense than having to worry about 32k compliance in future versions.

Snocksman, 2021-04-09:
As one of the people having one of these 32k Goteks I would prefer to have them integrated into the master branch... In the other case these devices would surely end up in an no longer updated and supported "32k branch" ...?

andr229k, 2021-04-09:
> As one of the people having one of these 32k Goteks I would prefer to have them integrated into the master branch... In the other case these devices would surely end up in an no longer updated and supported "32k branch" ...?

We are rocking the same boat, I have send one of these boards to @keirf for investigation. Im sure more people will in the near future run into same issue. I also informed Jean (HxC) about this and im sure he is also cooking on it, else he wont be able to provide his HxC licenses :)

keirf, 2021-04-09:
There's no reason to believe that volume production of Gotek will return to STM32F105 even if availability improves. The 32kB AT chips are sure to remain cheaper, and are sufficient for factory firmware. You may be able to special-order the STM chip, but small-volume production with a more expensive chip = more expensive Goteks.

Snocksman, 2021-04-11:
I´m still testing around a bit, with the Test-Version from Issue #453... The 7 segment Display that came with the Drive works fine, but the OLED Display looks like in the attached Foto. Is this a side effect of the Artery Chip, or do i have to confige the Display in some way...?
[image]

andr229k, 2021-04-11:
> I´m still testing around a bit, with the Test-Version from Issue #453... The 7 segment Display that came with the Drive works fine, but the OLED Display looks like in the attached Foto. Is this a side effect of the Artery Chip, or do i have to confige the Display in some way...?
> [image]

with @keirf last 32 kb ram FF build I get following reaction/result with OLD and 7-segment LED:

**OLED:**
https://user-images.githubusercontent.com/52201354/114299599-7efd7100-9abc-11eb-9e94-0147254a83c7.mp4

**7-Segment LED:**
https://user-images.githubusercontent.com/52201354/114299728-309ca200-9abd-11eb-8ee9-c3439306013f.mp4

Did you test it in use yet?

I have a Emu ESI-4000, which I want to test but not before we are able to confirm, that OLED and Rotary can be implemented, if that is not possible with the 48-pin chip, then all my drives with AT32 will be returned to retailer and probably @GOTEKsystem as result of insufficient change management with some kind of requred flashreport depending on how they have structure their supply-chain.

keirf, 2021-04-11:
The OLED problem will be incompatibility in the I2C or DMA peripherals, or in the DMA routing matrix, or something like that. I doubt it will be a biggie to sort out.

I need a 48-pin Gotek to support it. It doesn't need to be @andr229k who supplies it. I have another lead via email I will chase up who will courier me one on request, I just hope theirs are 48 pin. Seems likely.

keirf, 2021-04-11:
The other thing is I'm going to bring up the new chips on master branch which is future v4 series. I'm thinking that there's a fair bit of work to get 32kB chips working *well* (especially for writes) across all supported image types. This effort may be better spent releasing a v4.0a with AT32 support, and dealing with bug reports efficiently. This can work okay if I get the firmware update plan sorted, as users will be able to update to better firmware versions, and it means resellers can more aggressively send out these new Goteks.

RetroClinic, 2021-04-11:
> I need a 48-pin Gotek to support it. It doesn't need to be @andr229k who supplies it. I have another lead via email I will chase up who will courier me one on request, I just hope theirs are 48 pin. Seems likely.

I've sent you a Facebook PM.

Snocksman, 2021-04-11:
@keirf I don't want to be impatient, but what do you mean when we will get something to test ?

keirf, 2021-04-11:
I'll probably get a run at it Tuesday to Thursday this week and that may be enough to get a build out. And an actual (beta) release a week later if all goes smoothly. These are optimistic estimates.

Snocksman, 2021-04-11:
Hey, but thats much faster than i thought... :D  I thought it would take several months. To get something in the next two weeks (maybe) is really cool !!!

keirf, 2021-04-11:
I've got a 48 pin variant arriving Tuesday by the way. Kindly offered by someone in the UK.

keirf, 2021-04-11:
> Hey, but thats much faster than i thought... :D I thought it would take several months. To get something in the next two weeks (maybe) is really cool !!!

It definitely won't be months.

bleeptech-nl, 2021-04-11:
I made a schematic for the AT32F415CBT7 SFRC922AT3 Board for general reference. It should be complete and correct, any errors, please let me know and I will update.
[image]

I wonder how long they will keep operating though; The _SEL, _STEP and DIR signals are pulled up to 5V but the respective mcu pins are **not** 5V tolerant.
[image]
(Note; above are absolute maximums, normal operating max voltages are VDD+0.3V=3.6V, 5.5V for FT pins)

keirf, 2021-04-11:
This is true of every Gotek in existence. The result is that these pins sink a few milliamps via their clamping diodes. Out of spec but they survive.

keirf, 2021-04-11:
Nice schematic by the way!

andr229k, 2021-04-11:
HxC just added support for the AT32:
https://torlus.com/floppy/forum/viewtopic.php?f=33&t=4020

simonsunnyboy, 2021-04-13:
Side note only, as we use STM32 Cortex based devices in 100000s of devices per year ar my employer, our parts procurement tells me, they don't know about shortages.

ST seems to service mass customers at this moment who really buy 100000s and millions of devices.

andr229k, 2021-04-13:
> Side note only, as we use STM32 Cortex based devices in 100000s of devices per year at my employer, our parts procurement tells me, they don't know about shortages.
> 
> ST seems to service mass customers at this moment who really buy 100000s and millions of devices.

If I understand @GOTEKsystem above correctly, then is rather a question of profitability being lowered due to Increasing price on the MCU as the issue is here.

keirf, 2021-04-13:
Made a start today. I2C (OLED,LCD,OSD) is fixed. That plus shrinking RAM usage might be it. I'll give the floppy emulation itself a kick around this afternoon.

tizzulo, 2021-04-13:
> I made a schematic for the AT32F415CBT7 SFRC922AT3 Board for general reference. It should be complete and correct, any errors, please let me know and I will update.
> [image]
> 
> I wonder how long they will keep operating though; The _SEL, _STEP and DIR signals are pulled up to 5V but the respective mcu pins are **not** 5V tolerant.
> 
> [image]
> (Note, above are absolute maximums, normal operating max voltages are VDD+0.3V=3.6V, 5.5V for FT pins so in that regard they are even further out of spec)

This made it pretty easy to find the solder points for the rotary knob on the 48 pin unit ;)

[image]

keirf, 2021-04-13:
The easiest soldering points are at the unpopulated programming port, near the floppy power header.

Snocksman, 2021-04-14:
> 
> 
> Made a start today. I2C (OLED,LCD,OSD) is fixed. That plus shrinking RAM usage might be it. I'll give the floppy emulation itself a kick around this afternoon.

I'm curious ... How did the test go ?

keirf, 2021-04-14:
> I'm curious ... How did the test go ?

Didn't happen. It's taken a while to fit in the button remapping, and I'm currently dealing with the rotary remapping to the programming port pins. With that done, I can test across a bunch of Gotek models and then see how floppy emulation goes.

[43 more replies]