keirf/flashfloppy #456: eilasung FDD-UDD U144K display pinouts?

dpaschal, 2021-04-12:
I am trying to find the correct pinouts on this board to be able to connect up a OLED display.  I have de-soldered the old 7-segment display.  The old display only shows GND.  I need to know which pads to connect the VCC, SCL and SDA pins to.  I found a two year old issue and someone had posted a photo of the display connected up but when I zoom in, it is very blurry and I can't make out which pins go to which pads on the board.  I will attach that image below.
[image]

dpaschal, 2021-04-12:
Also, if someone could help me with the rotary encoder pin-outs and speaker pin-outs, I should be able to complete the project.

dpaschal, 2021-04-13:
[image]
[image]
For anyone that may want to also know this information in the future, I have figured out the display pin-outs.  Observe the photos.  Reading from right to left with the first pin closest to the USB port, it should read as:
GND,VCC,SDA,SCL

Now to figure out the rotary encoder and speaker pin-outs...

GOTEKsystem, 2021-04-13:
This version of PCB has stopped production

https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility
PCB SFRC922D/SFRBA16D STM32 64PIN J7(PB10, PB11, 3V3 GND) JK(PC10,PC11) (Small supply)
PCB SFRC922AT3 AT32 48PIN J7(PB10, PB11, 3V3 GND) (Mass production)
PCB SFRKC30AT3 AT32 48PIN J7(PB10, PB11, 3V3 GND) JK(PF6,PF7) J8(PA6, PA15, 5V GND) (Mass production started in May)
J7 for 3digits LED/OLED/LCD J8/JK for rotary
We will have the new model SFR1M44-LQD with 0.96‘ OLED and rotary in May，visit goteksystem.com

dpaschal, 2021-04-13:
Thanks. I will purchase one of the new units. For now, since I have purchased two of these units and all the hardware to mod them, I am hopeful for collaboration to get these two modded.

keirf, 2021-04-13:
I may add support for this as part of supporting the new 48pin Artery MCU boards. Since they too have no rotary header.

dpaschal, 2021-04-13:
> I may add support for this as part of supporting the new 48pin Artery MCU boards. Since they too have no rotary header.

Thanks @keirf.  For clarity, are you saying that I cannot connect a rotary encoder to the Eilasung FDD-UDD U144K board?

keirf, 2021-04-13:
Yes there's no header for the pins currently used for rotary encoders

dpaschal, 2021-04-13:
@keirf I realize there are no pins, but are there any solder points on the board that could be used for the CLK, DT, SW, + and GND signals though?  

Unfortunately this is the only page with **some** information that I have found.  It's very sparce on this board but I am just trying to work with what I have.

dpaschal, 2021-04-14:
With the help of a friend.  I have been able to figure out how to hook up the rotary encoder and the speaker.  I am including a diagram with colored areas on where to hook up the encoder.

[image]

From the email that he sent me:

Reference [Keirf rotary encoder](https://github.com/keirf/FlashFloppy/wiki/Hardware-Mods#rotary-encoder) image, that shows green, blue and red wires for the rotary. You'll see where those 3 colors are hooked up in the picture. I used the same green, blue and red color to denote. You can connect GND to the same GND as the OLED, and same for VCC/3V3, but look at the reference to when the orange dotted line is needed.

The swiggle I drew shows where pin one is on the LQFP64 chip, it is the dimple on the chip. There might be other circles on the chip, but the dimple that is the smaller hole denotes pin 1. To make sure you have the orientation on the chip right, check ohm between 48/VDD_2, 32/VDD_1 and 64/VDD_3, all these should be connected, 0 ohm to each other.

For the speaker, one wire goes to GND, the other goes to 16/PA2, circled in black on my attached picture.

Microscope is nice for this, magnifying glass at minimum. For the pins on the chip, they might be pulled to a via, so then it's much easier to solder to the via instead of directly to chip pin.

I will be uploading the completed photos once I get it all together and test.

dpaschal, 2021-04-14:
My plan going forward is to connect to any VIAs that I can find.  Once I have that all done, I will do a photo of the board with the connections diagrammed properly on the photo.  More to follow as I go through the process.

dpaschal, 2021-04-14:
Here is a photo of where you can solder the leads for the rotary encoder and the speaker.  Enjoy!  @keirf you can close this one now.

[image]

closed 2021-04-15 by keirf

quantum8, 2021-06-23:
Just to add to this, there's no need to solder to vias for the speaker. I've traced the connection out to J3 header and works great on mine.

J5 appears to be connected to rotary SW but I haven't tested yet. 

Unfortunately it looks like DT and CLK aren't connected on the PCB to anywhere.

elTwardy, 2022-11-29:
I modified my FDD-UDD U144K with an OLED Display a Rotary encoder and a speaker. It's working very fine on my Amiga. I want to use it also on my Atari 1040STF but can't find any information about setting jumpers to get it working. On Amiga is working for me as DF0 (jumper on J2) or DF1 (jumper on J1) but I can't run it on my Atari. After connecting to Atari and starting TOS there is clear desktop no one's drive icons drive visible neither A neither B. I connected it on twisted FDD cables. 
What jumpers I need to set to get it working on my Atari STF ?
FDD cable should be full twisted or partialy (like on PC) ??
Also maybe someone know how to set DF2 and DF3 on this for Amiga?
[image]
[image]
There are to hex files on latest firmware relase:
**flashfloppy-at415-st105-3.37.hex
flashfloppy-at435-3.37.hex**
Wich one I should flash on?
(now I have flashed flashfloppy-at415-st105-3.37.hex)

jojo-gfx, 2026-01-13:
Also on Amiga you don't need DF2, DF3 setting, because this is changed by external controler boards automatically.
So DF0 and DF1 is fine! :)