keirf/flashfloppy #460: STM drivers in DFU mode

roses2uk, 2021-04-24:
Trying to flash FF on a new Gotek, pins 1-2 and 3-7 are jumpered, connected to Win 7 Ultimate x64PC with USB-A to USB-A cable. In Devices and Printers it is showing as unrecognised DFU in FS mode, yellow triangle, cannot find the correct drivers to update it anywhere online. So frustrating, did this 18 months ago (on  a different PC) and it all went like a breeze.

keirf, 2021-04-24:
Latest Goteks are using a different microcontroller. See here:
https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility#update-april-2021-artery-microcontrollers

They require different programming software (and driver)

closed 2021-04-24 by keirf

roses2uk, 2021-04-24:
> Latest Goteks are using a different microcontroller. See here:
> https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility#update-april-2021-artery-microcontrollers
> 
> They require different programming software (and driver)

Mine is SFR1M44-U100K. Is that also affected?

keirf, 2021-04-24:
Look at what the PCB says. Or open the Gotek and look at what is written on the main chip.

roses2uk, 2021-04-24:
Already had it opened up to do the jumpering.

PCB says SFRC922AT3

main chip says AT32F415CBT7 although the writing is so small it's difficult to be 100% sure

So you think I've bought a Gotek which cannot be flashed by any of the usual means??

keirf, 2021-04-24:
There is a link on the wiki to instructions to follow for the usb dfu method on at32f415 microcontrollers. That's the only documented method right now.

roses2uk, 2021-04-24:
This?? http://www.arterychip.com/download/RM_AT32F415_EN_V1.02.pdf

roses2uk, 2021-04-24:
OK great, that seemed to work, very similar to the Dfuse method I used 18 months ago, with the difference being you have to delete the old firmware first.

Got to the stage where I can see F-F on the stock 3 character display (it's for a Roland S-760 sampler by the way - my previous Gotek was for a W-30).

At the moment, I cannot get it to boot up from a FAT32 formatted USB drive, with the S760224.OUT system disk image renamed to 000.IMG in the root directory, but that's my problem for another day.

I have a 0.91" OLED screen, but unfortunately have no Dupont cables to make the necessary connections, and the factory connectors are a 4-way block which I think cannot be separated easily. Have to wait till next Sat to get to a local shop to get what I need.

Thanks Kier. This is important information, and as I am the Admin of a W-30 user's group, I need to inform people who I know are in the process of buying Goteks in case I advise them on the wrong models.

keirf, 2021-04-24:
No problem. These new chips are basically unavoidable.

roses2uk, 2021-04-24:
Really sorry to bother you again.

I think the problem is the jumpers. Llama Music say to jumper JB & S0.  (From the factory mine is S1 and also across between J5 and JA, but he was referrring to SFR1M44-FU-DL).

Can I irreparably damage my Gotek if I jumper the wrong thing?

keirf, 2021-04-24:
Don't jumper JB, that's the speaker output on FlashFloppy. Try jumper at S0 only.

roses2uk, 2021-04-24:
Goes a bit further into the SCSI check, but the display alternates between O_ and 1_ (nothing on the first character)

roses2uk, 2021-04-24:
I'm so proud!! I solved it myself... Realised that there were four Dupont cables in the Rotary Encoder kit that arrived a few days ago. Attached OLED, and hey presto - now finds the sys disk image and boots.

But therein lies another problem: there are not even pin header holes on the circuit board to attach a rotary encoder to. Can you make another graphic in the Wiki/Hardware Mods to show how this is possible. The kit I got from eBay seller chipsdipsuk only seems to need 4 connectors, as opposed to the previous setup of 5.

keirf, 2021-04-24:
the CLK/DT wires can be connected to pins PA13 (SWDIO) and PA14 (SWCLK) on the programming port.
https://raw.githubusercontent.com/wiki/keirf/FlashFloppy/assets/pheader.jpg

roses2uk, 2021-04-24:
Seek and ye shall find.

"SFRC922AT3: Carries the smaller 48-pin AT32F415CBT7. This PCB is missing the usual header pins for installing a rotary encoder; Instead the CLK/DT wires can be connected to pins PA13 (SWDIO) and PA14 (SWCLK) on the programming port." https://raw.githubusercontent.com/wiki/keirf/FlashFloppy//assets/pheader.jpg

AlenMarkov, 2024-03-25:
Highly interesting topic - just bought a new GOTEK and ran into the same issue as roses2uk is describing it (last one did work without problems and new one shows up as "DFU in FS mode" in the device manager). So the reason is a new Microcotnroller... Will now read myself through this thread (perfect timing)...