keirf/flashfloppy #462: [FF v3.23] Strange issue flashing on SFRC922D with dfu-util 0.10 (libusb 1.0.23) on Linux

Drag0nFly, 2021-05-03:
I'm having no luck getting the FlashFloppy firmware to "stick" on this Gotek board. The board is a standard SFRC922D, and appears to be fresh from the factory (with whatever software it came preinstalled with).

Hooked it up to one of my A500 machines (which also has been tested with a different Gotek), and the only activity is the green LED (nothing on either LED or OLED display). No response to buttons, and no activity on USB stick (which also has been tested separately.)

I've run several (around 10) – apparently successful – firmware flashes using dfu-util (see output below). But it does not appear to actually get FlashFloppy on it (is there a way to *check* the .dfu file against what is actually written on the Gotek, btw.?)

Leaving a comment here out of desperation in case there is something I overlooked, as the wiki and several searches did not reveal a similar issue. Obviously there could be some kind of defect with this Gotek board (which would be a first as I have several)

Thanks for any input.

First step, removing the protection (if any)

```
sudo dfu-util -a 0 -s :unprotect:force -D /data/amiga/flashfloppy-v3.23/FF_Gotek-v3.23.dfu
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Device disconnects, erases flash and resets now

```

Listing the device after it resets.

```
sudo dfu-util -l
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Deducing device DFU version from functional descriptor length
Found Runtime: [21ee:1100] ver=0391, devnum=9, cfg=1, intf=3, path="3-1.5.4.1", alt=0, name="UNKNOWN", serial="0009DD402B65"
```

Actually flashing the firmware (one of many identical tries)

```
sudo dfu-util -a 0 -D /data/amiga/flashfloppy-v3.23/FF_Gotek-v3.23.dfu
dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0483
Match product ID from file: df11
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
[... 270 lines cut ...]
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

libusb version 1.0.23 (11397)
DFU suffix version 11a
Match vendor ID from file: 0483
Match product ID from file: df11
Deducing device DFU version from functional descriptor length
Opening DFU capable USB device...
ID 0483:df11
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuERROR, status = 10
dfuERROR, clearing status
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 2048
DfuSe interface name: "Internal Flash  "
Memory segment at 0x08000000 128 x 2048 = 262144 (rew)
Device disconnects, erases flash and resets now
```

keirf, 2021-05-04:
Perhaps if this is a new Gotek it has the AT32F415 chip? You need to use Artery DFU tool to program, and use latest firmware (v3.25). See the wiki: https://github.com/keirf/FlashFloppy/wiki/Gotek-Compatibility#update-april-2021-artery-microcontrollers

Drag0nFly, 2021-05-04:
I thought the same before opening the ticket based on https://github.com/keirf/FlashFloppy/issues/453 + https://github.com/keirf/FlashFloppy/issues/453

But it is the STM32 chip w/ 128k RAM-

```
Found DFU: [0483:df11] ver=2200, devnum=59, cfg=1, intf=0, path="3-1.1", alt=1, name="@Option Bytes  /0x1FFFF800/01*016 e", serial="STM32"
Found DFU: [0483:df11] ver=2200, devnum=59, cfg=1, intf=0, path="3-1.1", alt=0, name="@Internal Flash  /0x08000000/128*002Kg", serial="STM32"
```

The USB device listing identifies it as “STMicroelectronics STM Device in DFU Mode” (should be different if it's the AT chip)

Since this is my first attempt at flashing using dfu-util, I assumed it was something trivial. I looked for a “verify” option in the tool, but it appears that such an option does not exist (in order to verify what is on the board against the actual .dfu file)

Drag0nFly, 2021-05-05:
Did some more testing with this Gotek board, including flashing it again a number of times. I also compared with a different Gotek drive, running on older 0.9.16a firmware. The latter works fine (probably a coincidence).

However, it might be useful to attempt running it with the exact same FF version.

Still, it probably would not explain why the LED does not light up at all, and that USB activity is not registered. The only activity, assuming the jumper is correctly set to S0, is the blinking green led on the Gotek during Amiga bootup. Plugging in a USB drive does not change anything, nor does pressing the buttons.

This is a longshot I know, but as I see the 0.9.16a version is available I thought I'd be an idea to try it as well. Obviously it is missing a .dfu file – is there a way to process the .upd & .hex file into a .dfu in order to test this? If not I can probably revert to the "old way" of flashing providing this would work within a virtual machine.

Naturally, if there any other thoughts as to what might be going on with this board those would be appreciated as well. (I'm not ready yet to throw it away since it is my last spare Gotek drive...)

keirf, 2021-05-05:
There is a script in `scripts/dfu-convert.py` which can be used:
```
$(PYTHON) ./scripts/dfu-convert.py -i FF.hex FF.dfu
```

Drag0nFly, 2021-05-05:
Ok, I found it (or a version of it) after looking around a bit (github's search didn't turn anything up)

However, it appears to have been last updated 11 years ago. There were some syntax issues for print statements - which I corrected, but it currently bombs out with:

```
python ../dfu-convert.py -i FF_Gotek-v0.9.16a.hex FF_Gotek-v0.9.16a.dfu
Traceback (most recent call last):
  File "/data/amiga/flashfloppy_v0.9.16a/../dfu-convert.py", line 57, in <module>
    build(outfile,[target],device)
  File "/data/amiga/flashfloppy_v0.9.16a/../dfu-convert.py", line 19, in build
    tdata += struct.pack('<2I',image['address'],len(image['data']))+image['data']
TypeError: can only concatenate str (not "bytes") to str
```

Is this working on newer distros / python versions? (tried both python3 &2)

Drag0nFly, 2021-05-05:
Strike that. Missing some deps for a correctly-working python2 – sorry for the noise.

keirf, 2021-05-05:
It's included in the FlashFloppy github repository. It's the script I use to generate the release DFUs.

Drag0nFly, 2021-05-05:
Yes, it's just GitHub's search function which proved a little lacking...

We couldn’t find any code matching 'dfu-convert.py' in keirf/FlashFloppy
You could search all of GitHub or try an advanced search.

Drag0nFly, 2021-05-05:
Same behaviour with 0.9.16a (anything else would frankly have been even stranger). Leaves the question whether anything is actually being written to the Gotek, as it exhibits the exact same behaviour it had when it was unflashed from the factory.

Will the LED register when hooked up to an Amiga for such boards (i.e, unflashed from factory)? If so it must be a harddware issue.

The only difference is that the previous Goteks I flashed were done using the “DfuSe_Demo v3.0.5” tool. (However, running that successfully in a VM – or wrangling with a M$ OS in itself – has proved to be an introduction to modern mental torture techniques...)

closed 2021-08-07 by keirf