keirf/flashfloppy #463: Feature request: simulate READY signal / new disk inserted (or maybe I'm just stupid)

rmpel, 2021-05-04:
Hello.

Total noob here (Amiga-wise);
From the options page, I assumed the following configuration would allow me to virtually eject and insert a disk

twobutton-action = rotary
; 1: Up-dir, 2: Select/Eject/Insert

and although this works, it does not do what I expected.

Let me put this another way;
Software says "insert disk 2", so I rotate the knob to disk 2, and nothing happens.
So I eject, and insert, nothing happens.

I've tried mouse click, joystick button, any key on the keyboard, nothing helps.

So I see three possibilities
1. This is not the place to ask this question. if so, just tell me and I'll go away.
2. The game is broken. (Seems unlikely as I tried multiple titles)
3. I'm stupid, I am missing something (very likely).

My system: (and please forgive me, I only have my Amiga for 3 days and it's my first ever Amiga)

Amiga 1200 (Escom) with 4 MB Fast Memory expansion and Co processor (or so the previous owner said)
Gotek flash floppy emulator flash floppy 3.23
Kickstart 3.0

So; If I should be able to simulate a disk change already, please tell me how.
If not; that would be the feature request :)

keirf, 2021-05-04:
You probably have an ESCOM modded A1200 board. Try `interface = ibmpc`. Or revert the ESCOM mod as described here https://www.youtube.com/watch?v=G6fYOjTYvXM

rmpel, 2021-05-04:
Thank you for replying; I have thought of that and I indeed to have an Escom model (or though I had one, see below).

So for that reason, I have tried interface = ibmpc as well as shugart and amiga
I've tested the drive as DF1 on external, DF0 on external (with internal drive disconnected) and as DF0 on internal.

Thank you also for the video to undo the Escom hack.
Upon opening my 1200, I seem _not_ to have an Escom model.
I have an Escom model mouse, but I have no Escom "hack" wire, and the floppy unit is by Chicon FZ 357, which, apparently,  is an "Amiga Original" 

I really am puzzled ;P

thank you for the assist!

Remon.

keirf, 2021-05-04:
Run Amiga Test Kit and look at the drive signal test. Does the CHNG signal ever get asserted when a disk is changed.

rmpel, 2021-05-04:
I will try, will be tomorrow very late though. Thank you so far, will report back in 24 hours ;)

rmpel, 2021-05-04:
Having trouble putting it down.
I found the Amiga Test Kit image on your GitHub, downloaded it, put it on the Gotek. Amiga show it as DF1:NDOS and cannot open it.
When I select Workbench.adf, I see DF1:Workbench3.1

Going to bed now, too tired to think properly. Maybe, ..., tomorrow ... ;)

rmpel, 2021-05-05:
With the internal drive disconnected and the Gotek as DF0, preselected AmigaTestKit.adf and then reboot I have been able to run some test.
when I set motor on on DF0 (which is the Gotek) I can se a list of "flags", CHG is one of them, and that goes away. When I browse for another image (on the Gotek), CHG returns and once I select an image, CHG goes away again.

keirf, 2021-05-06:
Okay so you ran F5 "Signal Test" and observed the real-time set of signals asserted by the Gotek. And that included CHG. Here is the correct behaviour for CHG:
1. It should be asserted as soon as you browse for another image
2. It should disappear only after selecting a new image *and* stepping the drive head (F6)

If CHG is dependent on motor, that's wrong (in fact makes little sense as Gotek ignores the motor signal by default). If CHG disappears immediately on selecting a new image, that's wrong.

What jumpers do you have set at the rear of the Gotek?

What is your *full* FF.CFG? Is it just as in your OP?

Have you tried resetting cached config? To do this remove your USB stick and then press the rotary button for a few seconds until the Gotek display changes.

rmpel, 2021-05-06:
`It should be asserted as soon as you browse for another image`
it does

`It should disappear only after selecting a new image and stepping the drive head (F6)`
it does, see below

My current (full) config is as above, but I added one line later in the hopes it would help.
```
interface = shugart
ejected-on-startup = yes
autoselect-folder-secs = 0
autoselect-file-secs = 0
twobutton-action = rotary
chgrst=delay-4
```
With this config, the CHG disappears 2 seconds after selecting an image. (as intended, per ff.cfg examples)

But indeed; the CHG flag (or any flag) does not change on the screen unless I set motor on.

Jumpers:
I don't know these by heart.
They should be on 3 for S0, buzzer on 4, led on 6, but I have to check.

I have another Gotek on the way, should arrive tomorrow (hopefully, otherwise it will be Tuesday next week). We'll see what that one does.

`Have you tried resetting cached config?`
I don't know what that is, but changes in FF.CFG seem to work (like changing the CHGRST delay) so I don't think it is a caching issue.
`To do this remove your USB stick and then press the rotary button for a few seconds until the Gotek display changes.`
I think in the many things I tried, this was one to them, but, ...
I will do this tonight and report back.

During testing with The Lost Vikings (which is the game that asks for disk 2, and never continues after switching to it) I tried any key and mouse/joystick button to "continue" (just in case), but, it appears my right mouse button is broken.
I have investigated this last night, and it is the cable, not the Amiga, that is at fault. Found this out with AmigaTestKit, so, thank you for that as well :)
I inserted a 9-pin extension with removable plug-casings, and shorted pin 9 to 8 (RMB to GND) and then a right-click is registered. Tested Lost Vikings with that, and still no go, so it is, in fact, a disk-change that is the trigger to continue.

Unmentioned earlier; the supplier of the gotek sent me his "root files" which are HCXblala.CFG (binary) and automount.img. 
I also tried these same to files from the "amiga starter kit" on the website
In both cases; the drive will not function as expected, none of my images are visible.
So my USB disk has only the following:
```
/FF/FF.CFG
/Amiga/Games/*.hfe
/Amiga/Workbench/*.adf
/Amiga/AmigaTestKit.adf
```
Maybe that's wrong??? If so, let me know, please :)

Recap (pun intended):
- report jumper settings (tonight)
- test no-usb + rotary push-and-hold (tonight)
- test different gotek (when I get it)

keirf, 2021-05-06:
Why do you have `chgrst=delay-4` ?

Change it back to default: `chgrst=step`

keirf, 2021-05-06:
And I suspect you are misjumpered. Perhaps you have a jumper at MO. It should be at S0.

rmpel, 2021-05-06:
as stated; `I added one line later in the hopes it would help.`

After yet another unsuccessful test, I dropped everything and went to sleep ;) that's why my current config still has that line.
The intended config is the same minus that line.

keirf, 2021-05-06:
> as stated; `I added one line later in the hopes it would help.`
> 
> After yet another unsuccessful test, I dropped everything and went to sleep ;) that's why my current config still has that line.
> The intended config is the same minus that line.

You cannot just remove a line. Removed lines are cached in the Gotek unless explicitly reconfigured in FF.CFG. So removing `chgrst=delay-4` actually has no effect. You either must:
(a) Explicitly reconfigure, `chgrst=step`, or
(b) Reset cached config (hold the button for a few seconds with no stick inserted)

This is a subtlety of config handling, implemented so that display config works at power on before a stick is mounted. Users do tend to trip on it for obvious reasons!

rmpel, 2021-05-06:
wow..
o.k.

did not know that.

I will add to my list;
- get default full config file, change lines that need changing and try again ;)

Thanks :)

keirf, 2021-05-06:
Yeah it's non intuitive!

Your approach is fine and apart from the chgrst line your config is good. It's just sensible to reset cached config after each change. At the end of the day I think this is a chgrst config plus physical jumper issue.

rmpel, 2021-05-06:
chgrst was unlisted, therefore default, therefore "step" when I first got the device
jumper was set to "DF1" (so position 3, I think?)

That did not work
changed jumper to DF0, and connected it where the floppy was, did not work
changed the chgrst, did not help
switched back to DF1 (and gave up for the day)

So effectively I think I tried every relevant combination, but, I will start from scratch tonight and try and try again.

keirf, 2021-05-06:
Another possibility is: are you using a floppy ribbon cable with twist in it (pins 10-16 swapped round)? That would confuse matters!

rmpel, 2021-05-06:
Good question, answer is "no"; there is is no drive-select twist in the cable, and the problem does not go away with using either the flat cable that is attached to the 23-pin connector or the internal original floppy cable.

rmpel, 2021-05-06:
Today I learned...

External drive port is hardwired not to accept a DF0. the sel0 signal, IDC pin 10, is not available on external port. So setting an external Gotek to boot will never work.
Although this was not the question, it was something I learned today.

I've reset the config, written a more complete ff.cfg, essentially the last config, but with step for chgrst.
drive did not work.
set the interface to amiga (which should be the same as Shugart, according the comments in the example config file) 
Drive is connected to the external port, with a straight cable, jumper on 2 (S1), buzzer on 4 and some other thing on 6 (either LED or Rotary, wires go inside)
No other jumpers

Drive works.

With a hardware hack in the mouse cable I can now right-click (new old stock mouse to arrive in 2 weeks, so very temporary hack :P) and now things are better.

For some reason, I now get garbled video in Lost Vikings, but I will continue with that problem once I have my RGB cable.

So, conclusion

No idea what was the problem in the end, maybe it was a stuck config that got reset with 3-seconds-on-rotary, maybe it was something else. In any case, with a lot of trouble I can now boot from the Gotek while still being external.

Thank you for all your time. 

I will close this in a few days, just in case I need to report more ;)

keirf, 2021-05-07:
The external drive port isn't IDC so it depends how the dsub is wired to IDC. Many enclosures connect DF1 select to pin 10, but not all.

At least the problems are solved. Perhaps you had a few combined. `interface=amiga` may help make the Gotek detectable at boot even when no image is mounted, as described here: https://github.com/keirf/FlashFloppy/wiki/Host-Platforms#forcing-drive-identification

...but apart from that shouldn't fix the kinds of issues you were seeing.

rmpel, 2021-05-07:
according to the A1200 schematic, IDC pin10 (E580) is not connected to the dsub at all, so an external drive can never be DF0. 
Information on the external adaptor from dsub back to IDC is harder to come by
Some image show a simple schematic with connectivity from SEL1 on the dsub to pin 10 (SEL0) on the IDC, which would suggest that an external drive should be set as DF0 but then function as DF1 (kind of like the floppy cable twist)
Other show schematics with ics that my adaptor does not have.
I checked continuity and found a 1 to 1 pass through on all signals (except pin 10 sel0 and pin 16 mtr0, of course) so I must assume the schematic I found is wrong.

sources: 
https://www.amigawiki.org/dnl/schematics/A1200_R1.pdf
https://lh3.googleusercontent.com/proxy/qfnerCRe-BzimDAatcLdVG1cmr096JMoDCY2sueLbk3_beh2IScHKOckI6YrFKfhPQD0kVtAvkKh_0nask8P8NAhF9tofxwdoFduPqO7T4jorLbQ7HGc0zpnfw8gROO9MgWDjUOL_cWELtcp7KLk2VI

`interface=amiga` indeed makes things a little easier.
I am now however re-evaluating my decision to keep the A1200 stock and the gotek external.
Maybe I really should be internalising the Gotek and be done with all the hassle.
If only there was a way to set DF1 on highest boot priority ...

Anyways,
once more;

thank you for all the help :)

closed 2021-08-07 by keirf