keirf/flashfloppy #464: Mazak multiplex 620 with mazatrol T32-6 operation system read and write error

hikodaci, 2021-05-04:
Hello dear All!

We have a cnc machine brand Mazak which is working with floppy and has mazatrol T32-6 operation system.

The original floppy is brand by Mitsubishi

[image]

I used the standard Flash Floppy on it with a blank image as 000.IMG file and FF.cfg. Also, I added the files in one zip file as an attachment.
[Mazak.zip](https://github.com/keirf/FlashFloppy/files/6423148/Mazak.zip)

When I try to save files from the machine to the floppy disk The machine gives an error. Also, When I plug the USB stick the Gotek the green led is always on. 
[image]

I have tried S1, S1+JC, S0+JC and also, I have turned the data cableway but the machine never turns on. But there is no answer from the machine for our request.

There are some parameter information

[image]

[image]

[image]

When I search on google I learn the Mazak with T32-6 uses 360 RPM floppy drive.

> The disk format is the old 8" double sided 77 track 360 RPM format.
There is nothing that a standard 300 RPM drive can do to read these disks.> 

What should I do? What are your suggestions?

keirf, 2021-05-04:
You are best using an HFE image file. I will make you one.

keirf, 2021-05-04:
Try the image below (unzipped of course). It will need to be formatted. If it isn't reliable you can at least send the image back to me and I can probably suggest an IMG configuration so you can use the more efficient IMG image format.

[a.zip](https://github.com/keirf/FlashFloppy/files/6423943/a.zip)

hikodaci, 2021-05-05:
I have tried also .hfe file there is no command to format the disk in my machine. And When I try to save the file to floppy again it gives the same error. But I'm confused about the green led always on. Is it normal?

hikodaci, 2021-05-05:
If there is no jumper on the Gotek, the green led is off.

keirf, 2021-05-05:
The host may permanently select the drive in which case this is expected. Or it may mean that your cable is connected wrong way round (pin 1 at the wrong side).

hikodaci, 2021-05-05:
When I try to change the cableway, the machine doesn't turn on.

keirf, 2021-05-05:
Then you were right first time. The problem is that if you can't format a disk on this system, you need a preformatted disk image.

hikodaci, 2021-05-05:
How can I get a preformatted disk image? Or how can I create?

keirf, 2021-05-05:
Unfortunately things like number of sectors per track is not standardised. You need to know the format needed for your system. For example, by dumping an existing physical disk (using Greaseweazle, Kryoflux, Supercard Pro, or similar hardware).

closed 2022-06-22 by keirf