keirf/flashfloppy #465: There are some problems with the settings on a DOS PC.

sarafiel, 2021-05-13:
There are some problems with the settings on a DOS PC. I am using v3.24.

My gotek is artery new MCU and Jumper is JC and S1 and twisted FDD cable.
I Modify rotery encoder / piezo with 5v volume up / 1602A LCD.

[image]

1. If 2HD (1.2 or 1.44) is selected during the booting phase, only 2HD can be used normally until rebooting, a General Failure error when 2DD is recognized, and 2DD (360 or 720) this time when 2DD is selected and rebooted in this state. Only normal use is possible and General Failure error persists in 2HD. I don't know what's the problem.

2. When one image is recognized and replaced with another image, the replaced image is visible on the LCD, but the original image is retained. It means that the list of disk1 is still visible when the disk1 is recognized and changed to disk2. Obviously, accessing the file contents of disk1 in this state will result in an error. When accessing the disk1 image, an error appears, and when disk2 is recognized again in that state, the disk2 image is loaded normally. I expected it to be a cache problem, but the same problem occurs in DOS and Windows 95, where all caches such as smartdrv have been removed. The settings of ff.cfg are the same as the default options, but this does not solve the problem.

keirf, 2021-05-14:
Not recognising disk changes. Different (bad) drive cable compared with your 286 PC?

sarafiel, 2021-05-14:
I ran a few more tests. As a result, it was confirmed that it was not a problem with Gotek.

I tried replacing the FDD cable that was operating normally on the 286 and connecting it to the Pentium, but the symptoms are the same.

In addition, when Gotech finishes scanning the disk track to the end and inserts a new image, it is not reset to T:0, but it is confirmed that it is in T:80 state (probably, it is judged that the disk was replaced at some stage).

I actually connected the FDD drive to the Pentium, but even if the diskette was intermittently replaced, the existing data list was confirmed when the dir command was displayed.

The same problem occurred in the actual FDD, and the problem did not occur when the same Gotek was installed on the 286, so it was concluded that there was no problem with Gotek.

But I want to figure out the problem. Was the FDD io part of my Pentium board damaged? For example, if the part that handles the disk change signal is broken, it is not possible to properly process the signal corresponding to the disk change, and if so, there is no other simple solution other than a difficult solution such as changing the chip, after listening to the advice of an experienced expert. want.

I am sad. I am happy to have confirmed that Gotek is fine, but I am not happy to see that my favorite board is weird.

sorphin, 2021-05-29:
I have this exact same issue on varying systems. (the lack of disk change being detected without say changing to B: and then back to A:) and the General Failure when switching disk sizes (until reboot).

closed 2021-11-24 by keirf