keirf/flashfloppy #466: Revert to previous version

roses2uk, 2021-05-15:
Is it possible to downgrade to a previous FF version using the *.upd file in the root directory/push both buttons while powering up method, or does it only accept higher versions?

I'm having problems getting my Roland S760 sampler to see any disk images I use from my large library. The 1.44 MB system disk image (system224.out) loads fine, but any other files regardless of extension (out, img, etc) are not available for loading. If I revert back to a FDD,  system boots from floppy, and the sound disk floppy (S7-S760-O1, the only disk I have) also loads without issue.

So I want to see if a previous FF version will help.

Any 760/Gotek owners out there reading this, please chime in with your thoughts...

keirf, 2021-05-15:
Yes you can downgrade or upgrade the same method. I recommend to clear cached config if you downgrade (hold both buttons for a few seconds after power on, with no stick inserted, until the display changes).

closed 2021-06-22 by keirf

pelagos, 2021-08-20:
> Is it possible to downgrade to a previous FF version using the *.upd file in the root directory/push both buttons while powering up method, or does it only accept higher versions?
>
> I'm having [...]

Hi roses2UK - did you end up getting this working? I am having the same issue

Thanks

pelagos1