keirf/flashfloppy #474: FlashFloppy is unreliable / doesn't work with IBM 3174

blackbit42, 2021-05-26:
I use two GoTek emulators in an IBM 3174 controller. These emulators are of the 'new' variant, with Artery microcontrollers. FlashFloppy is at version 3.25 (confirmed via 7-segment display). The diodes that are required for 3174 operation are installed correctly as well.

There are two observations.

Observation [1]:
When I use an actual floppy drive in slot 1 and a GoTek in slot 2 to 'digitize' disks with the copy facility of the IBM microcode, this is works in some cases and doesn't in others.
During copying there is progress reporting both on the terminal screen and the 7-segment display of the Gotek in terms of tracks that are accessed. I have made some notes during testing.
In the following list, OK obviously means 'no error'. A number means the track at which an error was reported and the copy operation aborted. The copy process operates on descending tracks.

```
disk number | size | result
---------------------------
          0 | 2.4M |     39
          0 | 2.4M |     40
          0 | 2.4M |     OK
          1 | 2.4M |     45
          1 | 2.4M |     OK
          2 | 2.4M |     OK
          3 | 1.2M |     OK
          4 | 1.2M |     OK
          4 | 1.2M |     OK
```

Observeration [2]:
When I try to IML (i.e. 'boot') the machine from a GoTek emulator, there are clearly disk accesses, but eventually an error is reported and the boot process is aborted.
I've made a video to illustrate what happens.

https://user-images.githubusercontent.com/19274131/119686381-ce61eb80-be35-11eb-8fbe-694556511646.mp4

I have tried this several times in a row and wasn't able to make it work a single time.
Status code 385 means 'Disk drive not ready.'

How do I debug this?

keirf, 2021-05-26:
The problem is the storage and data rate required for a 2.4MB track, an Artery Gotek can probably not quite buffer that amount, and certainly nowhere near if you are trying to use HFE images (even an STM32 based Gotek may be problematic in that case).

So what type of image files are you using? Can you provide an example one (formatted at least partially would be good). If you're using HFE perhaps we can at least fix you up using IMG files, which are a lot lighter on RAM and USB bandwidth.

blackbit42, 2021-05-26:
Thanks for the quick response.
I am using img images.
The particular image file that I tried to 'boot' was created by overwriting the content of a 2.4M image that I put on the USB drive as described under 'Observation [1]' above.
So, with the Artery uC GoTeks that I have, is there anything I can do?

keirf, 2021-05-26:
Is it possible to boot/read a clean known working image? Or are we dealing only with potentially broken part written image files here?

I think artery should probably be able to deal with 2.4 IMG read ok, but whole track write/format may be problematic. That would be true for stm32 as well though. Single sector writes should be ok.

Other thing to try is a build from v4 prerelease. I can dig out a link for you tomorrow.

blackbit42, 2021-05-26:
To rule out that the image file that was produced by the 3174 when writing on the GoTek, I can try to boot another image that was made by other means, tomorrow (CET).

blackbit42, 2021-05-27:
I've booted the 3174 with a known good image. Same result.

keirf, 2021-05-27:
Can you upload an image and your IMG.CFG?

You can also try the logfile firmware https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive
This leaves even less memory available for track buffering but can still give us an idea what's going on.

blackbit42, 2021-05-27:
[C6_3_Control.zip](https://github.com/keirf/FlashFloppy/files/6553504/C6_3_Control.zip)
[FF.CFG.zip](https://github.com/keirf/FlashFloppy/files/6553508/FF.CFG.zip)
I do not have an IMG.CFG.
Will try the pre-v4 build.

keirf, 2021-05-27:
Ah I forgot FF has built in 3174 support, which someone contributed last year.

The first release to have this support was v3.12a. Perhaps it has since rotted. You could try v3.12a and see if you have any joy. It is available from the FF Downloads page.

keirf, 2021-05-27:
I will give it a test myself (with debug logging) when I get a chance, and see if there's anything obvious going wrong when setting up for a 2.4M image.

keirf, 2021-05-28:
I did some light testing, dumping the Gotek output using a Greaseweazle. The output was as expected.

Perhaps it is worth trying to contact the original 3174 author via their pull request #314

blackbit42, 2021-06-13:
I did some tests with different flash drives.
IMLing (booting) the machine requires two disks and is a read-only operation.
I did a series of tests with an actual floppy drive for disk 1 and a GoTek with respective image for disk 2.
The operation succeeded 5 times in a row, so this seems to work as expected.
I then proceeded with two GoTeks for the same operation.
This did not succeed once in a series of 5 tries. It all starts promising. The first few tracks are accessed, then the 3174 bails out with status code 385 (disk drive not ready).

After having conversations with two people who use GoTeks with FlashFloppy in their machines I was assured that the setup works reliably for them, for reading and writing. These people have the older hardware revision that does not yet have the Artery controller, so my suspicion is that this is the culprit.

How can I positively confirm that the controller with it's smaller RAM makes the setup fail? Are there any debug logging functions that I could use?

keirf, 2021-06-13:
There is alt/logfile firmware you can run to gather logs: https://github.com/keirf/FlashFloppy/wiki/Logging-To-USB-Drive

There is the v4 prerelease firmware you can try which drastically changes write buffering (linked in an earlier post too): https://github.com/keirf/FlashFloppy/suites/2685959106/artifacts/59212216

blackbit42, 2021-06-14:
The link to the v4 prerelease isn't valid anymore.
I pulled the file back then and gave it a try.
Interestingly the firmware version reported on the 7-segment display is 3.23, which makes me unsure I got the right file.
This is the sha1sum.
05df4198bc1fb617243188c2b58c02ba951ad830  FF_Gotek-3932716.upd
When I use this build, the whole thing doesn't get very far. When the disk is accessed, the 3174 bails out with the 'Check Cond' LED lit.

I used the logging firmware of 3.25 recently to track what happens when writing a floppy. Can the log be made more verbose?

keirf, 2021-06-14:
Ah the track may be too long in combination with the reduced Artery RAM, for the approach currently taken in v4. You probably did have correct firmware, the version number is not currently modified.

The logging cannot be made automatically more verbose, but extra printk() statements can be added. We are hoping that statements in function `raw_write_track()` will be printed.

blackbit42, 2021-06-14:
I've written a normal log with 3.25. The results are a bit confusing to me.
While booting, I made notes which tracks are accessed according to the 7-segment display of the GoTek.
This is the list:
1 2 3 4 5 6  <pause>  0 1 0 6 7 8 0 <error status code on 3174, end>

All those accesses should be read-only. During booting nothing is written to the disk.
I do not see these accesses reflected in the log.

```** FlashFloppy v3.25 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Apr 25 2021 15:06:36
Board: Standard
Config: Flash Slot 0 (ver 2, size 69)
Interface: IBM PC (pin2=high, pin34=chg)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5581
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra
 Serial Number : 4C530001190731110394
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 42 items
Mode: HxC
[DSKA0001*.*]
Current slot: 1/5
Name: 'DSKA0001_Control' Type: img
Attr: 20 Clus: 0000009a Size: 2442240
Fast Seek: 1 frags
Cache 17 items
Write 1[02]/15... 7511 us
Write 2[03]/15... 7917 us
Write 3[04]/15... 7935 us
Write 4[05]/15... 7859 us
Cache 42 items
[DSKA0002*.*]
[DSKA0002*.*]
Current slot: 2/5
Name: 'DSKA0002_Extention' Type: img
Attr: 20 Clus: 00000130 Size: 2442240
Fast Seek: 1 frags
Cache 17 items
Cache 42 items
[DSKA0001*.*]
[DSKA0001*.*]
Current slot: 1/5
Name: 'DSKA0001_Control' Type: img
Attr: 20 Clus: 0000009a Size: 2442240
```

If I interpret the log correctly, then there were only write and no read accesses, and the list of tracks doesn't match what was seen on the 7-seg display. Does all of that make sense? I struggle to see the relation between the 7-seg display and the log content.

keirf, 2021-06-14:
Reads aren't logged, actually a floppy drive is always reading while its motor is turned on and it's not writing. That is, while motor is on, the drive is continuously serving up the track that the drive head is currently seeked to. FlashFloppy deals with extracting the sector data from the IMG file and, in an endless loop, framing that into a raw encoded track for transmission to the host.

The write statements in your log above are FlashFloppy extracting sector data from the host raw writes, working out which sector is being written, and then the microseconds latency figure is time taken to write to the USB drive (which here looks pretty long!).

You could dump information about each track stepped to by changing `verbose_image_log` definition in inc/floppy.h to `TRUE` and then rebuild (`make logfile=y`)

blackbit42, 2021-06-14:
Here we go.
Does this provide any interesting insights?

```** FlashFloppy v3.25 for Gotek
** Keir Fraser <keir.xen@gmail.com>
** https://github.com/keirf/FlashFloppy

Build: Jun 14 2021 16:15:25
Board: Standard
Config: Flash Slot 0 (ver 2, size 69)
Interface: IBM PC (pin2=high, pin34=chg)
I2C: Invalid bus SCL=1 SDA=0
Display: 3-Digit LED

> USBH_USR_Init
> USBH_USR_ResetDevice
> USBH_USR_DeviceAttached
> USBH_USR_ResetDevice
> USBH_USR_DeviceSpeedDetected
> Device speed: Full
> USBH_USR_DeviceDescAvailable
 VID : 0781
 PID : 5581
> USBH_USR_DeviceAddressAssigned
> USBH_USR_ConfigurationDescAvailable
> Class connected: 08 (MSC)
 Manufacturer : SanDisk
 Product : Ultra
 Serial Number : 4C530001190731110394
> USBH_USR_EnumerationDone
> USBH_USR_UserInput
Cache 42 items
Mode: HxC
[DSKA0001*.*]
Current slot: 1/5
Name: 'DSKA0001_Control' Type: img
Attr: 20 Clus: 0000009a Size: 2442240

[lost 25037]
,2} {3,0,23,2} {3,0,24,2} {3,0,25,2} {3,0,26,2} {3,0,27,2} {3,0,28,2} {3,0,29,2} {3,0,30,2} 
C2 S0:: MFM 80-2-30:
 rpm: 180, tracklen: 333344, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 348
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{2,0,1,2} {2,0,2,2} {2,0,3,2} {2,0,4,2} {2,0,5,2} {2,0,6,2} {2,0,7,2} {2,0,8,2} {2,0,9,2} {2,0,10,2} {2,0,11,2} {2,0,12,2} {2,0,13,2} {2,0,14,2} {2,0,15,2} {2,0,16,2} {2,0,17,2} {2,0,18,2} {2,0,19,2} {2,0,20,2} {2,0,21,2} {2,0,22,2} {2,0,23,2} {2,0,24,2} {2,0,25,2} {2,0,26,2} {2,0,27,2} {2,0,28,2} {2,0,29,2} {2,0,30,2} 
C1 S0:: MFM 80-2-30:
 rpm: 180, tracklen: 333344, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 348
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{1,0,1,2} {1,0,2,2} {1,0,3,2} {1,0,4,2} {1,0,5,2} {1,0,6,2} {1,0,7,2} {1,0,8,2} {1,0,9,2} {1,0,10,2} {1,0,11,2} {1,0,12,2} {1,0,13,2} {1,0,14,2} {1,0,15,2} {1,0,16,2} {1,0,17,2} {1,0,18,2} {1,0,19,2} {1,0,20,2} {1,0,21,2} {1,0,22,2} {1,0,23,2} {1,0,24,2} {1,0,25,2} {1,0,26,2} {1,0,27,2} {1,0,28,2} {1,0,29,2} {1,0,30,2} 
C0 S0:: MFM 80-2-15:
 rpm: 360, tracklen: 166688, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 102
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{0,0,1,2} {0,0,2,2} {0,0,3,2} {0,0,4,2} {0,0,5,2} {0,0,6,2} {0,0,7,2} {0,0,8,2} {0,0,9,2} {0,0,10,2} {0,0,11,2} {0,0,12,2} {0,0,13,2} {0,0,14,2} {0,0,15,2} 
C0 S1:: MFM 80-2-15:
 rpm: 360, tracklen: 166688, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 102
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{0,1,1,2} {0,1,2,2} {0,1,3,2} {0,1,4,2} {0,1,5,2} {0,1,6,2} {0,1,7,2} {0,1,8,2} {0,1,9,2} {0,1,10,2} {0,1,11,2} {0,1,12,2} {0,1,13,2} {0,1,14,2} {0,1,15,2} 
Write 1[02]/15... 5361 us
Write 2[03]/15... 4758 us
Write 3[04]/15... 5302 us
Write 4[05]/15... 4705 us
Cache 42 items
[DSKA0002*.*]
[DSKA0002*.*]
Current slot: 2/5
Name: 'DSKA0002_Extention' Type: img
Attr: 20 Clus: 00000130 Size: 2442240
Fast Seek: 1 frags
Cache 17 items
C0 S0:: MFM 80-2-15:
 rpm: 360, tracklen: 166688, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 102
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{0,0,1,2} {0,0,2,2} {0,0,3,2} {0,0,4,2} {0,0,5,2} {0,0,6,2} {0,0,7,2} {0,0,8,2} {0,0,9,2} {0,0,10,2} {0,0,11,2} {0,0,12,2} {0,0,13,2} {0,0,14,2} {0,0,15,2} 
C0 S1:: MFM 80-2-15:
 rpm: 360, tracklen: 166688, datarate: 500
 gap2: 22, gap3: 104, gap4a: 80, gap4: 102
 ticks_per_cell: 1151, write_bc_ticks: 72, has_iam: 1
 interleave: 1, cskew 0, hskew 0
  file-layout: 0
{0,1,1,2} {0,1,2,2} {0,1,3,2} {0,1,4,2} {0,1,5,2} {0,1,6,2} {0,1,7,2} {0,1,8,2} {0,1,9,2} {0,1,10,2} {0,1,11,2} {0,1,12,2} {0,1,13,2} {0,1,14,2} {0,1,15,2} 
Cache 42 items
[DSKA0001*.*]
[DSKA0001*.*]
Current slot: 1/5
Name: 'DSKA0001_Control' Type: img
Attr: 20 Clus: 0000009a Size: 2442240```

blackbit42, 2021-06-15:
I cannot find the list of access tracks I mentioned above in the verbose log.
Also, I don't see any indication that a command sent from the host to the GoTek wasn't properly answer or something like that.
Is there anything else I can do here?

keirf, 2021-06-15:
The `C0 S1:: ...` and similar lines are indicating seek to cylinder (0) and side (1). It is quite verbose though and the log buffer is only 2kB so a lot of logging gets lost, as it's only flushed to USB drive when the image is unmounted.

Agreed there's no sign anything is lost.

Your best bet is to try and find an STM32 based Gotek... At least then you can see if the Artery RAM is the issue.

blackbit42, 2021-06-15:
> Your best bet is to try and find an STM32 based Gotek... At least then you can see if the Artery RAM is the issue.

Yes, seems so.
It seems like the easiest way to obtain an STM32 based PCB is OpenFlops. The sellers of GoTek emulators on eBay and so forth usually do not advertise which microcontroller is on their boards and it seems like traditional GoTeks with STM32 are becoming rare to be offered rather quickly.
@keirf would you have concerns regarding going for an OpenFlops or is that the way to go?

keirf, 2021-06-15:
Should be no reason not to go with openflops, it's just a Gotek clone really.

ejona86, 2021-06-19:
I hacked `identify_mcu` to unconditionally set `ram_kb` to 32 and didn't see any particular difference in behavior compared to 64, on both stable-v3 and v4. Just the expected faster head swapping with 64k and especially v4. Fluxengine read the data just fine and nothing looked strange with a logic analyzer. v4 is able to handle the large IMG track in the same way it handles large HFE tracks; it's just a circular readahead buffer. v4 would have trouble with a large track size if there is sector interleaving, but that doesn't seem to be the case here.

I'm concerned to hear v4 failed sooner than stable-v3, but it seems this is a case where you really need 64k anyway so I guess I won't fret about it much.

keirf, 2021-06-20:
@ejona86 but your evidence indicates that 64k isn't needed here? Not that we have strong evidence that v4 is worse than v3 here, and certainly nothing that can be immediately acted on.

kwoot, 2021-07-07:
Hi. Newbie here. I have a 3174 and love to buy 2 Gotek emulators but don't know what type exactly. I seem multiple models on their website. If someone could point me to the correct product that would be awesome. Regards, Jeroen.

blackbit42, 2021-07-07:
Hi @kwoot

I've been assured by several people by now that the GoTek models with STM32 micro-controller work reliably in 3174s.
Hence my conclusion is that the issues I am experiencing are due to the Artery controller I have in my emulators - root cause being the smaller RAM or something else.
Two OpenFlops emulators are currently en-route to me, but stuck somewhere. Once I was able to validate they work as expected I'll report here.

Unrelated off-topic question: Which model of 3174 do you have and what setup do you run around it?
If you are up for a chat, https://discord.gg/zQbDV23A #terminals is a great place for that. I would appreciate if we could talk.
We shouldn't spam this GitHub issue with off-topic stuff.

blackbit42, 2021-07-14:
I received two OpenFlops boards today.
They work as intended. Yay!
Hence, some aspect of the Artery controller is the culprit. If it's the smaller RAM or something else, I do not know.
Anyways, my concrete issue is solved.
I recommend to add a note to the wiki, stating that the Artery controller is incompatible with the IBM 3174.

closed 2021-07-14 by blackbit42

keirf, 2021-07-15:
Would you like to try a firmware with the Ram size forced to 32kb, on Openflops?

blackbit42, 2021-07-15:
I am happy to, if it helps to confirm the smaller RAM is the issue.
https://github.com/keirf/FlashFloppy/blob/5c71b087fda4e8a77a3e29f21f55b1a5136c6406/src/stm32f10x.c#L14
just s/64/32/ , build, install and try?

keirf, 2021-07-15:
Yes that should do it!

jmaynard, 2023-02-18:
Just to provide a pointer for folks who come across this discussion from Google searches or the like:

The latest version of the Gotek, with the Artery AT32F435 MCU, works fine on the 3174 with FlashFloppy+ 3.38. You still need the diode as described in the documentation, as well as FF.CFG. Disk images for the 3174 must be exactly 1228800 or 2442240 bytes in length, to be seen by FF and presented to the host as 1.2 or 2.4 MB floppies respectively. You can create disk images by using `dd if=/dev/zero of=`filename` bs=1024 count=2385` to make a 2.4 MB blank floppy, and then use the 3174's full copy function to copy a real floppy to the emulated one. (Use `count=1200` for a 1.2 MB floppy, though those are probably pretty rare.)