keirf/flashfloppy #477: Drive reacts as if selected if either CS0 or CS1 is asserted.

stephen-usher, 2021-05-29:
Background:
I've been attempting to get FF devices to act as floppy devices on a couple of ACT Apricot F Series machines, both singally and together with one of the original drives.

I've had to use a converter cable between the Apricot's 26 PIN connector and a standard Shugart 34 pin connector with the following Shugart pins connected:

2 - density select
4 - head load
8 - index
10 - motor enable 0 (tied to ground as otherwise the BIOS diagnostic fails as the Gotek is too slow to respond when the system initially toggles "motor on". For normal operation this makes no odds as the "motor on" signal is always enabled subsequently. Access and data flow is controlled using CS0/CS1 and "head load" only.)
12 - CS1
14 - CS0
16 - motor enable 1 (tied to ground, see above)
18 - direction
20 - head step
22 - write data
24 - write enable
26 - track zero
28 - write protect
30 - read data
32 - head select
34 - ready

Note 1: The CS1 output is driven by a NOT gate which is fed the CS0 signal, so if one is high the other it automatically low.
Note 2: There is only one "motor on" line on the Apricot 26 pin connector.

Issue:
The Gotek (and Goex) drive responds to when CS0 is either high or low (and hence CS1 is low or high).

When the drive is the only one connected to the computer then the machine sees two floppy drives. If it is connected with another drive then the data from the other drive is corrupted by output from the Gotek.

I've seen a similar problem when the Gotek is used as a second drive with an Amstrad CPC464 + DDI + FD1 floppy drive and the Gotek being "drive B".

keirf, 2021-05-29:
If the Gotek responded regardless of drive select then it would not work correctly in multi drive setups with which I'm more familiar (primarily Amiga). I've also seen it work correctly as a second drive in a CPC setup too albeit not tested it myself recently.

keirf, 2021-05-29:
If the Gotek has stm32 chip then you can safely test older firmware versions. I recommend to clear cached config (hold both gotek buttons method) when you do this.

I don't understand comments regarding motor enable lines in your pin connection summary as by default gotek ignores motor enable entirely. Motor is assumed always on.

stephen-usher, 2021-05-29:
I have no idea what's actually going on.

Unfortunately, all I can report is the empiracle evidence.

If I connect the machine's "motor on" line to both "motor enable 0" and "motor enable 1" the BIOS diagnostics error with "drive not found" (Error 35). Pull them to ground and things work.

On the CPC the drive works fine at the end of the ribbon cable but not when it's on the second connector (this apparently merely swaps the CS0 and CS1 lines in the cable. If I then fit an FD-1 drive to the end IDC connector things don't work correctly. Of course, this could be an entirely different problem.

With respect to the Goteks and Goex, they all are older models.

keirf, 2021-05-29:
I think you perhaps have your drive pin assignments for pins 10-16 confused. Your assignments are correct on a PC mainboard connector. However on all drives these pins are assigned as follows: 10=SEL0, 12=SEL1, 14=SEL2, 16=MOTOR.

The weird PC mainboard assignment is so that two drives can have independent motor signals, all drives can be jumpered to respond on SEL1, and a drive responds as A or B depending on whether it has a straight connection to the mainboard (it is drive B aka 1) or has pins 10-16 on the ribbon twisted/reversed (it is drive A aka 0). Think about that.

So I suggest you ignore your motor signals, connect CS0 and CS1 to pins 10 and 12 respectively, and identify your Gotek drive as drive A or B by jumpering S0 or S1 (respectively) on the Gotek jumper block.

stephen-usher, 2021-05-29:
Oh, I see!

That's one of the perils of using Google to find out pinouts, you always get incomplete information. The pinout stated Shugart, which was obviously wrong. *sigh*

I'll have to try to re-wire the converter cable in that case and try again. Unfortunately I've had to temporarily put the Apricots into store whilst I get some machines ready for a display that's due to happen (COVID permitting) at the beginning of July. I'll leave this case open until I've done the testing. (Of course, if the event is cancelled then this will be sooner rather than later.)

keirf, 2021-08-07:
Please reopen this issue if further information arises requiring discussion.

closed 2021-08-07 by keirf