keirf/flashfloppy #480: firmware 3.23 with fixed skew/interleave on MSDMF (1.68M) images

gfernval, 2021-05-31:
Could you build a 3.23 firmware with skew/interleave on MSDMF (1.68M) images fix?
Problem is that 3.24 and 3.25 doesn´t work well with 2 gotek STM based connected (Drive A and Drive B)
to a Ohio Scientific C4P computer (uses HFE images). Appears random writes that trash the HFE images.
It would be advisable to release separate firmware versions optimized for STM and Artery microcontrollers.

thanks

keirf, 2021-05-31:
There should be no degradation in v3.24, the changes are straightforward and RAM changes affect Artery only. If not the case it needs tracking down and fixing.

keirf, 2021-05-31:
Here is an automated build immediately after the MSDMF fix but before the Artery changes. You will need to be logged in to Github to download the Artifact at the bottom of the page:
https://github.com/keirf/FlashFloppy/actions/runs/620521215

If it works better than v3.24 with HFE images then there's an issue to be tracked down.

gfernval, 2021-05-31:
Thanks for the file, it is possible to build the firmware under Windows (using msys32)?

keirf, 2021-05-31:
You don't need to, it's already built. Unzip the non-debug zip file and use the UPD file within.

closed 2021-08-07 by keirf