keirf/flashfloppy #483: 720KB Cnc machine Floppy Write error

hikodaci, 2021-05-31:
Hi, I have a CNC machine which is work with a 720KB floppy disk.

I have tried to operate with FF v3.23 and I have successes to read The file on the USB stick. But When I try to Format or write a file from the Machine to the USB stick. The machine always gives the Error.

The Jumpers (Only M0, Only S1, Only S0, M1 With JB, S1 With JB, S0 with JA, S0 with JC) are tried by me which is not working.

I have read the files on the USB stick only M0+JB configuration. There is a video which is including try to format, read and write operations.

[WhatsApp Video 2021-05-31 at 22.04.06.zip](https://github.com/keirf/FlashFloppy/files/6571946/WhatsApp.Video.2021-05-31.at.22.04.06.zip)

What can I do? What is your suggestion and please give me your all suggestions at once because I am going to be near the machine only one time for 30 minutes? Thank you!

hikodaci, 2021-06-01:
Also, I tried .hfe, .img and .ima file.

hikodaci, 2021-06-02:
Can someone help me please?

keirf, 2021-06-02:
I know nothing about the CNC machine I'm afraid. Other things to try are a prerelease of v4 (I can find a link later to an automated build) and another usb stick.

closed 2021-06-24 by keirf