keirf/flashfloppy #484: LCD and OLED do not work after flashing firmware?  (ie. blank display)

sofakng, 2021-06-01:
I've found a refurbished Gotek with an STM32 chip and I turned it on with the default firmware and LCD worked fine.  (ie. it displayed "0")

However, after flashing v3.25 firmware it now doesn't display anything.  I can reflash the firmware (using serial) without any problems and I also see 3.3v / GND on the LCD header fine.

It's possible this STM32 is defective, but is there anything I can try?

Can I reflash the OEM firmware (I don't have a copy) to see if the LCD comes back to life?

keirf, 2021-06-24:
The factory firmware is not available to try reflashing. I assume it is a normal 3-digit display Gotek?

sofakng, 2021-06-24:
Yes, it is a normal 3-digit display.

keirf, 2021-06-24:
No idea I'm afraid. I would probably put serial logging firmware on it and see if I could diagnose what's up but without that level of diagnostic who can really say what's up. If you had a second Gotek you could try swapping display to see whether it is PCB or display at fault.

closed 2021-06-24 by keirf

sofakng, 2021-06-24:
Where can I obtain serial logging firmware?

(Sorry for the comment after the issue is closed;  I'll continue troubleshooting on my own if I can find this other firmware?)

keirf, 2021-06-24:
Look in CI workflow here: https://github.com/keirf/FlashFloppy/actions/workflows/ci.yml
Pick latest build on branch stable-v3.
Click on it, and download the Artifact at bottom of the page.
Unzip the debug zip file within.
Use that HEX file.
You can now collect serial logs on the serial programming header at 3Mbaud (3 million baud rate).

sofakng, 2021-06-24:
Thanks so much!  Are you referring to the UART RX/TX pins near the BOOT0 pin, or the SWD pins?

(That's quite a high baud rate, but I think I have an FTDI FT232L that supports it...)

keirf, 2021-06-24:
UART RX/TX pins. I mostly use crappy PL2303 clones and they support 3M just fine. It may depend what your OS/terminal will pass through to the USB device.

sofakng, 2021-06-24:
What OS/terminal are you using to view the information?

I've successfully flashed and verified the firmware, but Windows Putty (setting "3000000" as the baud rate") doesn't seem to produce any output.

keirf, 2021-06-24:
I use miniterm on Linux. I'm pretty sure putty can work. Perhaps the firmware isn't running properly in that case.