keirf/flashfloppy #488: Agilent 546xx cannot initialize SFRC2DA-board

linklocal, 2021-06-04:
hi!

i purchased a "SFRM72-DU26 720K USB SSD" as a replacement for an original TEAC FD-05HF (1.44Mb) slim line FD.

flashed the most recent FF-FW (v3.25) using a serial-connection with a galvanic isolated USB2SERIAL (and an external 5V power supply) .  unfortunately my board has **no display** - but a single LED showing traffic at PB10 (7seg-DISPLAY-DAT line).

regardless of:

- any jumpers set or not
- factory default settings (both buttons)
- a given FF.CFG
- any modifications to the FF.CFG (pin02 / pin36 / host /  interface)

at **power on** 
the above mentioned LED is flashing for 2-3s, while in between the usb-stick-activity-led enables for ~1s; followed by a message on the scope:
`system concerns detected:
floppy drive failed startup initialization
`
the same message appears, when removing the floppy at all. no probelms will be detected when connecting the old TEAC-FD. 

i tried to update to the **DEBUG-version: '''alt/logfile/FF_Gotek_Logfile-v3.25.upd'''** to see at least a file of some output.  i placed this file in a partitioned USB-stick (partition of 1GB fat32) root-directory. but i don't see, that this FW is been touched in any way - when releasing the two buttons - nothing happens. even observing  of the current consumption does not show any changes in activity (~83mA for a single board, ~161mA with an USB stick attached) . while waiting for 2min nothing happened. 

as an alternative: how could i flash the debug-version with serial?

TEAK data sheet tells something about **PIN02** which offers **INDEX** .. but this is not select able in configuration. but i don't know if this is the marker for the presence of an FD in the system.

my best regards from linklocal

keirf, 2021-06-04:
You can download a serial debug build here: https://github.com/keirf/FlashFloppy/actions/runs/782964740 (you must be logged in to Github to see the Artifact download link)

Debugging issues with no display/logging at all is no fun so that's your first mission. I daresay the Gotek PCB has the header for attaching OLED/LCD if you obtained one. Or you should be able to get serial out which is v helpful.

linklocal, 2021-06-04:
found a oled-display in my **toy-box** and immediately saw the message: ``FF Update Flash v3.25`` .

FAQ tells: [https://github.com/keirf/FlashFloppy/wiki/Troubleshooting-FAQ](url) remove jumper **JA** (short to GND). someone found in the issue #155 -> remove resistor: **RA**. :)

display switched to:``FlashFloppy v3.25`` and when placing an empty USB-stick ``ERROR* 34`` which leads to an empty USB space - at this time - ok.

lets see what happened when attaching this HW to the scope ...

linklocal, 2021-06-04:
jumpered: **JE, S1** and **none on JD** combined with the following **FF.CFG** the scope is not complaining a missing floppy drive any more. 

FF.CFG
```text
interface = shugart
display-type = oled-128x64
display-order = default
twobutton-action = eject
```

but ``no disk`` could be found from scope, so writing a file will fail.

linklocal, 2021-06-05:
meanwhile i created a few **disks** 

```# mkfs.msdos -C /media/x/xxxx-xxxx/image_00.img 1440```

inside FF this img-file is a legal image of floppy. but i don't know if H/C/S:2/80/18 for a 1.44 is sufficient. 

```# mount -t msdos -o loop /media/x/xxxx-xxxx/image_00.img /mnt```

brings up the possible content of that disk.

but the scope is claiming ``Disk not initialized`` ... for my point of view: there is a difference to a real floppy-img (which i don't have)

keirf, 2021-06-05:
Is the Agilent known to use PC 1M44 FAT formatted disks?

linklocal, 2021-06-05:
yes, it uses normal preformated disks. 

after all i have to report a first success. 

this settings:

```ff.cfg
interface = shugart
host = unspecified
motor-delay = 10
twobutton-action = eject
display-type = oled-128x64
```

combined with a 720kb SD-floppy image
``mkfs.msdos -C /media/x/xxxx-xxxx/disk_720.img 720``

adding ``motor-delay`` was the key which made the communication possible

 [image]

now i have to find the **density** or **!density** signal for enabling 1440kb.

keirf, 2021-06-05:
Did you do hardware mod to bring the motor line to the Gotek MCU then? Otherwise motor-delay should have no effect.

linklocal, 2021-06-06:
i didn't mod the HW. but i had two observations while peeking and poking around. i _prepared a couple of different disk-images in sizes and formats_ on the same USB-memory - non worked. the other observation was _booting the scope with a USB-stick connected failed_ but worked without. 

the scope can only do simple things. like writing an image or reading or writing a configuration. but failed in any case - with different error-messages - when stepping through the possible candidates. 

at least i configured ``track-change = realtime`` and it started to work on the 720kb-image - but writing was very slow. 

finally i changed back and for other possible candidates: ``write-drain = realtime/instant`` and even ``motor-delay = 100`` which worked and all other switches set to their defaults. so i guessed `` motor-delay = 10`` is the one. even booting with an attached USB-key is now working - which was the reason for the "hello world"-image.

latest available original instrument-firmware  ``sys5462x.jzp, 700188bytes, Apr 25  2005``.  
while checking the size i come to the conclusion that the 720kb-limit might be correct.

keirf, 2021-06-06:
Perhaps it just provides enough delay when first image is inserted.... I wonder if you need instead something like `chgrst=delay-3` and another experiment would be `interface=ibmpc`

linklocal, 2021-06-07:
tried ``interface = ibmpc`` which didn't work. 
in a second guess i set  ``motor-delay = ignore``  and added ``chgrst=delay-3`` which worked ... in the same way like the above configured stuff.

thank you for the assistance :)

keirf, 2021-06-07:
Ok, that chgrst line is the magic :)

closed 2021-06-07 by keirf

linklocal, 2021-06-10:
nevertheless of the closed issue - i would like to add the solution for the 720 / 1440kb problem.

the original TEAK FD-05HF disk drive offers to the host the signal **HD OUT** on **pin9**. the data sheet tells if: ``HD-OUT is LOW for 2DD  -> 720kb`` resp. **HD-OUT is HIGH for 2HD -> 1440kb** - which i wanted to achieve . 

this pin is connected to a pullup-resistor ``PU14`` and could be pulled-down only by a jumper: ``JE`` - unluckily not by a signal from the MCU. removing jumper JE and offering a ``mkfs.msdos -C FD-THREE.img 1440`` generated image and everything worked perfectly.

with my best regards